NixOS/nixos-hardware #830: Towards a better NixOS on ARM (Raspberry Pi 4 with UEFI Firmware)

MarkusLohmayer, 2024-01-06:
When I got my Raspberry Pi 4, I figured that according to the post [Planning for a better NixOS on ARM](https://discourse.nixos.org/t/planning-for-a-better-nixos-on-arm-and-other-non-x86-64-systems/15346), it would be a good idea to first install the [Raspberry Pi 4 UEFI Firmware](https://github.com/pftf/RPi4). Similarly to the guide [Installing NixOS on the Raspberry Pi 4 with UEFI and ZFS](https://carjorvaz.com/posts/nixos-on-raspberry-pi-4-with-uefi-and-zfs/), I went ahead and installed NixOS in a fairly standard way.

Unfortunately, the built-in sound card (and presumably other peripherals) do not work and so I figured I should see what the nixos-hardware project has to offer. Unfortunately though it seems to be the case that nixos-hardware or rather the [device-specific kernel](https://github.com/raspberrypi/linux/) that nixos-hardware uses does not work with the UEFI firmware. So I wonder if anything is already known about this problem.

More to the point of this issue though, the "better NixOS on ARM" approach suggests to stick to the mainline kernel, which begs the following question. Is there a way to leverage the device-specific configuration (for rpi4) offered by nixos-hardware, while using the mainline kernel?

Mic92, 2024-01-06:
You can override the used kernel like this:

```
{
 boot.kernelPackages = pkgs.linuxKernel.packages.linux;
}
```

Mic92, 2024-01-06:
I don't know enough of the raspberry pi 4 to know wether the upstream kernel has much too offer these days over the mainline one.

MarkusLohmayer, 2024-01-06:
Dear @Mic92 thanks for your answers!

> ```
> {
>  boot.kernelPackages = pkgs.linuxKernel.packages.linux;
> }
> ```

The above line gives me an error saying that the attribute `linux` is missing.
However, I had already tried `kernelPackages = pkgs.linuxPackages;` before together with `loader.generic-extlinux-compatible.enable = false;` as I have `loader.systemd-boot.enable = true;`.

Perhaps this is really would be all that is needed to apply the "ARM with UEFI and mainline kernel approach".

I now realized that the issue which prevents further progress is probably that audio also seems to be broken with the traditional approach, see #703. I plan to update this issue once #703 is fixed. If you prefer to meanwhile close this issue, it is okay with me. Besides me wanting to get audio working, I guess it is important to consider (and document) the "UEFI approach".

Mic92, 2024-01-06:
Right it should be

```
{
 boot.kernelPackages = pkgs.linuxPackages;
}
```

Also:

```
{
 boot.kernelPackages = pkgs.linuxPackages_latest;
}
```

gives you the latest lts kernel.

closed 2024-01-06 by Mic92