NixOS/nixos-hardware #854: Add hydra jobset to repository

damianoognissanti, 2024-02-02:
This is for discussing the addition of a hydra jobset to the repository,

damianoognissanti, 2024-02-02:
I have set up a local hydra instance and have tried to add linux-surface as a flake: 
https://github.com/damianoognissanti/TestHydra/tree/main

I get the following error message:
```
in job ‘packages.x86_64-linux.microsoft-surface-pro-intel.microsoft-surface.surface-control.enable’:
error: attribute 'packages.x86_64-linux.microsoft-surface-pro-intel.microsoft-surface.surface-control.enable' is a Boolean, which is not supported

in job ‘packages.x86_64-linux.microsoft-surface-pro-intel.microsoft-surface.ipts.enable’:
error: attribute 'packages.x86_64-linux.microsoft-surface-pro-intel.microsoft-surface.ipts.enable' is a Boolean, which is not supported

in job ‘packages.x86_64-linux.microsoft-surface-pro-intel.imports’:
error: attribute 'packages.x86_64-linux.microsoft-surface-pro-intel.imports' is a list, which is not supported
```

Does this mean that the surface files must be rewritten, or is there something wrong with the flake?

damianoognissanti, 2024-03-05:
@Mic92 do you know how to write code for adding a hydra jobset? I can't get it to work and I don't find any documentation or examples that I can use either. I am just fumbling in the dark...

RossComputerGuy, 2025-09-06:
We've been looking into this, it is definitely something we're considering now.