NixOS/nixos-hardware #871: Starlabs Support

w3irdrobot, 2024-02-20:
It'd be great to have some support for [Star Labs](https://starlabs.systems) laptops. In my case, specifically the Starbook VI.

w3irdrobot, 2024-02-21:
```
~ 2 $ cat /sys/class/dmi/id/product_sku 
B6-I
```

i have been running NixOS on this machine for a few months now. i use it almost daily in the evenings. there's nothing that's a show stopper. however, there definitely seem to be some issues. admittedly it could just be the way it is configured on my end and it's not a hardware issue. 

specifically, the only way to update the firmware is to mount ubuntu via usb and use it that way because `fwupdmgr` doesn't recognize the Star Labs hardware. Star Labs does not officially support NixOS which causes some issues in some places. i think the hardware not being recognized is because `flashrom-starlabs` is not packaged in nixpkgs or anything. 

it also seems to struggle when doing anything somewhat "intensive," which with my setup shouldn't be difficult. however, the machine slows down immensely even when just updating the flake, for example. this could just be the way i have it configured. the initial instance was installed using the the NixOS Gnome image on the NixOS website. however, i've noticed there seem to be small tweaks available for other hardware in this repo and thought that perhaps others might have this issue and could commit them here if they had these tweaks fixed. it appears i'll just have to figure out my problems myself if i'm the only one having them.

samueldr, 2024-02-21:
> because `fwupdmgr` doesn't recognize the Star Labs hardware.

Weird, it *just works* on my end.

```
services.fwupd.enable = true;
services.fwupd.extraRemotes = [
  "lvfs-testing" # might not be needed, config shared across systems
];
```

It's also weird that they don't tell you to use the "pure" UEFI-based updater, and to use Ubuntu. Though they might not have those for coreboot-based builds. (The AMD variant *still* does not have coreboot support...)

> *[other slight issues in last paragraph]*

Yeah, the way they end-up in here is people figuring out things that solve their issue, and sharing.

So two ways you could find solutions are

 - See if anything from this repo for as-similar-as-possible hardware helps
 - See what people on other distros do to help.

As for the latter, really, if the vendor supports a distro, they should be able to tell you *on any of those distro* what to do from a clean slate install to get proper hardware support.

samueldr, 2024-02-21:
[renamed the issue so it's a bit more actionable]

matdibu, 2024-03-29:
The only way I got `fwupd` to work on a `StarBook VIr2 (B62-I)` was with modified versions of `fwupd` and `flashrom`, because nixpkgs doesn't ship a version of fwupd with the flashrom plugin enabled, and for that to work it needs the meson build of flashrom.

This was my working config: https://codeberg.org/mateidibu/nix-config/src/commit/3f82645c9e5b5ae091530ac09e9ac484379fa071/hosts/nix-starbook/hardware/fwupd.nix

https://github.com/Roger/flashrom-meson-nix

matdibu, 2024-03-29:
Aside from that, things like `GuC` and `HuC` seem to be enabled by default, so I don't see anything else that needs to be done other than `common-pc` and `common-pc-ssd`.

matdibu, 2024-03-29:
@samueldr so on your `B6-A` you could update the BIOS via fwupd without any additional config other than enabling the service?

samueldr, 2024-03-29:
AFAIK nothing else is needed.

The only config about fwupd in my personal config is the one I shared earlier.