NixOS/nixos-hardware #874: gigabyte b550 ds3h ac

robabo, 2024-02-26:
Hi, I just updated the bios on my gigabyte motherboard to the latest (F18d), and Nixos will no longer boot without manually selecting the drive.  I have tried changing the kernel to 'latest' and 'zen'.  I have tried both systemd and grub.  I have tried both a nvme drive and a solid state drive.  When I do not manually select the boot device, uefi fails to find the EFI target and just hangs.  Nixos was booting perfectly fine before the bios update.  I'm dual booting with Arch, and Arch boots without problem.   PLease inform me if this is the wrong place to post this.  Thanks. 
  
Motherboard: gigabyte b550 ds3h ac
CPU: Ryzen 7 5800X

samueldr, 2024-02-27:
Putting on my guessing hat, I am guessing the BIOS update for the board also resets the NVRAM, which includes the EFI boot order. (It's a common enough issue through many motherboard vendors).

I believe using `sudo nixos-rebuild switch --install-bootloader` will re-configure the boot entries here.

robabo, 2024-02-27:
I ran that command whenever I changed the bootloader and kernel. I also ensured Nixos was first in the bios boot sequence. I still have to manually select Nixos or it will hang.

robabo, 2024-03-30:
The f18 bios update has been removed from the gigabyte support site.  The f17 bios update boots nixos without problem.  So i'm closing this issue.

closed 2024-03-30 by robabo