NixOS/nixos-hardware #882: pinebook pro: update known issue info

xeniarose, 2024-03-04:
The current (readme for pinebook pro)[https://github.com/NixOS/nixos-hardware/blob/master/pine64/pinebook-pro/README.md] mentions a known issue with "`rockchipdrm` and `efifb`"

It looks like this was fixed in <https://github.com/Tow-Boot/Tow-Boot/issues/8>, in kernel 5.14. I spent a bit of time working on booting the UEFI installer image on pinebook pro with the additional `efifb=off` option only to find out it is currently unnecessary. Should the section on it be removed from the README, or at least updated to state that 5.14+ resolves it?

samueldr, 2024-03-04:
I think it could be removed entirely. There's been three kernel LTSes since.

xeniarose, 2024-03-04:
should i submit a PR?

samueldr, 2024-03-04:
Sure, the `nixos-hardware` repo is for users to update with the current state of things :)

Mic92, 2024-03-25:
Feel free to submit a PR.

closed 2024-03-25 by Mic92