NixOS/nixos-hardware #883: Surface Pro 6 Front Camera Not Working (Green) and Back Camera Upside Down

balintbarna, 2024-03-04:
Hi everyone

I have a Surface Pro 6 running NixOS, including the microsoft/surface/surface-pro-intel nixos-hardware config.
When I open the gnome camera app I can switch between the back and the front camera. The front camera image is fully completely green, the back camera image works but is upside down and not amazing quality-wise.

Is there anything I can try to get that front camera working and maybe even rotate the back camera the right side up?
@damianoognissanti @mexisme @Mic92

Mic92, 2024-03-05:
🤷

damianoognissanti, 2024-03-05:
Did you use the Gnome camera app when trying your camera? I had that issue too after just installing the kernel with patches. What you need to do is:

1) Install a custom `libcamera` that follows the instruction on the linux-surface page:
This one works for me: https://github.com/damianoognissanti/libcamera-surface
You can install it by putting the file in your configuration.nix folder and import it with "./libcamera.nix" right under the import of hardware-configuration.nix

2) Install kernel modules for `v4l2loopback`, check this:
https://github.com/NixOS/nixos-hardware/issues/523#issuecomment-1396797530

3) Run the `gst-launch-1.0` command found in step 2 to launch the camera (after a reboot). You also need to install gstreamer plugins for this. I installed
```
gst_all_1.gstreamer
gst_all_1.gst-plugins-base
gst_all_1.gst-plugins-good
```

While the command is running the camera should work.

balintbarna, 2024-03-06:
@damianoognissanti Thank you for the reply. It still doesn't work. What happens now and also before, is that when I open the camera app, it activates the white led by the front camera, but the image is just a green blob. If I switch cameras, it activates the little white led by the back camera and I can actually see some image from that camera, but it is upside down and very dark.

My setup:

```
let
  nixos_hardware = fetchGit {
    name = "nixos-hardware-2024-03-03";
    url = "https://github.com/NixOS/nixos-hardware/";
    rev = "59e37017b9ed31dee303dbbd4531c594df95cfbc";
    submodules = true;
  };
  libcamera_surface = fetchGit {
    name = "libcamera-surface-0.2.0";
    url = "https://github.com/damianoognissanti/libcamera-surface/";
    rev = "7268e5ed4389d8b2390321dd4f47da200bd75fde";
    submodules = true;
  };
  ...
in {
  imports = [
    /etc/nixos/hardware-configuration.nix
    "${nixos_hardware}/microsoft/surface/surface-pro-intel/"
    "${libcamera_surface}/libcamera.nix"
    ...
  ];
  ...
}
```

Do any of these show something useful?

```
$ cam -l
[0:35:11.528759281] [12121]  INFO IPAManager ipa_manager.cpp:143 libcamera is not installed. Adding '/nix/store/src/ipa' to the IPA search path
[0:35:11.534521505] [12121]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0
[0:35:11.577528953] [12124] ERROR IPAProxy ipa_proxy.cpp:149 Configuration file 'ov8865.yaml' not found for IPA module 'ipu3'
[0:35:11.577839762] [12124]  INFO IPU3 ipu3.cpp:1141 Registered Camera[0] "\_SB_.PCI0.I2C3.CAMR" connected to CSI-2 receiver 0
[0:35:11.579105282] [12124] ERROR IPAProxy ipa_proxy.cpp:149 Configuration file 'ov5693.yaml' not found for IPA module 'ipu3'
[0:35:11.579287291] [12124]  INFO IPU3 ipu3.cpp:1141 Registered Camera[1] "\_SB_.PCI0.I2C2.CAMF" connected to CSI-2 receiver 1
```

```
[nix-shell:~]$ gst-launch-1.0 _SB_.PCI0.I2C2.CAMF
ERROR: pipeline could not be constructed: unexpected reference "_SB_" - ignoring.

[nix-shell:~]$ gst-launch-1.0 _SB_.PCI0.I2C3.CAMR
ERROR: pipeline could not be constructed: unexpected reference "_SB_" - ignoring.
```

```
$ dmesg | grep ov5693
[    7.604542] ov5693 i2c-INT33BE:00: supply avdd not found, using dummy regulator
[    7.604590] ov5693 i2c-INT33BE:00: supply dovdd not found, using dummy regulator
[    7.604610] ov5693 i2c-INT33BE:00: supply dvdd not found, using dummy regulator
```

```
$ LIBCAMERA_LOG_LEVELS=*:0 cam --list
[0:46:41.863450278] [16515]  INFO IPAManager ipa_manager.cpp:143 libcamera is not installed. Adding '/nix/store/src/ipa' to the IPA search path
[0:46:41.863588405] [16515] DEBUG IPAModule ipa_module.cpp:334 ipa_ipu3.so: IPA module /nix/store/1adx0amavy8ms0pl0ddg1vbwgvzz4kjz-libcamera-0.2.0/lib/libcamera/ipa_ipu3.so is signed
[0:46:41.863632677] [16515] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/nix/store/1adx0amavy8ms0pl0ddg1vbwgvzz4kjz-libcamera-0.2.0/lib/libcamera/ipa_ipu3.so'
[0:46:41.863664470] [16515] DEBUG IPAModule ipa_module.cpp:334 ipa_vimc.so: IPA module /nix/store/1adx0amavy8ms0pl0ddg1vbwgvzz4kjz-libcamera-0.2.0/lib/libcamera/ipa_vimc.so is signed
[0:46:41.863693520] [16515] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/nix/store/1adx0amavy8ms0pl0ddg1vbwgvzz4kjz-libcamera-0.2.0/lib/libcamera/ipa_vimc.so'
[0:46:41.863729242] [16515]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0
[0:46:41.863898382] [16518] DEBUG Camera camera_manager.cpp:69 Starting camera manager
[0:46:41.872100825] [16518] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "ipu3-imgu" created from /dev/media1
[0:46:41.872148693] [16518] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media1 due to 12 missing dependencies
[0:46:41.873898936] [16518] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media1 found
[0:46:41.873907984] [16518] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media1: ipu3-imgu
[... 102 lines cut ...]
	/dev/video5
	/dev/video6
	/dev/video7
	/dev/video8
	/dev/video9
	/dev/video10
	/dev/video11
	/dev/video12
	/dev/video13
	/dev/media1

Intel IPU3 CIO2 (PCI:0000:00:14.3):
	/dev/video0
	/dev/video1
	/dev/video2
	/dev/video3
	/dev/media0
```

```
$ ffplay /dev/video0
...
[video4linux2,v4l2 @ 0x7f5d2c000c80] Not a video capture device.
/dev/video0: No such device
```

balintbarna, 2024-03-06:
As a side note, the camera is not the only issue, the power and volume buttons worked a few times but most of the time they just don't. And about 1/3 of the times when shutting down `iptsd` gets stuck and I have to long-press the power button to power it down.

damianoognissanti, 2024-03-07:
> `gst-launch-1.0 _SB_.PCI0.I2C2.CAMF`

This might be the problem, you don't run the command I told you about. You must have a backslash in the camera name (the name is "\_SB_.PCI0.I2C2.CAMF" as you can see from the `cam -l` command), and you must escape both the backslash as well as the underscore. What happens when you run the following?

```
gst-launch-1.0 libcamerasrc camera-name='\\\_SB_.PCI0.I2C2.CAMF' ! videoconvert ! v4l2sink device=/dev/video0
```

balintbarna, 2024-03-07:
Thank you, that works, sort of. The image is (not) mirrored, so when I move left my image moves right which is pretty disorienting. Do you know if there's a flag or a simple way to mirror it? In addition the quality is pretty low, seems like the white balance is off and either the resolution is low or it looks out of focus. Anything to do about that? And is there a way to avoid having to manually run the terminal command each time? I guess I could create a service but then the camera will be always on...

damianoognissanti, 2024-03-08:
I have a script called startcamera that runs the above command. That command can either be run before you start some camera application by changing the .desktop file, or can be bound to a keybinding. I start it in a terminal though, since I don't use the camera that frequently and it feels like I'm "in control" that way (but that's just a feeling).

If I google gstreamer flip video the very first search result is:
https://gstreamer.freedesktop.org/documentation/videofilter/videoflip.html
which can flip the image in whatever way you want
```
Video-flip-method
The different flip methods.
Members
none (0) – Identity (no rotation)
clockwise (1) – Rotate clockwise 90 degrees
rotate-180 (2) – Rotate 180 degrees
counterclockwise (3) – Rotate counter-clockwise 90 degrees
horizontal-flip (4) – Flip horizontally
vertical-flip (5) – Flip vertically
upper-left-diagonal (6) – Flip across upper left/lower right diagonal
upper-right-diagonal (7) – Flip across upper right/lower left diagonal
automatic (8) – Select flip method based on image-orientation tag
```

Try to check the manual for the other things you wonder too.

balintbarna, 2024-03-09:
Thank you, that's very helpful. Of course it would be nice if we could make it work "out of the box" using the nix configuration, but this will work for now.

closed 2024-03-09 by balintbarna