NixOS/nixos-hardware #885: Feature: NixOS Modules

Lyndeno, 2024-03-09:
Instead of importing individual configs, import a single modules that enables something along the lines of setting options in `hardware.nixos-hardware` or similar.

This would play nicer with some situations such as me wanting to use specializations with different hardware support than the default config. If I inherit the default config, I currently cannot easily disable one hardware import and use another. For example, having a specialization that enables nvidia optimus while the default config disables the nvidia card.

Of course, maybe there is a reason it is currently the way it is, and I am not aware?

Lyndeno, 2024-03-09:
This would make it cleaner to add optionals to hardware configs. I could optionally have an opencl setting for a laptop with an enable option. Without importing another module.

Mic92, 2024-03-25:
> Of course, maybe there is a reason it is currently the way it is, and I am not aware?

We have a wide range of hardware in nixos-hardware with sometimes only 0-2 people that are available to test a particular hardware. As such I think it's a good strategy to avoid too many abstractions and instead make hardware profiles independent from each other so that we can test and change it without breaking different profiles. However feel free to modularize a particular model with more options if you feel that the current structure forces you to use options that you don't want to use. I think this is an independent feature from merging all modules into one namespace.

techieAgnostic, 2024-03-27:
Id like to express want for something like this, even if it is only the .enable option for each module.

As a use-case, a flake that is supporting multiple pieces of hardware, may very likely have a module similar to this:

```
{ lib, pkgs, config, inputs, ... }:
with lib;                      
let
  cfg = config.foo.hardware.framework;
in {

  imports = [
    inputs.nixos-hardware.nixosModules.framework-12th-gen-intel
  ];

  options.foo.hardware.framework = {
    enable = mkEnableOption "default framework options";
  };

  config = mkIf cfg.enable {
    <some other default framework configuration>
  };
}
```

With the intention being to set some default configuration in addition to the contents of the `nixos-hardware` module.

The downside to this is that you cannot gate the import of nixos-hardware, and so it gets included in every configuration when the flakes module list is loaded, which is exasperated for every other piece of hardware in the flake.

The work around to this is to import nixos-hardware directly in each nixosConfiguration, which means you cannot group it logically with the rest of the hardware configuration.

Giving each module an enable option would allow this usecase to work.

Lyndeno, 2024-03-27:
Write now I use `disabledModules` for my specialisations, but it feels gross

non-bin, 2026-10-09:
Right now I have to use things like `"${inputs.nixos-hardware}/common/cpu/intel/haswell"` because not every module is even exposed. I would love to, and be happy to start turning this into a single import. It doesn't have to be a replacement for the current system, just another way to use it

I imagine we would have to:
- Create a global flag like `config.hardware.nixos-hardware.enable` that defaults to `false` but is set to `true` when using the new system
- Rewrite every current option that defaults to `true` to default to `!config.hardware.nixos-hardware.enable`
- Then we can start combining the modules into a single unified import.
 
This would also mean users can reference `nixos-hardware` options in the rest of their config. Right now I have custom options I have to set in addition to importing the hardware modules, but this way the options would always be defined. Here are 2 of my use cases

```nix
# customOptions.nix
{
  options.hardware = {
    amdgpu.enable = lib.mkEnableOption "Enable AMD GPU features";
    intelgpu.enable = lib.mkEnableOption "Enable Intel GPU features";
    nvidia.enable = lib.mkEnableOption "Enable nvidia GPU features";
  };

  config = {
    hardware.intel-gpu-tools.enable = config.hardware.intelgpu.enable; # Allow viewing performance without root
  };
}
```

```nix
# home.nix
{
  home.packages = with pkgs; [
    (btop.override 
      cudaSupport = config.hardware.nvidia.enable;
      rocmSupport = config.hardware.amdgpu.enable;
    })
  ];
}
```