NixOS/nixos-hardware #886: Surface Pro 6 volume and power buttons don't without reloading the kernel module manually

balintbarna, 2024-03-09:
@NobbZ @mexisme 
Have you seen this issue elsewhere?
Basically after every reboot the buttons don't work until I run `sudo modprobe -r soc_button_array && sudo modprobe soc_button_array`
That solves the problem temporarily. Is there a way to automate this?
Also, just running modprobe without -r does not fix the issue.

lVentus, 2024-06-23:
Not only this problem. When I set config of iptsd, then touchscreen in gtk won't function.

Cal-Nine, 2024-09-22:
I have the same issue. For me buttons will work roughly half the time and have to be reset. I submitted the bug at Linux on Surface [here](https://github.com/linux-surface/linux-surface/issues/1481) and somebody commented the drivers need to be built into the kernel option.

You can automate that command by creating a systemd service.
```
  systemd.services.surfaceButtonsFix = {
    script = "modprobe -r soc_button_array && modprobe soc_button_array";
    after = [ "system.slice" ];
  };
```
This creates a systemd service called surfaceButtonsFix that runs the command after loading(?) the unit system.slice. System.slice was kind of an arbitrary guess so it may not work every time but I did several restarts and it seems to have fixed it for me. This was all a bit of guesswork as I'm no expert so use at your own risk.

Cal-Nine, 2024-09-26:
> ```
>   systemd.services.surfaceButtonsFix = {
>     script = "modprobe -r soc_button_array && modprobe soc_button_array";
>     after = [ "system.slice" ];
>   };
> ```

There is a more reliable way to do this with a timer. The name of the service may also require quotes (that's how it is used on [the community wiki page](https://nixos.wiki/wiki/Systemd/Timers)) although I'm sure I've used this option before without quotes and got it to run scripts.

```
  systemd.services."surfaceButtonsFix" = {
    script = "modprobe -r soc_button_array && modprobe soc_button_array";
    serviceConfig = {
      Type = "oneshot";
    };
  };
  systemd.timers."surfaceButtonsFix" = {
    wantedBy = [ "timers.target" ];
    timerConfig = {
      OnBootSec = "30s";
      Unit = "surfaceButtonsFix.service";
    };
  };
```

30 seconds after boot it runs the command.

Cal-Nine, 2024-10-07:
OK I got around to working out a proper solution. 

Find out what your pinctrl driver is for your Surface model by running lsmod and looking for "pinctrl". For my SP7 it's "pinctrl_icelake" – replace that with yours in the examples below.

Then you can load it before soc_button array with:

```
  boot.extraModprobeConfig = ''
    softdep soc_button_array pre: pinctrl_icelake
  '';
```

It seems like another option recommended [here](https://github.com/linux-surface/linux-surface/issues/1392#issuecomment-1989558759) would be to add the pinctrl drivers to initrd with:
`  boot.initrd.kernelModules = [ "pinctrl_icelake" ];` – I haven't tested that, but I wanted to suggest it in case somebody who knows more than me can let us know if it is better for some reason.

mexisme, 2024-10-18:
> OK I got around to working out a proper solution.
> 
> Find out what your pinctrl driver is for your Surface model by running lsmod and looking for "pinctrl". For my SP7 it's "pinctrl_icelake" – replace that with yours in the examples below.
> 
> Then you can load it before soc_button array with:
> 
> ```
>   boot.extraModprobeConfig = ''
>     softdep soc_button_array pre: pinctrl_icelake
>   '';
> ```
> 
> It seems like another option recommended [here](https://github.com/linux-surface/linux-surface/issues/1392#issuecomment-1989558759) would be to add the pinctrl drivers to initrd with: ` boot.initrd.kernelModules = [ "pinctrl_icelake" ];` – I haven't tested that, but I wanted to suggest it in case somebody who knows more than me can let us know if it is better for some reason.
> 
> I wouldn't know how to add that to the nixos-hardware module since you'd need to make sure it uses the correct drivers for the model in question.

Rather than hard-coding this for every Surface model, what about adding a feature flag / module option, allowing people to add this functionality to their build if they need it?
That would allow you to merge it into the `master` branch without impacting everyone.

I do that for Asus Zephyrus GA402x for some finnicky keyboard-like hardware, that only seems to be needed for certain kernel revs:

- https://github.com/NixOS/nixos-hardware/blob/a8dd1b21995964b115b1e3ec639dd6ce24ab9806/asus/zephyrus/ga402x/shared.nix#L25-L34
```
  options.hardware.asus.zephyrus.ga402x = {
    # Kernels earlier than 6.9 (possibly even earlier) tend to take 1-2 key-presses
    # to wake-up the internal keyboard after the device is suspended.
    # Therefore, this option disables auto-suspend for the keyboard by default, but
    # enables it for kernel 6.9.x onwards.
    #
    # Note: the device name is "ASUS N-KEY Device".
    keyboard.autosuspend.enable = (
      mkEnableOption "Enable auto-suspend on the internal USB keyboard (ASUS N-KEY Device) on Zephyrus GA402X"
    ) // { default = defaultAutosuspendEnable; };
```

- https://github.com/NixOS/nixos-hardware/blob/a8dd1b21995964b115b1e3ec639dd6ce24ab9806/asus/zephyrus/ga402x/shared.nix#L70-L76
```
    (mkIf (! cfg.keyboard.autosuspend.enable) {
      services.udev.extraRules = ''
        # Disable power auto-suspend for the ASUS N-KEY device, i.e. USB Keyboard.
        # Otherwise on certain kernel-versions, it will tend to take 1-2 key-presses to wake-up after the device suspends.
        ACTION=="add", SUBSYSTEM=="usb", TEST=="power/autosuspend", ATTR{idVendor}=="0b05", ATTR{idProduct}=="19b6", ATTR{power/autosuspend}="-1"
      '';
    })
```

mexisme, 2024-10-18:
> I wouldn't know how to add that to the nixos-hardware module since you'd need to make sure it uses the correct drivers for the model in question.

Rather than hard-coding this for every Surface model, what about adding a feature flag / module option, allowing people to add this functionality to their build if they need it?
That would allow you to merge it into the `master` branch without impacting everyone.

I do that for Asus Zephyrus GA402x for some finnicky keyboard-like hardware, that only seems to be needed for certain kernel revs:

- https://github.com/NixOS/nixos-hardware/blob/a8dd1b21995964b115b1e3ec639dd6ce24ab9806/asus/zephyrus/ga402x/shared.nix#L25-L34

- https://github.com/NixOS/nixos-hardware/blob/a8dd1b21995964b115b1e3ec639dd6ce24ab9806/asus/zephyrus/ga402x/shared.nix#L70-L76

balintbarna, 2024-10-20:
Closing as workarounds are found and I replaced my SP6

closed 2024-10-20 by balintbarna