NixOS/nixos-hardware #895: T2 MacBook Pro 2019 unable to build

safr-world, 2024-03-23:
Following the instructions [here](https://wiki.t2linux.org/distributions/nixos/installation/) results in the following error:

error: unused option: VIDEO_STK1160_COMMON
error: builder for '/nix/store/pSsluf40rx9ma5z4jsfaxl 7nph119p13-linux-config-6.5.drv' failed with exit code 255
error: 1 dependencies of derivation
Inix/store/ 78cunq2wj81bq9bsc8vw867d4431661-14กนx-6.5. drv
failed to build
error:
1 dependencles of derivation/nix/store/vzm6vyhug5jvs4x126hp3x46w6qw55v1-n1xos-systen-nixos-23.05. 20230222, 988cc95. dry"
failed to build

Mic92, 2024-03-25:
cc @networkException

networkException, 2024-03-25:
I don't have time to look into t2 stuff currently, ill try to get things working properly again though at some point

Mic92, 2024-03-25:
@safr-world maybe a kernel version bump would resolve the option. 
The alternative is to fixup this kernel constraint if its wrong in nixpkgs: https://github.com/NixOS/nixpkgs/blob/1b1c743525de9db9a4d0b86fc48f3c23177ac54f/pkgs/os-specific/linux/kernel/common-config.nix#L850

jdevsc, 2025-06-10:
This issue is resolved right?