NixOS/nixos-hardware #896: Error running test.py on my machine.

jkoking, 2024-03-23:
- When I run this.

```bash
 './tests/run.py '<nixos-hardware/framework/13-inch/7040-amd>'
 ```

-  I get
 
```nix
warning: unknown setting 'allowed-users'
warning: unknown setting 'trusted-users'
error: path has a trailing slash

       at /run/user/1000/tmp5oo2fahn/eval-test.nix:17:124:

           16| {
           17|   "<nixos-hardware/framework/13-inch/7040-amd>" = buildProfile { profile = import /home/jacob/Documents/dev/nixos-hardware/<nixos-hardware/framework/13-inch/7040-amd>; pkgs = pkgs.x86_64-linux; };
             |                                                                                                                            ^
           18| }
error: path has a trailing slash

       at /run/user/1000/tmp5oo2fahn/eval-test.nix:17:124:

           16| {
           17|   "<nixos-hardware/framework/13-inch/7040-amd>" = buildProfile { profile = import /home/jacob/Documents/dev/nixos-hardware/<nixos-hardware/framework/13-inch/7040-amd>; pkgs = pkgs.x86_64-linux; };
             |                                                                                                                            ^
           18| }
error: path has a trailing slash

       at /run/user/1000/tmp5oo2fahn/eval-test.nix:17:124:

           16| {
           17|   "<nixos-hardware/framework/13-inch/7040-amd>" = buildProfile { profile = import /home/jacob/Documents/dev/nixos-hardware/<nixos-hardware/framework/13-inch/7040-amd>; pkgs = pkgs.x86_64-linux; };
             |                                                                                                                            ^
           18| }
error: path has a trailing slash

       at /run/user/1000/tmp5oo2fahn/eval-test.nix:17:124:

           16| {
           17|   "<nixos-hardware/framework/13-inch/7040-amd>" = buildProfile { profile = import /home/jacob/Documents/dev/nixos-hardware/<nixos-hardware/framework/13-inch/7040-amd>; pkgs = pkgs.x86_64-linux; };
             |                                                                                                                            ^
           18| }
error: path has a trailing slash

       at /run/user/1000/tmp5oo2fahn/eval-test.nix:17:124:

           16| {
           17|   "<nixos-hardware/framework/13-inch/7040-amd>" = buildProfile { profile = import /home/jacob/Documents/dev/nixos-hardware/<nixos-hardware/framework/13-inch/7040-amd>; pkgs = pkgs.x86_64-linux; };
             |                                                                                                                            ^
           18| }
error: path has a trailing slash

       at /run/user/1000/tmp5oo2fahn/eval-test.nix:17:124:

           16| {
           17|   "<nixos-hardware/framework/13-inch/7040-amd>" = buildProfile { profile = import /home/jacob/Documents/dev/nixos-hardware/<nixos-hardware/framework/13-inch/7040-amd>; pkgs = pkgs.x86_64-linux; };
             |                                                                                                                            ^
           18| }
error: path has a trailing slash

       at /run/user/1000/tmp5oo2fahn/eval-test.nix:17:124:

           16| {
           17|   "<nixos-hardware/framework/13-inch/7040-amd>" = buildProfile { profile = import /home/jacob/Documents/dev/nixos-hardware/<nixos-hardware/framework/13-inch/7040-amd>; pkgs = pkgs.x86_64-linux; };
             |                                                                                                                            ^
           18| }
error: path has a trailing slash

       at /run/user/1000/tmp5oo2fahn/eval-test.nix:17:124:

           16| {
           17|   "<nixos-hardware/framework/13-inch/7040-amd>" = buildProfile { profile = import /home/jacob/Documents/dev/nixos-hardware/<nixos-hardware/framework/13-inch/7040-amd>; pkgs = pkgs.x86_64-linux; };
             |                                                                                                                            ^
           18| }
error: path has a trailing slash

       at /run/user/1000/tmp5oo2fahn/eval-test.nix:17:124:

           16| {
           17|   "<nixos-hardware/framework/13-inch/7040-amd>" = buildProfile { profile = import /home/jacob/Documents/dev/nixos-hardware/<nixos-hardware/framework/13-inch/7040-amd>; pkgs = pkgs.x86_64-linux; };
             |                                                                                                                            ^
           18| }
error: path has a trailing slash

       at /run/user/1000/tmp5oo2fahn/eval-test.nix:17:124:

           16| {
           17|   "<nixos-hardware/framework/13-inch/7040-amd>" = buildProfile { profile = import /home/jacob/Documents/dev/nixos-hardware/<nixos-hardware/framework/13-inch/7040-amd>; pkgs = pkgs.x86_64-linux; };
             |                                                                                                                            ^
           18| }
error: path has a trailing slash

       at /run/user/1000/tmp5oo2fahn/eval-test.nix:17:124:

           16| {
           17|   "<nixos-hardware/framework/13-inch/7040-amd>" = buildProfile { profile = import /home/jacob/Documents/dev/nixos-hardware/<nixos-hardware/framework/13-inch/7040-amd>; pkgs = pkgs.x86_64-linux; };
             |                                                                                                                            ^
           18| }
error: path has a trailing slash

       at /run/user/1000/tmp5oo2fahn/eval-test.nix:17:124:

           16| {
           17|   "<nixos-hardware/framework/13-inch/7040-amd>" = buildProfile { profile = import /home/jacob/Documents/dev/nixos-hardware/<nixos-hardware/framework/13-inch/7040-amd>; pkgs = pkgs.x86_64-linux; };
             |                                                                                                                            ^
           18| }
error: path has a trailing slash

       at /run/user/1000/tmp5oo2fahn/eval-test.nix:17:124:

           16| {
           17|   "<nixos-hardware/framework/13-inch/7040-amd>" = buildProfile { profile = import /home/jacob/Documents/dev/nixos-hardware/<nixos-hardware/framework/13-inch/7040-amd>; pkgs = pkgs.x86_64-linux; };
             |                                                                                                                            ^
           18| }
error: path has a trailing slash

       at /run/user/1000/tmp5oo2fahn/eval-test.nix:17:124:

           16| {
           17|   "<nixos-hardware/framework/13-inch/7040-amd>" = buildProfile { profile = import /home/jacob/Documents/dev/nixos-hardware/<nixos-hardware/framework/13-inch/7040-amd>; pkgs = pkgs.x86_64-linux; };
             |                                                                                                                            ^
           18| }
error: path has a trailing slash

       at /run/user/1000/tmp5oo2fahn/eval-test.nix:17:124:

           16| {
           17|   "<nixos-hardware/framework/13-inch/7040-amd>" = buildProfile { profile = import /home/jacob/Documents/dev/nixos-hardware/<nixos-hardware/framework/13-inch/7040-amd>; pkgs = pkgs.x86_64-linux; };
             |                                                                                                                            ^
           18| }
error: path has a trailing slash

       at /run/user/1000/tmp5oo2fahn/eval-test.nix:17:124:

           16| {
           17|   "<nixos-hardware/framework/13-inch/7040-amd>" = buildProfile { profile = import /home/jacob/Documents/dev/nixos-hardware/<nixos-hardware/framework/13-inch/7040-amd>; pkgs = pkgs.x86_64-linux; };
             |                                                                                                                            ^
           18| }
error: worker error: error: path has a trailing slash

              at /run/user/1000/tmp5oo2fahn/eval-test.nix:17:124:

                  16| {
                  17|   "<nixos-hardware/framework/13-inch/7040-amd>" = buildProfile { profile = import /home/jacob/Documents/dev/nixos-hardware/<nixos-hardware/framework/13-inch/7040-amd>; pkgs = pkgs.x86_64-linux; };
                    |                                                                                                                            ^
                  18| }
 
```

8bitbuddhist, 2024-05-24:
I know this is an old issue, and I can't say this is the "correct" answer, but try this: 

```sh
./tests/run.py 'framework/13-inch/7040-amd'
```

The test script seems to treat this as a file path instead of a reference to a module, so when it tries interpreting "<" as part of the path, it throws an error.

Mic92, 2024-05-25:
Yes. You need to remove the `<>`.

closed 2024-05-25 by Mic92