NixOS/nixos-hardware #911: network dns error

et2300, 2024-04-13:
i am in qemu dns not really but i in centos
 dns is really

et2300, 2024-04-13:
@Lyndeno

Mic92, 2024-04-13:
Sorry, this repository is about NixOS, not CentOS.

closed 2024-04-13 by Mic92

et2300, 2024-04-14:
[image]

et2300, 2024-04-14:
[image]

et2300, 2024-04-14:
@Mic92

et2300, 2024-04-14:
[image]

et2300, 2024-04-14:
[image]