NixOS/nixos-hardware #913: network dns error

et2300, 2024-04-15:
[image]
[image]

SuperSandro2000, 2024-04-15:
Wrong repo and termux is known to have wonky DNS.

et2300, 2024-04-15:
thus is start qemu command 
`qemu-system-x86_64 -hda  nixos.vmdk -cdrom /storage/emulated/0/Download/latest-nixos-minimal-x86_64-linux.iso -boot d -m 8g -netdev user,id=n1,hostfwd=tcp::2222-:22,dns=223.5.5.5 -device e1000,netdev=n1 -vnc :1 -monitor stdio`

et2300, 2024-04-16:
@everyone

closed 2024-06-19 by Mic92