NixOS/nixos-hardware #918: Is disabling scatter/gather on AMD necessary in newer kernels?

nyabinary, 2024-04-19:
Currently, we just disable scatter and gather on any kernel version newer than 6.2, but is this necessary or has the issue causing it to be disabled in the first place been resolved?
https://github.com/NixOS/nixos-hardware/blob/5d48925b815fd202781bfae8fb6f45c07112fdb2/common/cpu/amd/raphael/igpu.nix#L16

JohnRTitor, 2024-06-15:
Looks like it was added in https://github.com/NixOS/nixos-hardware/pull/598

I don't think that's relevant now on 6.6 LTS or later. Can you double check though, before I send a PR?

Mic92, 2024-06-15:
I am already closing the pull request here, because the causing code will be soon in nixpkgs.

JohnRTitor, 2024-06-15:
@Mic92 could you link the PR? If you are referring to #319865, it's not related to this.

Mic92, 2024-06-15:
I thought it was.

JohnRTitor, 2024-06-15:
Looks like the issue is no longer observed in kernel 6.6+. Thus it is not necessary to set this kernel param. #980 should fix it.

PR #980 (2024-06-15): amd/cpu/raphael-igpu: enable scatter/gather for kernel 6.6+
###### Description of changes
Scatter/gather was introduced in 6.2 kernel and caused flickering and various issues when running iGPU on Ryzen 7000-series CPUs. This issue is not observed in kernel 6.6 or newer versions so we can enable it for kernel 6.6 and newer versions.

Closes #918

###### Things done

- [x] Tested the changes in your own NixOS Configuration
- [x] Tested the changes end-to-end by using your fork of `nixos-hardware` and
      importing it via `<nixos-hardware>` or Flake input
closed 2024-06-15 by mergify[bot]