NixOS/nixos-hardware #936: iptsd failing on suspension

Flameopathic, 2024-05-10:
`iptsd` fails when suspended, but can be manually restarted just fine. reported this to main repo [linux-surface/iptsd#162](https://github.com/linux-surface/iptsd/issues/162), but they noted that it functions well without the alterations made by this flake. why, then, does the flake not use the implementation of `iptsd` that is in `nixpkgs` and instead its own?

Flameopathic, 2024-06-17:
oops, never pinged anyone. @iwanders @mexisme

Flameopathic, 2024-06-17:
in addition, I find it odd how the folders under `microsoft` are structured. why is there one for `surface-pro/(3/9)` and one for `surface/pro-intel`? would there be any issues if I changed this to `surface/pro/(3/9/intel)` or `surface-pro/(3/9/intel)`? i know that some relative paths would have to be updated, but past that, it would be fine, right? also the configuration for the surface pro 3 seems kind of opinionated, is that okay here?

iwanders, 2024-06-17:
I closed https://github.com/NixOS/nixos-hardware/pull/879 because the advice was to use the one from nixpkgs, so I feel we should consolidate on that version.

This is a good find though, does using the module from nixpkgs actually fix this? Because I've seen it happen on my device and I use the [microsoft-surface.ipts](https://github.com/iwanders/nixos-surface/blob/c5d9432fcf2ce9eadeef709d163e17c12be60935/module-desktop.nix#L17) module currently... I didn't really pursue this as I hardly use suspend, but let me know if using the one from nixpkgs works, in that case I'll switch to that for my install :)

I don't know too much about folder structures, I'm still learning when it comes to nix best practices.

Flameopathic, 2024-06-17:
I have had no issues after using the one from nixpkgs. it's possible there are situations where it will break, but I havent encountered any. my recent pull request switches it to the version on nixpkgs and I've been running with it without issue.

Flameopathic, 2024-06-17:
> I don't know too much about folder structures, I'm still learning when it comes to nix best practices.
it's just personal preference here, but consistency would be nice.

iwanders, 2024-06-18:
> I have had no issues after using the one from nixpkgs. it's possible there are situations where it will break, but I havent encountered any.

Switched over and gave it a few suspend tests and the touch / pen kept working, so :+1:  Thanks!

PR #1004 (2024-06-23): surface: remove broken iptsd and surface-control configuration
###### Description of changes
previously, iptsd and surface-control had custom implementations because they weren't on the stable branch of nixpkgs. they are now on the stable branch of nixpkgs, and the custom implementations have bugs that the stable versions on nixpkgs do not have. this fixes those bugs and removes unnecessary modules.

edit for clarification: surface-control configuration was not causing problems, but its configuration seemed to be created to solve a bug that no longer exists in the package, thus it is being rem [... cut]
closed 2024-06-23 by Mic92