NixOS/nixos-hardware #956: [Visionfive 2] Can't cross-compile sd-image linux kernel with current nixpkgs-unstable.

malteneuss, 2024-05-25:
I'm trying to build the Visionfive 2 sd card image, but get stuck a buiilding the Linux kernel using the nixos-hardware guide https://github.com/NixOS/nixos-hardware/blob/d9e0b26202fd500cf3e79f73653cce7f7d541191/starfive/visionfive/v2/README.md :

```
error: builder for '/nix/store/61iivqkj2wh24bbq8dr9arck5ijd8gil-linux-riscv64-unknown-linux-gnu-6.6.0-starfive-visionfive2.drv' failed with exit code 2;
       last 10 log lines:
       > 0000000000000000 R_RISCV_NONE      *ABS*
       > 0000000000000000 R_RISCV_NONE      *ABS*
       > 0000000000000000 R_RISCV_NONE      *ABS*
       >   CPREL   vmlinux.relocs
       > cp: error copying 'vmlinux' to 'vmlinux.relocs': No space left on device
       > make[3]: *** [../arch/riscv/Makefile.postlink:34: vmlinux] Error 1
       > make[2]: *** [../scripts/Makefile.vmlinux:36: vmlinux] Error 2
       > make[2]: *** Deleting file 'vmlinux'
       > make[1]: *** [/build/source/Makefile:1165: vmlinux] Error 2
       > make: *** [../Makefile:234: __sub-make] Error 2
       For full logs, run 'nix log /nix/store/61iivqkj2wh24bbq8dr9arck5ijd8gil-linux-riscv64-unknown-linux-gnu-6.6.0-starfive-visionfive2.drv'.
error: 1 dependencies of derivation '/nix/store/ff06cfzpz0yjqvibrnbpv56qlk5m9fqr-nixos-system-nixos-24.05.20240507.b211b39.drv' failed to build
error: 1 dependencies of derivation '/nix/store/ibsyzihqykwf82rja36pzq3jc85w1ldf-ext4-fs.img-riscv64-unknown-linux-gnu.drv' failed to build
error: 1 dependencies of derivation '/nix/store/pkpwgs986i7rcxqy441bzwxr0zcjn2zn-nixos-sd-image-24.05.20240507.b211b39-riscv64-linux-starfive-visionfive2.img-riscv64-unknown-linux-gnu.drv' failed to build
```

NickCao, 2024-05-25:
> cp: error copying 'vmlinux' to 'vmlinux.relocs': No space left on device

This seems to be a transient error due to your tmpfs filling up.

NickCao, 2024-05-25:
Also may you help test https://github.com/NixOS/nixos-hardware/pull/958?

malteneuss, 2024-05-25:
Thanks for the tmpfs hint. I'll see what i can do about it. Once it works i gladly test your PR.

malteneuss, 2024-05-26:
I really had `boot.tmp.useTmpfs = True` in my NixOS configuration without really understanding the implications. Apparently 22 GB wasn't enough to build the riscv linux kernel. Disabling it, resolved my issue. Many thanks.

closed 2024-05-26 by malteneuss