NixOS/nixos-hardware #970: [ Surface Pro 6 ] No sound on NixOS 24.05

balintbarna, 2024-06-08:
After upgrading my flake based setup to nixos-24.05 and also upgrading NixOS Hardware to the latest version the built-in speaker and microphone disappears. Removing the import of the nixos-hardware module changes to the default kernel where the sound works fine again, however I lose the other features, such as touch and camera.

balintbarna, 2024-06-08:
@mexisme hope you might have some ideas

mexisme, 2024-06-08:
> After upgrading my flake based setup to nixos-24.05 and also upgrading NixOS Hardware to the latest version the built-in speaker and microphone disappears. Removing the import of the nixos-hardware [...]

I had the same problem this week on my Surface Go, and discovered that the driver for my camera no longer works.
This also seems to cause WirePlumber to crash on starting, which means sound no longer works in any shell that needs it ... which I think is "all of them" these days?

Reports online seem to suggest a regression in more recent kernels (around late 6.7.x, from memory) and a few patches, but I haven't had a chance to try them.

However, it also seems weird that this change would be caused by NixOs 24.05, as nixos-hardware is not version-dependent, I believe?

mexisme, 2024-06-09:
> @mexisme hope you might have some ideas

This is the reference I found about the camera:
- https://github.com/linux-surface/linux-surface/issues/1359#issuecomment-1925668220

I will try to test whether this helps me.

Do you think you would also be able to investigate if it might be related to your issue, as well, @balintbarna ?

mexisme, 2024-06-09:
> After upgrading my flake based setup to nixos-24.05 and also upgrading NixOS Hardware to the latest version the built-in speaker and microphone disappears. Removing the import of the nixos-hardware [...]

FYI:
It may also be related to needing to provide a quirk param to the snd-hda driver; this is a very common difficulty, I'm afraid, e.g:
- https://bbs.archlinux.org/viewtopic.php?id=292139

And the reason I mention this is I did discover that there were several updates and quirks added to the snd-hda-intel driver in 6.7, e.g:
- https://www.phoronix.com/news/Linux-6.7-Sound

balintbarna, 2024-06-09:
I don't really know how to test these things. My biggest issue is the lack of sound on surface-linux, I didn't even get as far as to test the cameras. If you can provide me with a couple of lines to test in my nix config I can run it and tell you if it works or not, but I don't know what changes to make on my own.

mexisme, 2024-06-13:
> I don't really know how to test these things.

Ah, ok.

> My biggest issue is the lack of sound on surface-linux, I didn't even get as far as to test the cameras. If you can provide me with a couple of lines to test in my nix config I can run it and tell you if it works or not, but I don't know what changes to make on my own.

I've just seen that https://github.com/llinux-surface/linux-surface has some support for kernel 6.9, which suggests there might be some fixes available.
Kernel 6.9.x had quite a few important fixes, IIRC, so this is probably worth looking at, anyway.

I'll try to build with this new kernel over the next few days, and see if it fixes my problems;  if it does, then will raise a PR and some instructions on how you might be able to test it yourself?

// @balintbarna

mexisme, 2024-06-14:
> > I don't really know how to test these things.
> 
> Ah, ok.
> 
> > My biggest issue is the lack of sound on surface-linux, I didn't even get as far as to test the cameras. If you can provide me with a couple of lines to test in my nix config I can run it and tell you if it works or not, but I don't know what changes to make on my own.
> 
> I've just seen that https://github.com/llinux-surface/linux-surface has some new support for kernel 6.9, which suggests there might be some fixes available. Kernel 6.9.x had quite a few important fixes, IIRC, so this is probably worth looking at, anyway.
> 
> I'll try to build with this new kernel over the next few days, and see if it fixes my problems; if it does, then will raise a PR and some instructions on how you might be able to test it yourself?
> 
> // @balintbarna

Oops, nevermind.
I see that [@stigtsp merged an update to kernel 6.9.3 yesterday.](https://github.com/NixOS/nixos-hardware/pull/973)

balintbarna, 2024-06-14:
I tested with the new kernel and still have no sound. The built-in devices just don't show up as an option. I'm not sure what to look for exactly. I see some wireplumber crashes and error messages, and one of them was making a reference to libcamera and others are always around bluetooth errors. I don't know if that's related at all. Otherwise I'm not sure how to go about debugging.
@mexisme  @stigtsp

lVentus, 2024-06-23:
> > 我确实不知道如何测试这些东西。
> 
> 喔好吧。
> 
> > 我最大的问题是 Surface Linux 上没有声音，我甚至还没来得及测试摄像头。如果你能提供几行代码让我在我的 nix 配置中进行测试，我可以运行它并告诉你它是否有效，但我不知道自己该做哪些更改。
> 
> 我刚刚看到https://github.com/llinux-surface/linux-surface对内核 6.9 提供了一些新支持，这表明可能有一些修复可用。 内核 6.9.x 有相当多的重要修复，如果我没记错的话，所以无论如何，这可能值得一看。
> 
> 我将在接下来的几天内尝试使用这个新内核进行构建，看看它是否能解决我的问题；如果能，那么是否会提出一个 PR 并提供一些关于如何自己测试它的说明？
> 
> //@balintbarna
Sorry, my question is not related to this question, but it is also a problem with surface. When I use the configuration of surface in nixos-hardware directly. Once I enable service.iptsd.enable (because I want to enable disableOnPalm), then all finger touch screens in gtk cannot work. But when I use the following configuration instead of nixos-hardware, this problem is solved, so I think there should be some problem with the wurface part of nixos-hardware.

> > > I don't really know how to test these things.
> > 
> > 
> > Ah, ok.
> > > My biggest issue is the lack of sound on surface-linux, I didn't even get as far as to test the cameras. If you can provide me with a couple of lines to test in my nix config I can run it and tell you if it works or not, but I don't know what changes to make on my own.
> > 
> > 
> > I've just seen that https://github.com/llinux-surface/linux-surface has some new support for kernel 6.9, which suggests there might be some fixes available. Kernel 6.9.x had quite a few important fixes, IIRC, so this is probably worth looking at, anyway.
> > I'll try to build with this new kernel over the next few days, and see if it fixes my problems; if it does, then will raise a PR and some instructions on how you might be able to test it yourself?
> > // @balintbarna
> 
> Oops, nevermind. I see that @stigtsp [merged an update to kernel 6.9.3 yesterday.](https://github.com/NixOS/nixos-hardware/pull/973)
> 
> (Thanks for that, btw, @stigtsp!)

Sorry, my question is not related to this question, but it is also a problem with surface. When I use the configuration of surface in nixos-hardware directly. Once I enable service.iptsd.enable (because I want to enable disableOnPalm), then all finger touch screens in gtk cannot work. 
But when I use the following configuration instead of nixos-hardware, 
https://github.com/NixOS/nixpkgs/issues/240612#issuecomment-1635859341
this problem is solved, so I think there should be some problem with the wurface part of nixos-hardware.

mexisme, 2024-06-27:
> I tested with the new kernel and still have no sound. The built-in devices just don't show up as an option. I'm not sure what to look for exactly. I see some wireplumber crashes and error messages, and one of them was making a reference to libcamera and others are always around bluetooth errors. I don't know if that's related at all. Otherwise I'm not sure how to go about debugging. @mexisme @stigtsp
> 
> ```
> wireplumber[1620]: segfault at 0 ip 00007f93e3f23aaf sp 00007f93d7dff110 error 4 in libcamera.so.0.2.0[7f93e3f04000+b8000] likely on CPU 0 (core 0, socket 0)
> ```
> 
> ```
> jun 14 17:31:27 balint-surfacepro bluetoothd[1111]: src/gatt-database.c:database_add_chrc() Failed to create characteristic entry in database
> jun 14 17:31:27 balint-surfacepro bluetoothd[1111]: src/gatt-database.c:database_add_service() Failed to add characteristic
> jun 14 17:31:27 balint-surfacepro bluetoothd[1111]: src/gatt-database.c:database_add_app() Failed to add service
> jun 14 17:31:27 balint-surfacepro bluetoothd[1111]: src/gatt-database.c:client_ready_cb() Failed to create GATT service entry in local database
> jun 14 17:31:27 balint-surfacepro wireplumber[2068]: spa.bluez5.midi.server: org.bluez.GattManager1.RegisterApplication() failed: GDBus.Error:org.bluez.Error.Failed: Fa>
> ```
> 
> ```
> jun 14 17:31:27 balint-surfacepro bluetoothd[1111]: Endpoint unregistered: sender=:1.88 path=/MediaEndpoint/A2DPSink/aptx_hd
> jun 14 17:31:27 balint-surfacepro systemd[1749]: wireplumber.service: Main process exited, code=dumped, status=11/SEGV
> ```
> 
> ```
> jun 14 17:31:27 balint-surfacepro bluetoothd[1111]: Endpoint unregistered: sender=:1.88 path=/MediaEndpoint/A2DPSource/aptx_hd
> jun 14 17:31:27 balint-surfacepro systemd[1749]: wireplumber.service: Failed with result 'core-dump'.
> ```

Yeah, the new kernel didn't fix my camera for me either — wireplumber is still crashing in the same way.
Some additional googling suggested it might be because the firmware for the camera has been renamed in a way that might be causing this.

I'll try to investigate further this week.

stigtsp, 2024-06-27:
I don't have the Surface Pro hardware, so unable to help debug this. Maybe bumping to 6.9.6 will help?

balintbarna, 2024-06-28:
I’m happy to debug if someone can support me in changing my configuration

alexisspacegirl, 2024-07-22:
I'm also having this same issue, in combination with #886 which makes it difficult to test config tweaks given that the linux-surface kernel currently won't compile, but I'm happy to help debug/test. I'm fairly new to both NixOS and kernel compilation---How would I set up my config to try 6.9.6? Is it a matter of setting up overrides for the version and checksum in `microsoft/surface/common/repos.nix` and/or `microsoft/surface/common/kernel/linux-6.9.x/default.nix`, or would the patch list need to be tweaked too?

alexisspacegirl, 2024-08-07:
This issue is still present for me, even with the recent update to kernel version 6.9.12.

fortydeux, 2024-08-23:
I'm having the same issue on my Surface Book 2 - I can fix my sound by disabling/removing the surface kernel from my Nix Flake and rebuilding with latest linux kernel, but of course touchscreen support is gone. 
As soon as I re-enable the linux-surface kernel, the soundcard no longer shows up. 

I'm getting the following errors:
"""
$> systemctl --user status wireplumber.service 
× wireplumber.service - Multimedia Service Session Manager
     Loaded: loaded (/etc/systemd/user/wireplumber.service; enabled; preset: enabled)
    Drop-In: /nix/store/c5vz6h6rrfbm754j2l35c60a0ldjmp4h-user-units/wireplumber.service.d
             └─overrides.conf
     Active: failed (Result: core-dump) since Fri 2024-08-23 13:16:08 EDT; 6min ago
   Duration: 649ms
 Invocation: dc1958dd19434cd09eca5adf6f522bb1
    Process: 2464 ExecStart=/nix/store/q907bprl8vlzbwzi2x0y031pn1ysjz40-wireplumber-0.5.5/bin/wireplumber (code=dumped, signal=SEGV)
   Main PID: 2464 (code=dumped, signal=SEGV)

Aug 23 13:16:08 archerfish-nixos systemd[1766]: wireplumber.service: Main process exited, code=dumped, status=11/SEGV
Aug 23 13:16:08 archerfish-nixos systemd[1766]: wireplumber.service: Failed with result 'core-dump'.
Aug 23 13:16:08 archerfish-nixos systemd[1766]: wireplumber.service: Scheduled restart job, restart counter is at 5.
Aug 23 13:16:08 archerfish-nixos systemd[1766]: wireplumber.service: Start request repeated too quickly.
Aug 23 13:16:08 archerfish-nixos systemd[1766]: wireplumber.service: Failed with result 'core-dump'.
Aug 23 13:16:08 archerfish-nixos systemd[1766]: Failed to start Multimedia Service Session Manager.
"""

And... 

"""
$> journalctl --user -u wireplumber.service
Jun 16 22:48:40 archerfish-nixos systemd[1807]: Started Multimedia Service Session Manager.
Jun 16 22:48:41 archerfish-nixos wireplumber[1911]: spa.bluez5.midi.server: org.bluez.GattManager1.RegisterApplication() failed: GDBus.Error:org.bluez.Error.Failed: Failed to create entry in database
Jun 16 22:48:41 archerfish-nixos wireplumber[1911]: [0:00:25.843275155] [1911]  INFO IPAManager ipa_manager.cpp:143 libcamera is not installed. Adding '/nix/store/src/ipa' to the IPA search path
Jun 16 22:48:41 archerfish-nixos wireplumber[1911]: [0:00:25.843493366] [1911]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0
Jun 16 22:48:41 archerfish-nixos wireplumber[1911]: [0:00:25.941873624] [2026]  INFO IPAProxy ipa_proxy.cpp:198 libcamera is not installed. Loading proxy workers from '/nix/store/src/libcamera/proxy/worker'
Jun 16 22:48:41 archerfish-nixos wireplumber[1911]: [0:00:25.941906210] [2026] ERROR IPAProxy ipu3_ipa_proxy.cpp:52 Failed to get proxy worker path
Jun 16 22:48:41 archerfish-nixos wireplumber[1911]: [0:00:25.941917493] [2026] ERROR IPAManager ipa_manager.h:43 Failed to load proxy
Jun 16 22:48:42 archerfish-nixos systemd-coredump[2032]: [🡕] Process 1911 (wireplumber) of user 1000 dumped core.
                                                         
                                                         Module libatomic.so.1 without build-id.
                                                         Module liblttng-ust-tracepoint.so.1 without build-id.
                                                         Module liblttng-ust-common.so.1 without build-id.
                                                         Module libnuma.so.1 without build-id.
                                                         Module libyaml-0.so.2 without build-id.
                                                         Module liblttng-ust.so.1 without build-id.
                                                         Module libgcc_s.so.1 without build-id.
                                                         Module libstdc++.so.6 without build-id.
                                                         Module libcamera-base.so.0.2 without build-id.
                                                         Module libcamera.so.0.2 without build-id.
                                                         Module libspa-libcamera.so without build-id.
                                                         Module libspa-v4l2.so without build-id.
                                                         Module libpipewire-module-spa-node-factory.so without build-id.
                                                         Module liblc3.so.1 without build-id.
                                                         Module libspa-codec-bluez5-lc3.so without build-id.
                                                         Module libopus.so.0 without build-id.
                                                         ...
"""

But I'm not sure how to troubleshoot further.  Happy to look into it more and report back if someone has ideas on how to troubleshoot! For now I guess I'll be disabling linux-surface kernel again, since I need sound more than I need touchscreen.

FWIW this issue started for me probably around the same time @balintbarna created the issue, I just hadn't been using this device as much, and didn't get around to trying to troubleshoot it until this week.

fortydeux, 2024-08-23:
As a temporary workaround, what _appears_ to be working for me: I do not need camera, and was seeing a lot of the errors being related to camera support. I don't have deep understanding/insight into this, but a little research showed camera support for my surface book 2 is provided by IPU3 (different for other surface devices, see: https://github.com/linux-surface/linux-surface/wiki/Camera-Support) 
What seems to be working since I don't need camera support on my device, is blacklisting the ipu3_imgu kernel module:
```
  boot.blacklistedKernelModules = [ 
    "ipu3_imgu"
  ]; 
```
I'd still love to get ideas for a better solution, but wanted to follow up in case this helps someone else.

alexisspacegirl, 2024-08-24:
> As a temporary workaround, what _appears_ to be working for me: I do not need camera, and was seeing a lot of the errors being related to camera support. I don't have deep understanding/insight into [...]

I can confirm that this also works on my Surface Pro 6, which uses the same kernel module.

balintbarna, 2024-10-20:
Closing as workaround was found and I stopped using my SP6.

closed 2024-10-20 by balintbarna