NixOS/nixos-hardware #979: hardware.amdgpu.amdvlk conflicts with recently added nixpkgs module

qbit, 2024-06-14:
It looks like `hardware.amdgpu.amdvlk` was recently added to nixpkgs: https://github.com/NixOS/nixpkgs/pull/318175

This conflicts with the one in nixos-hardware:

```
       error: The option `hardware.amdgpu.amdvlk' in module `/nix/store/hnlhs6x2jjzaf2iimaps105h32drgi1n-source/common/gpu/amd' would be a parent of the following options, but its type `boolean' does not support nested options.
        - option(s) with prefix `hardware.amdgpu.amdvlk.enable' in module `/nix/store/dfsgc1mry2f4avzrvcfsll7i5s0hz72c-source/nixos/modules/services/hardware/amdvlk.nix'
        - option(s) with prefix `hardware.amdgpu.amdvlk.package' in module `/nix/store/dfsgc1mry2f4avzrvcfsll7i5s0hz72c-source/nixos/modules/services/hardware/amdvlk.nix'
        - option(s) with prefix `hardware.amdgpu.amdvlk.settings' in module `/nix/store/dfsgc1mry2f4avzrvcfsll7i5s0hz72c-source/nixos/modules/services/hardware/amdvlk.nix'
        - option(s) with prefix `hardware.amdgpu.amdvlk.support32Bit' in module `/nix/store/dfsgc1mry2f4avzrvcfsll7i5s0hz72c-source/nixos/modules/services/hardware/amdvlk.nix'
        - option(s) with prefix `hardware.amdgpu.amdvlk.supportExperimental' in module `/nix/store/dfsgc1mry2f4avzrvcfsll7i5s0hz72c-source/nixos/modules/services/hardware/amdvlk.nix'
```

qbit, 2024-06-14:
Ah, taken care of in https://github.com/NixOS/nixos-hardware/pull/977

closed 2024-06-14 by qbit

bbigras, 2024-06-14:
Why not wait for the PR to be merged before closing this issue?

Mic92, 2024-06-15:
it's merged now.