NixOS/nixos-hardware #992: Let's upstream all profiles in `common/` to nixos

Mic92, 2024-06-20:
Having generic hardware modules profiles available allows us to include them automatically
in tools like nixos-generate-config and co.

Hence I am proposing to upstream all our modules in `common` and remove them from this repository:

```
- [ ] ./cpu/intel/tiger-lake/default.nix
- [ ] ./cpu/intel/sandy-bridge/default.nix
- [ ] ./cpu/intel/kaby-lake/default.nix
- [ ] ./cpu/intel/cpu-only.nix
- [ ] ./cpu/intel/default.nix
  * Note: We are currently doing igpu  configuration in our CPU configuration. We should probably separate these two things when upstreaming, since not all Intel CPUs have a GPU.
- [ ] ./cpu/intel/jasper-lake/default.nix
- [ ] ./cpu/intel/elkhart-lake/default.nix
- [ ] ./cpu/intel/comet-lake/default.nix
- [ ] ./cpu/amd/raphael/igpu.nix
- [ ] ./cpu/amd/zenpower.nix
- [ ] ./cpu/amd/default.nix
- [ ] ./cpu/amd/pstate.nix
- [ ] ./pc/default.nix
- [ ] ./pc/ssd/default.nix
- [ ] ./pc/hdd/default.nix
- [ ] ./pc/laptop/acpi_call.nix
- [ ] ./pc/laptop/hdd/default.nix
- [ ] ./pc/laptop/default.nix
- [ ] ./pc/laptop/ssd
- [ ] ./gpu/nvidia/prime-sync.nix
- [ ] ./gpu/nvidia/prime.nix
- [ ] ./gpu/nvidia/disable.nix
- [ ] ./gpu/nvidia/default.nix
- [ ] ./gpu/amd/default.nix
- [ ] ./gpu/amd/sea-islands
- [ ] ./gpu/amd/sea-islands/default.nix
- [ ] ./gpu/amd/southern-islands
- [ ] ./gpu/amd/southern-islands/default.nix
- [ ] ./gpu/intel/default.nix
- [ ] ./gpu/intel/disable.nix
- [ ] ./gpu/intel.nix
- [ ] ./gpu/nvidia.nix
- [ ] ./gpu/nvidia-disable.nix
```

Open questions:

- Do we upstream them as module options or as importable profiles?

SuperSandro2000, 2024-06-20:
I want to throw the following snippet from [nixos-modules](https://github.com/NuschtOS/nixos-modules/blob/main/modules/intel.nix) into the ring, too.

```nix
    hardware.opengl = {
      extraPackages = with pkgs; [
        intel-compute-runtime # OpenCL library for iGPU
        # video encoding/decoding hardware acceleration
        intel-media-driver # broadwell or newer
        intel-vaapi-driver # older hardware like haswell
      ];
      extraPackages32 = with pkgs.pkgsi686Linux; [
        # video encoding/decoding hardware acceleration
        intel-media-driver # broadwell or newer
        intel-vaapi-driver # older hardware like haswell
      ];
    };
```

I think we should not upstream them as is, otherwise we have 20+ new modules which only contain minimal things. We should probably combine them into a general intel module where we have some options for CPU/GPU/etc.

Mic92, 2024-06-20:
We should be soon able to drop the amdgpu module from nixos-hardware after https://nixpk.gs/pr-tracker.html?pr=319865 landed.

Lyndeno, 2024-06-24:
I like the idea of configurable modules. Then we can have generic cpu/gpu modules where we can specify say, "kaby lake" or "rdna2", etc.

Lyndeno, 2024-06-25:
Something along the idea of:

```nix
hardware.graphics.family = "rdna2";
hardware.processor.family = "zen3";
hardware.graphics.opencl.enable = true;
hardware.processor.enableMicrocode = true;
```

Then have the modules enable the proper settings: opencl packages, vaapi, thermald, patches, microcode (if unsupported in tree, perhaps for apple m series?) all configured for the given hardware families.

Since it is possible to have multiple GPUs, perhaps the graphics.family setting could take a list of families, and ensure setup for each.

Mic92, 2024-06-25:
We should have at least a vendor for both gpu and cpus, to avoid collision names between family names.

Mic92, 2024-06-25:
We want to provide the needed information with this project: https://github.com/numtide/nixos-facter

MattSturgeon, 2024-07-18:
What's the migration path if I'm using some of the deprecated/removed modules?

Neither the warning or the PR description describe what I should do as an end-user :wink: 

The PR talks about upstreaming stuff, but I can't see any obvious recent changes in [nixpkgs/nixos/hardware](https://github.com/NixOS/nixpkgs/commits/master/nixos/modules/hardware)'s history :thinking: 

---

For my case specifically:

I was using the `common-cpu-intel-kaby-lake` module, which was renamed to `common-gpu-intel-kaby-lake` without any deprecation warnings or aliases.
I just got an "attr doesn't exist" error after updating my lockfile, so I had to go hunting for recent changes.

The new name has a deprecation warning, but I can't work out what the alternative is supposed to be?

Lyndeno, 2024-07-18:
There is no alternative currently.

I plan to upstream some of this soon. Perhaps we should not have a deprecation warning until then.

ahydronous, 2024-11-22:
Before you upstream things, a large corrective+feature pass would be nice to do, as I assume afterwards it might be harder to get things in there.

https://github.com/NixOS/nixos-hardware/issues/1205
https://github.com/NixOS/nixos-hardware/issues/1243
https://github.com/NixOS/nixos-hardware/issues/1244
https://github.com/NixOS/nixos-hardware/issues/1245
https://github.com/NixOS/nixos-hardware/issues/1246

In general the Intel GPU side needs a lot of work. There is for example a `comet-lake.nix`, but no such Intel iGPU generation exists. 10th gen (Comet Lake) CPUs use the Ice Lake GPU generation. There's a lot of places where `psr`, `fbc` and `guc` are manually set to `1` or `2` when they already already set to `-1` ( = turn on, unless problematic) by default by the kernel for those devices.

I'm fully willing to do the labor for that once I have the time, but I thought I should bring it up before a merge happens.

Mic92, 2024-11-22:
Why not bring it up, while you upstream a module? The modules that we bring upstream do not need to be literally the same but can be improved version. They just need to cover the same area so. But you can also do pull requests in nixos-hardware if you want. Sorry, I don't have time to go and address your issues, but I would merge your pull requests.

MattSturgeon, 2025-05-14:
It would be really nice if the warnings were only added once there is actually a migration path. 

And additionally, there should be a grace period after the warning is added where deprecated modules continue to work, e.g. as aliases.

The other issue is that currently warnings are only shown when using the flake outputs, not when importing modules directly.

I.e.

1. module gets upstreamed
2. a deprecation warning is added
3. in-tree module remains, or is aliased to the nixos module if possible
4. after a release cycle, in-tre module/alias is removed

An alias module could look like:

```nix
let
  loc =  "/path/to/upstream/module";
  warning = "This module is deprecated and will be removed after 25.11, use the upstream nixos module at `${loc}' instead.";
in
builtins.warn warning
{ modulesPath, ... }:
{
  imports = [ (modulesPath + loc) ];
}
```

By including the warning withing the module itself, you will warn whenever the file is used, even via `"${inputs.hardware}/common/cpu/intel/kaby-lake"`, for instance.

However I still feel strongly that warnings should not be added until there is actually a way for the user to resolve the warning.