NixOS/nixos-hardware #996: References to hardware.opengl.driSupport need to be removed.

lugeha, 2024-06-21:
The option definition `hardware.opengl.driSupport' in `...' no longer has any effect; please remove it.

rdbeni0, 2024-06-21:
the same for :

``` error: The option `hardware.opengl.setLdLibraryPath' does not exist. ```

only1thor, 2024-06-21:
not sure this is related, but i am unable to build my config, and i get this error:
```
 The option `hardware.graphics' does not exist. Definition values:
       - In `/nix/store/6qh2ywkl3lidw6ys45l0zf5xdm125hrl-source/common/gpu/intel': 
           {
             extraPackages = [
               <derivation intel-vaapi-driver-2.4.1>
               <derivation libvdpau-va-gl-0.4.2>
               <derivation intel-media-driver-24.2.1>
           ...
``` 

(sorry if this is unrelated, I'm still quite lost in how all this nix stuff works)

bbigras, 2024-06-21:
You need to update nixpkgs-unstable.

hoh, 2024-06-21:
> not sure this is related, but i am unable to build my config, and i get this error:
>
> ```
>  The option `hardware.graphics' does not exist. Definition values:
>        - In [...]

I am facing the same error today, since this change: https://github.com/NixOS/nixos-hardware/commit/cc634b69c8312c4e88469d3c7e8fb5ecc72e7dc6

Removing the use of nixos-unstable does not solve the problem (my system uses nixos-24.05).

only1thor, 2024-06-21:
> I am facing the same error today, since this change: [cc634b6](https://github.com/NixOS/nixos-hardware/commit/cc634b69c8312c4e88469d3c7e8fb5ecc72e7dc6)
> 
> Removing the use of nixos-unstable does not solve the problem (my system uses nixos-24.05).

I found that [this spesific](https://github.com/NixOS/nixos-hardware/commit/083823b7904e43a4fc1c7229781417e875359a42) commit states that it is the reason for the conflict. 

so i pinned [my input for nixos-hardware](https://github.com/only1thor/framework-nix/commit/59481ce35c8d77e1efd21c716bdd25086b4741e6) to the commit before:
```
...
    nixos-hardware.url = "github:NixOS/nixos-hardware/2f893e185c850bcd6dbf4fbc0c61b1b90d23ff79";
...
```

(also turns out i was not using unstable)

cig0, 2024-06-21:
> @hoh
> 
> > I am facing the same error today, since this change: [cc634b6](https://github.com/NixOS/nixos-hardware/commit/cc634b69c8312c4e88469d3c7e8fb5ecc72e7dc6)
> > Removing the use of nixos-unstable does not solve the problem (my system uses nixos-24.05).
> 
> I found that [this spesific](https://github.com/NixOS/nixos-hardware/commit/083823b7904e43a4fc1c7229781417e875359a42) commit states that it is the reason for the conflict.
> 
> so i pinned [my input for nixos-hardware](https://github.com/only1thor/framework-nix/commit/59481ce35c8d77e1efd21c716bdd25086b4741e6) to the commit before:
> 
> ```
> ...
>     nixos-hardware.url = "github:NixOS/nixos-hardware/2f893e185c850bcd6dbf4fbc0c61b1b90d23ff79";
> ...
> ```
> 
> (also turns out i was not using unstable)

Thank you! I was rolling up my sleeves to start looking at this :sweat_smile:

@toastal, would it help to add a quick note pointing to https://github.com/NixOS/nixos-hardware/issues/996#issuecomment-2183266924 to help folks quickly find the solution and reduce noise here, given that we still have a long way to go before `24.11` is out? Let me know and I'd be glad to quickly create a PR.

only1thor, 2024-06-21:
maybe make a 24.05 branch?
so we can update the inputs in lockstep?

Redlonghead, 2024-06-21:
It looks like with the latest commits it now shows
```
error: The option `hardware.graphics.extraPackages' does not exist. Definition values:
       - In `/nix/store/m0l0hp71pxhjwvqc0mdwf52xlrn0c39f-source/common/gpu/intel':
           [
             <derivation intel-vaapi-driver-2.4.1>
             <derivation libvdpau-va-gl-0.4.2>
             <derivation intel-media-driver-24.2.1>
           ]
```
so its closer now.

Mic92, 2024-06-22:
> maybe make a 24.05 branch on this repo as well? so we can update the inputs in lockstep?

Should work again after: https://github.com/NixOS/nixos-hardware/pull/999

Mic92, 2024-06-22:
driSupport references got removed here: https://github.com/NixOS/nixos-hardware/pull/997

closed 2024-06-22 by Mic92