NixOS/nixos-hardware #1000: 'hardware.opengl.enable' has conflicting definition values

DrymarchonShaun, 2024-06-22:
I just updated my flake.lock, and I'm now getting the following error. 

`/nix/store/m0l0hp71pxhjwvqc0mdwf52xlrn0c39f-source` is this repo, 
`/nix/store/d36839ih5vn074bbpvz9avkgpraf03nh-source` is nixpkgs

```nix
error:
       … while calling the 'head' builtin

         at /nix/store/d36839ih5vn074bbpvz9avkgpraf03nh-source/lib/attrsets.nix:1575:11:

         1574|         || pred here (elemAt values 1) (head values) then
         1575|           head values
             |           ^
         1576|         else

       … while evaluating the attribute 'value'

         at /nix/store/d36839ih5vn074bbpvz9avkgpraf03nh-source/lib/modules.nix:809:9:

          808|     in warnDeprecation opt //
          809|       { value = builtins.addErrorContext "while evaluating the option `${showOption loc}':" value;
             |         ^
          810|         inherit (res.defsFinal') highestPrio;

       (stack trace truncated; use '--show-trace' to show the full trace)

       error: The option `hardware.opengl.enable' has conflicting definition values:
       - In `/nix/store/m0l0hp71pxhjwvqc0mdwf52xlrn0c39f-source/common/gpu/24.05-compat.nix': false
       - In `/nix/store/d36839ih5vn074bbpvz9avkgpraf03nh-source/nixos/modules/programs/steam.nix': true
       Use `lib.mkForce value` or `lib.mkDefault value` to change the priority on any of these definitions.
```

DrymarchonShaun, 2024-06-22:
adding `hardware.opengl.enable = lib.mkForce true;` just gives the same error for hardware.opengl.driSupport32Bit, presumably the same for the other things in `common/gpu/24.05-compat.nix`?

Mic92, 2024-06-22:
Can you check if the `improve-compat` branch fixes the issue for you?

sokai, 2024-06-22:
@Mic92 I tried but got another error. :/

I added `inputs.nixos-hardware.url = "github:nixos/nixos-hardware?ref=improve-compat";` to my flake and …

```
~# nix flake check && nix flake update
• Updated input 'nixos-hardware':
    'github:nixos/nixos-hardware/e6d40db8924c3a663e1f76e0daed09510fea51c3' (2024-06-21)
  → 'github:nixos/nixos-hardware/4631b9e83c8deb9f24a86bc8400b02cac82a03a5' (2024-06-22)
error:
       … while checking flake output 'nixosConfigurations'

         at /nix/store/1m9nx6h3gfr8lnv1jg872n8xm6llwd4p-source/flake.nix:86:5:

           85|   in {
           86|     nixosConfigurations = {
             |     ^
           87|       # NixOS with Gnome on myhost

       … while checking the NixOS configuration 'nixosConfigurations.xolotl'

         at /nix/store/1m9nx6h3gfr8lnv1jg872n8xm6llwd4p-source/flake.nix:88:7:

           87|       # NixOS with Gnome on myhost
           88|       myhost = nixpkgs.lib.nixosSystem {
             |       ^
           89|         system = system;

       (stack trace truncated; use '--show-trace' to show the full trace)

       error: A definition for option `hardware.opengl.enable' is not of type `boolean'. Definition values:
       - In `/nix/store/bagcjirhiixnlpk07i75wxm1mbv0imvb-source/common/gpu/24.05-compat.nix': null
```

Mic92, 2024-06-22:
@sokai my bad. Check again, please.

Mic92, 2024-06-22:
@sokai please check again.

closed 2024-06-22 by Mic92

sokai, 2024-06-23:
Now I can build with OpenGL enabled and with the regular/default input. Thanks a lot!