NixOS/nixos-hardware #1011: TUXEDO Pulse-14 Gen 3: Display artifacts indicating issues with Radeon 780M iGPU

britter, 2024-06-25:
## Description

I'm running GNOME and when I press the Super key, which brings up the activities overview, the animation creates artifacts on the screen. I suspect this is related to the Radeon 780M iGPU not being configured correctly.

@gabyx has reported the same on this device but with Sway.

gabyx, 2024-06-25:
@britter : Thanks for this.
I was looking into `amdgpu.apmd=0` amgpu kernel module flag, that dit not help etc.

britter, 2024-06-26:
@gabyx any idea how to get to the bottom of this? Maybe we should reach out to TUXEDO support asking them if then know anything about this?

gabyx, 2024-06-26:
I already tried, and they said and they mostly say, its not a supported system lol. Do you also see that in Ubuntu?

gabyx, 2024-06-26:
@britter : I mean on the Tuxedo installed Ubuntu?

britter, 2024-06-26:
I immediately replaced Tuxedo OS with NixOS, so I can't tell 😇

gabyx, 2024-06-27:
Lol. hehe, does that thing for you work:
tuxedo-rs -> https://github.com/AaronErhardt/tuxedo-rs
I cannot execute it on my nixos: `tailor_hdwcaps`

On Ubuntu it apparently seems to work, I guess. I have to check again today when I have time.

britter, 2024-06-28:
I installed it from nixpkgs as some point, but it was quite outdated. It also did not seem to work, or I did not understand it. When I started the application there was nothing to click. I can try again installing from the tuxedo-rs flake directly.

Why are you bringing this up? Do you think it is related? As far as I understand this program just brings better ways of configuring the hardware of tuxedo devices.

gabyx, 2024-06-28:
@britter No I dont think its related. For me the application also does not work.
Same issue.  https://github.com/AaronErhardt/tuxedo-rs/issues/103

Maybe we should reach to Tuxedo once more and relate to this technical issue: I would like to know what they actually do for Ubuntu to configure this amdgpu thing...
Would you maybe able to write a ticket to them, casn you CC to my email: gabriel.nuetzi (the thing goes here) sdsc dot ethz dot ch

That would be great!

cguentherTUChemnitz, 2025-11-20:
I can confirm rendering artifacts also on the TUXEDO InfinityBook Pro AMD Gen10 using AMD Ryzen AI 9 HX 370 w/ Radeon 890M × 24.

gabyx, 2025-12-02:
It could be that we verywell miss some kernel parameter compared to the known Ubuntu image.
However my kernel switch above resolved artifacts consitently.

gabyx, 2026-01-17:
Thanks for the proper description: I have a NixOS with Sway where I dont have this rendering problem anymore fortunately:

https://github.com/gabyx/dotfiles/blob/main/nix/hosts/tuxedo-pulse-14/boot.nix
kernel params and 
+ the xan mod kernel,
maybe its also due to something else

and I have never seen the glitches anymore on the internal screen (for me it also only happens there)

also I use way-display with the config in the repo