NixOS/nixos-hardware #1015: libva-vdpau-driver rename from vaapiVdpau breaks common-gpu-nvidia when config.allowAliases is false

x123, 2024-06-26:
This PR https://github.com/NixOS/nixpkgs/pull/317542 breaks the use of `nixos-hardware.nixosModules.common-gpu-nvidia-nonprime` If you are using `config.allowAliases` of `false`. Assume that we need to use the new `libva-vdpau-driver` name:

Related partial stack error when using `config.allowAliases` of `false`:
```       (stack trace truncated; use '--show-trace' to show the full trace)

       error: undefined variable 'vaapiVdpau'

       at /nix/store/s4q588p6gh8jnibxcd67xc1h2g10baps-source/common/gpu/nvidia/default.nix:7:5:

            6|   hardware.graphics.extraPackages = with pkgs; [
            7|     vaapiVdpau
             |     ^
            8|   ];
```

Mic92, 2024-06-27:
Can you make a pull request request that uses the new package if it's present?

```
(if pkgs ? libva-vdpau-driver then pkgs.libva-vdpau-driver else pkgsvaapiVdpau)
```

x123, 2024-06-27:
@Mic92 Absolutely. PR submitted here: https://github.com/NixOS/nixos-hardware/pull/1018

closed 2024-06-27 by mergify[bot]