NixOS/nixos-hardware #1021: Surface pro 6 fail to build

d-dedrat, 2024-06-28:
I have a fresh of install of nixos on my surface and tried to add the surface-pro-intel build to my system through the guidelines. 

`  imports =
    [ # Include the results of the hardware scan.
      <nixos-hardware/microsoft/surface/surface-pro-intel>
      ./hardware-configuration.nix
    ];
`

However I encountered this error:

`/nix/store/r73z9i18vbjcph7k2f3isrysxzx6sqjx-gcc-wrapper-13.2.0/bin/ld: line 269: 177963 Killed                  /nix/store/7v7g86ml0ri171gfcrs1d442px5bi1p3-binutils-2.41/bin/ld ${extraBefore+"${extraBefore[@]}"} ${params+"${params[@]}"} ${extraAfter+"${extraAfter[@]}"}
  BTF     .btf.vmlinux.bin.o
libbpf: failed to get EHDR from .tmp_vmlinux.btf
pahole: .tmp_vmlinux.btf: Invalid argument
  LD      .tmp_vmlinux.kallsyms1
.btf.vmlinux.bin.o: file not recognized: file format not recognized
make[2]: *** [../scripts/Makefile.vmlinux:37: vmlinux] Error 1
make[1]: *** [/build/linux-6.9.3/Makefile:1160: vmlinux] Error 2
make: *** [../Makefile:240: __sub-make] Error 2
error: builder for '/nix/store/ih1vbzmmpd60rwj26ks1bl5p62wiris2-linux-6.9.3.drv' failed with exit code 2
error: 1 dependencies of derivation '/nix/store/f68dvsfs5q4j6cxhc1npz7kqa9rf5wa1-nixos-system-nixos-24.05.2150.89c49874fb15.drv' failed to build

`

I have seen another issue talking about the BTF, but the situation they were in seemed a little different. 

I can provide additional information if need be.

Mic92, 2024-06-29:
cc @stigtsp

alexisspacegirl, 2024-07-22:
I'm also having this same issue, and I'm also happy to help debug/test.

alexisspacegirl, 2024-07-25:
The message no longer seems related to BTF errors, but the build still doesn't finish.

```
[...snip...]
  CC [M]  drivers/net/usb/qmi_wwan.o
  CC [M]  drivers/net/usb/cdc_mbim.o
  CC [M]  drivers/net/usb/ch9200.o
  CC [M]  drivers/net/usb/aqc111.o
  CC [M]  drivers/net/usb/r8153_ecm.o
  LD [M]  drivers/net/usb/asix.o
  LD [M]  drivers/infiniband/hw/hfi1/hfi1.o
make[2]: *** [../scripts/Makefile.build:485: drivers] Error 2
make[1]: *** [/build/linux-6.9.3/Makefile:1919: .] Error 2
make: *** [../Makefile:240: __sub-make] Error 2
error: builder for '/nix/store/dl2h0r7srbdq26mv71l6dll43zylzava-linux-6.9.3.drv' failed with exit code 2;
       last 25 log lines:
       >   CC [M]  drivers/infiniband/hw/hfi1/verbs.o
       >   CC [M]  drivers/net/usb/usbnet.o
       >   CC [M]  drivers/infiniband/hw/hfi1/verbs_txreq.o
       >   CC [M]  drivers/net/usb/int51x1.o
       >   CC [M]  drivers/infiniband/hw/hfi1/vnic_main.o
       >   CC [M]  drivers/infiniband/hw/hfi1/vnic_sdma.o
       >   CC [M]  drivers/infiniband/hw/hfi1/debugfs.o
       >   CC [M]  drivers/net/usb/cdc-phonet.o
       >   CC [M]  drivers/net/usb/kalmia.o
       >   CC [M]  drivers/net/usb/ipheth.o
       >   CC [M]  drivers/net/usb/sierra_net.o
       >   CC [M]  drivers/net/usb/cx82310_eth.o
       >   CC [M]  drivers/net/usb/cdc_ncm.o
       >   CC [M]  drivers/net/usb/huawei_cdc_ncm.o
       >   CC [M]  drivers/net/usb/lg-vl600.o
       >   CC [M]  drivers/net/usb/qmi_wwan.o
       >   CC [M]  drivers/net/usb/cdc_mbim.o
       >   CC [M]  drivers/net/usb/ch9200.o
       >   CC [M]  drivers/net/usb/aqc111.o
       >   CC [M]  drivers/net/usb/r8153_ecm.o
       >   LD [M]  drivers/net/usb/asix.o
       >   LD [M]  drivers/infiniband/hw/hfi1/hfi1.o
       > make[2]: *** [../scripts/Makefile.build:485: drivers] Error 2
       > make[1]: *** [/build/linux-6.9.3/Makefile:1919: .] Error 2
       > make: *** [../Makefile:240: __sub-make] Error 2
       For full logs, run 'nix-store -l /nix/store/dl2h0r7srbdq26mv71l6dll43zylzava-linux-6.9.3.drv'.
error: 1 dependencies of derivation '/nix/store/4bjrn083aw20zm48ibapiwn4bzya5xcy-linux-6.9.3-modules.drv' failed to build
```

alexisspacegirl, 2024-07-25:
The error seems to be slightly different every time now, but always with a reference to `../scripts/Makefile.build:485` (but not always `drivers`). I'm watching the build more closely this time to see if it might have something to do with running out of memory. I've also added the following to my config to see if it helps at all, based on advice from <https://wiki.nixos.org/wiki/Linux_kernel#Build_fails>:

```nix
boot.kernelPatches = [{
  name = "debug-info-config";
  patch = null;
  extraConfig = ''
    DEBUG_INFO_BTF n
  '';
}];
```

alexisspacegirl, 2024-07-25:
The build succeeded, so I assume the issue was indeed OOM-related. I'll try building again without the `DEBUG_INFO_BTF` patch and see if it'll still finish if I give it enough headroom.

alexisspacegirl, 2024-07-25:
That worked too, so it looks like this might be resolved.