NixOS/nixos-hardware #1039: Legion 16irx8h Speaker Issue

MAkcanca, 2024-07-14:
Using the nixos-hardware/lenovo/legion/16irx8h with latest (6.9.8) linux kernel. Speaker seem to work for a few minutes after boot, then get disabled/shutdown. I tried the default kernel 6.6 as well, same result. There is a firmware specified in https://forums.lenovo.com/t5/Ubuntu/Ubuntu-and-legion-pro-7-16IRX8H-audio-issues/m-p/5210709?page=22#6155866 firmware-TAS-2781... However this requires copying to /lib/firmware normally. I'm not very familiar with NixOS architecture just yet, any help is appreciated to make the 16irx8h work perfectly.

MAkcanca, 2024-08-27:
Running this script https://e2e.ti.com/cfs-file/__key/communityserver-discussions-components-files/6/8306.2pa_2D00_byps.sh as root via `./8306.2pa_2D00_byps.sh 13` works after enabling i2c via `hardware.i2c.enable = true;` on configuration.nix

MAkcanca, 2024-08-27:
Another related discussion including several hacks https://forums.lenovo.com/t5/Ubuntu/Ubuntu-and-legion-pro-7-16IRX8H-audio-issues/m-p/5210709?page=32

felipelalli, 2024-09-23:
Same issue here with my Lenovo Legion Pro 7 16IRX9H.

nixos-discourse, 2024-09-23:
This issue has been mentioned on **NixOS Discourse**. There might be relevant details there:

https://discourse.nixos.org/t/sound-speaker-problem-after-connect-bluetooth-device/52443/1

felipelalli, 2024-11-06:
I found a (temporary?) solution at:

[New Lenovo Legion 7 Pro (2023) no sound from speakers in Linux (Reddit)](https://www.reddit.com/r/LenovoLegion/comments/12752y7/comment/llwx8zj/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button)

It worked to me.

I added it to my configuration file:

```
{ config, lib, pkgs, modulesPath, ... }:

{

  boot.extraModprobeConfig = ''
    options snd-hda-intel model=auto
  '';

  boot.blacklistedKernelModules = [ "snd_soc_avs" ];
}
```

nixos-discourse, 2024-11-06:
This issue has been mentioned on **NixOS Discourse**. There might be relevant details there:

https://discourse.nixos.org/t/sound-speaker-problem-after-connect-bluetooth-device/52443/2

MAkcanca, 2025-01-28:
> I found a (temporary?) solution at:
> 
> [New Lenovo Legion 7 Pro (2023) no sound from speakers in Linux (Reddit)](https://www.reddit.com/r/LenovoLegion/comments/12752y7/comment/llwx8zj/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button)
> 
> It worked to me.
> 
> I added it to my configuration file:
> 
> ```
> { config, lib, pkgs, modulesPath, ... }:
> 
> {
> 
>   boot.extraModprobeConfig = ''
>     options snd-hda-intel model=auto
>   '';
> 
>   boot.blacklistedKernelModules = [ "snd_soc_avs" ];
> }
> ```
> 
> PR: [#1227](https://github.com/NixOS/nixos-hardware/pull/1227)

This sadly does not work for me. Same behavior as before, there is sound right after booting. So you need to constantly play audio. After 10 minutes or so, when you stop playing and speakers become silent, they don't start again. After restart rinse and repeat. Problem is possibly related to speaker drivers going into some kind of power saving mode and you need to either manually trigger it to wake up, or linux driver needs to handle that power management.

felipelalli, 2025-07-14:
Finally I was able to fix it: [Guide: Fixing Speaker Audio on Lenovo Legion Pro 7 with NixOS](https://gist.github.com/felipelalli/6179aac72735fd35ea3a9854beb490e5)

Bladeage, 2026-08-27:
Same laptop here (Legion Pro 7 16IRX8H with the TAS2781 amps), but on Fedora 44 with kernel 7.1, so this isn't a NixOS thing. I spent a day on it last week and can maybe add the actual cause behind the "works after boot, then dies after a few minutes of silence" variant that @MAkcanca describes.

What happens is this: a few seconds after the last sound stops, snd_hda_intel puts the HDA controller into runtime suspend (the power_save parameter). The codec itself wakes up fine afterwards, but the TAS2781 amplifiers behind it are not re-initialised on runtime resume, so the speakers stay silent until the next reboot. A full suspend/resume of the laptop takes a different code path in the driver and brings them back, which is why closing the lid sometimes "fixes" it. It also explains why the inaudible 22 kHz tone works: the controller simply never gets to suspend.

The annoying part is that setting `options snd_hda_intel power_save=0` is often not enough, because power management daemons overwrite the value at runtime. On Fedora it's tuned, which writes 10 into it with every profile switch (both balanced and powersave do that, so plugging or unplugging the charger re-breaks it). On NixOS I'd expect TLP (SOUND_POWER_SAVE_ON_AC / SOUND_POWER_SAVE_ON_BAT) or powerManagement.powertop.enable to do the same. You can see what is actually in effect with

```
cat /sys/module/snd_hda_intel/parameters/power_save
```

If that says anything other than 0, the speakers will die after that many seconds of silence.

What has been stable for me since: keep that value at 0 permanently. So the kernel parameter `snd_hda_intel.power_save=0` plus telling the PM daemon to leave it alone (for TLP that's SOUND_POWER_SAVE_ON_AC=0 and SOUND_POWER_SAVE_ON_BAT=0; on Fedora I ended up with a small tuned profile that forces the audio timeout to 0). I also set WirePlumber's session.suspend-timeout-seconds to 0 for the ALSA nodes so PipeWire doesn't idle the device on its own. No firmware files, no I2C scripts, no background tone.

The real fix would be in the kernel driver, which should restore the amp state on runtime resume.