NixOS/nixos-hardware #1043: Legion 16irx8h Sleep Issue

MAkcanca, 2024-07-16:
**System Information**

    Device: Lenovo Legion 16irx8h
    Kernel Version: 6.9.8
    OS: NixOS with mostly default setup
    Hardware Configuration: Using nixos-hardware/lenovo/legion/16irx8h

**Problem Description**

When waking the laptop from sleep, it successfully resumes and prompts for login. However, after logging in, the system automatically goes back to sleep within approximately 5 seconds. The screen turns off, and the power button starts blinking slowly, indicating sleep mode. This behavior is consistent and repeatable.
**Steps to Reproduce**

1.     Put the laptop to sleep.
2.     Wake the laptop from sleep.
3.     Login when prompted.
4.     Observe the laptop going back to sleep after a few seconds.

**Expected Behavior**

After logging in, the laptop should remain awake and operational.
**Additional Information**

-     This issue occurs consistently after each wakeup from sleep.
-     No additional configurations have been made beyond the default setup for the hardware and kernel.

MAkcanca, 2024-07-18:
`journalctl -b` logs, last 400 lines

```
Jul 18 15:45:42 nixos dockerd[2853]: time="2024-07-18T15:45:42.575700888+03:00" level=error msg="[resolver] failed to query DNS server: 1.1.1.1:53, query: ;queue.amazonaws.com.\tIN\t A" error="read udp 172.18.0.8:55916->1.1.1.1:53: i/o timeout"
Jul 18 15:45:45 nixos .tracker-miner-[2692]: Could not launch metadata extractor: Failed to execute child process “/nix/store/gq81ix20x1qm7nqjpd9cdy5bd9n9x3yn-tracker-miners-3.7.3/libexec/tracker-extract-3” (Permission denied)
Jul 18 15:45:49 nixos .tracker-miner-[2692]: Could not launch metadata extractor: Failed to execute child process “/nix/store/gq81ix20x1qm7nqjpd9cdy5bd9n9x3yn-tracker-miners-3.7.3/libexec/tracker-extract-3” (Permission denied)
Jul 18 15:45:52 nixos systemd[1]: NetworkManager-dispatcher.service: Deactivated successfully.
Jul 18 15:45:55 nixos systemd-logind[1584]: The system will suspend now!
Jul 18 15:45:55 nixos ModemManager[2879]: <msg> [sleep-monitor-systemd] system is about to suspend
Jul 18 15:45:55 nixos dbus-daemon[1528]: [system] Activating via systemd: service name='org.freedesktop.nm_dispatcher' unit='dbus-org.freedesktop.nm-dispatcher.service' requested by ':1.11' (uid=0 pid=1675 comm="/nix/store/3kimhm61nra85879sxqqg7ilsm200zwi-networ" label="kernel")
Jul 18 15:45:55 nixos rtkit-daemon[1919]: Recovering from system lockup, not allowing further RT threads.
Jul 18 15:45:55 nixos .gnome-shell-wr[1885]: Failed to make thread 'KMS thread' normally scheduled: Operation not permitted
Jul 18 15:45:55 nixos systemd[1]: Starting Network Manager Script Dispatcher Service...
Jul 18 15:45:55 nixos dbus-daemon[1528]: [system] Successfully activated service 'org.freedesktop.nm_dispatcher'
Jul 18 15:45:55 nixos systemd[1]: Started Network Manager Script Dispatcher Service.
Jul 18 15:45:55 nixos kernel: wlp0s20f3: deauthenticating from 5c:64:8e:64:3b:12 by local choice (Reason: 3=DEAUTH_LEAVING)
Jul 18 15:45:55 nixos wpa_supplicant[1754]: wlp0s20f3: CTRL-EVENT-DISCONNECTED bssid=5c:64:8e:64:3b:12 reason=3 locally_generated=1
Jul 18 15:45:55 nixos wpa_supplicant[1754]: wlp0s20f3: CTRL-EVENT-DSCP-POLICY clear_all
Jul 18 15:45:55 nixos avahi-daemon[1520]: Interface wlp0s20f3.IPv4 no longer relevant for mDNS.
Jul 18 15:45:55 nixos avahi-daemon[1520]: Leaving mDNS multicast group on interface wlp0s20f3.IPv4 with address 192.168.1.42.
Jul 18 15:45:55 nixos avahi-daemon[1520]: Joining mDNS multicast group on interface wlp0s20f3.IPv4 with address 192.168.1.42.
Jul 18 15:45:55 nixos avahi-daemon[1520]: New relevant interface wlp0s20f3.IPv4 for mDNS.
Jul 18 15:45:55 nixos systemd[1]: Stopped target Host and Network Name Lookups.
Jul 18 15:45:55 nixos systemd[1]: Stopping Host and Network Name Lookups...
Jul 18 15:45:55 nixos systemd[1]: Stopped target User and Group Name Lookups.
Jul 18 15:45:55 nixos systemd[1]: Stopping User and Group Name Lookups...
Jul 18 15:45:55 nixos systemd[1]: Stopping Name Service Cache Daemon (nsncd)...
Jul 18 15:45:55 nixos systemd[1]: nscd.service: Deactivated successfully.
Jul 18 15:45:55 nixos systemd[1]: Stopped Name Service Cache Daemon (nsncd).
Jul 18 15:45:55 nixos systemd[1]: nscd.service: Consumed 36ms CPU time, received 1.7K IP traffic, sent 1.2K IP traffic.
Jul 18 15:45:55 nixos avahi-daemon[1520]: Got SIGHUP, reloading.
Jul 18 15:45:55 nixos avahi-daemon[1520]: No service file found in /etc/avahi/services.
Jul 18 15:45:55 nixos avahi-daemon[1520]: Leaving mDNS multicast group on interface wlp0s20f3.IPv4 with address 192.168.1.42.
Jul 18 15:45:55 nixos avahi-daemon[1520]: Interface wlp0s20f3.IPv4 no longer relevant for mDNS.
Jul 18 15:45:56 nixos systemd[1]: Starting Name Service Cache Daemon (nsncd)...
Jul 18 15:45:56 nixos systemd[1]: Started Name Service Cache Daemon (nsncd).
Jul 18 15:45:56 nixos systemd[1]: Reached target Host and Network Name Lookups.
Jul 18 15:45:56 nixos nsncd[13214]: Jul 18 12:45:56.023 INFO started, config: Config { ignored_request_types: {}, worker_count: 8, handoff_timeout: 3s }, path: "/var/run/nscd/socket"
Jul 18 15:45:56 nixos systemd[1]: Reached target User and Group Name Lookups.
Jul 18 15:45:56 nixos rtkit-daemon[1919]: Recovering from system lockup, not allowing further RT threads.
Jul 18 15:45:56 nixos .gnome-shell-wr[1885]: Failed to make thread 'KMS thread' normally scheduled: Operation not permitted
Jul 18 15:45:56 nixos wpa_supplicant[1754]: p2p-dev-wlp0s20: CTRL-EVENT-DSCP-POLICY clear_all
Jul 18 15:45:56 nixos wpa_supplicant[1754]: p2p-dev-wlp0s20: CTRL-EVENT-DSCP-POLICY clear_all
Jul 18 15:45:56 nixos wpa_supplicant[1754]: nl80211: deinit ifname=p2p-dev-wlp0s20 disabled_11b_rates=0
Jul 18 15:45:56 nixos wpa_supplicant[1754]: wlp0s20f3: CTRL-EVENT-DSCP-POLICY clear_all
Jul 18 15:45:56 nixos wpa_supplicant[1754]: wlp0s20f3: CTRL-EVENT-DSCP-POLICY clear_all
Jul 18 15:45:56 nixos wpa_supplicant[1754]: nl80211: deinit ifname=wlp0s20f3 disabled_11b_rates=0
Jul 18 15:45:57 nixos rtkit-daemon[1919]: Recovering from system lockup, not allowing further RT threads.
Jul 18 15:45:57 nixos .gnome-shell-wr[1885]: Failed to make thread 'KMS thread' normally scheduled: Operation not permitted
Jul 18 15:45:57 nixos systemd[1]: Starting NVIDIA system suspend actions...
Jul 18 15:45:57 nixos systemd[1]: Starting Pre-Sleep Actions...
Jul 18 15:45:57 nixos systemd[1]: pre-sleep.service: Deactivated successfully.
Jul 18 15:45:57 nixos systemd[1]: Finished Pre-Sleep Actions.
Jul 18 15:45:57 nixos systemd[1]: Reached target Sleep.
[... cut ...]
Jul 18 15:46:53 nixos .org.gnome.Char[13519]: JS LOG: Characters Application exiting
Jul 18 15:46:57 nixos rtkit-daemon[1919]: Recovering from system lockup, not allowing further RT threads.
Jul 18 15:46:57 nixos .gnome-shell-wr[1885]: Failed to make thread 'KMS thread' normally scheduled: Operation not permitted
Jul 18 15:46:58 nixos rtkit-daemon[1919]: Recovering from system lockup, not allowing further RT threads.
Jul 18 15:46:58 nixos .gnome-shell-wr[1885]: Failed to make thread 'KMS thread' normally scheduled: Operation not permitted
```

4JX, 2024-08-04:
Chiming in to say that I've forever had this as well on my 16ACH6H. I think this may be a "general for legion/lenovo devices" issue. Never really dived into why it happens, though.

felipelalli, 2024-09-23:
Same issue here with my Lenovo Legion Pro 7 16IRX9H.

nixos-discourse, 2024-09-23:
This issue has been mentioned on **NixOS Discourse**. There might be relevant details there:

https://discourse.nixos.org/t/lenovo-legion-pro-7-hibernation/52384/1