NixOS/nixos-hardware #1050: Legion 16irx8h Random Freeze

MAkcanca, 2024-07-19:
System Information

Device: Lenovo Legion 16irx8h
Kernel Version: 6.9.8
OS: NixOS with mostly default setup
Hardware Configuration: Using nixos-hardware/lenovo/legion/16irx8h, Quiet Mode(blue) and Hybrid Graphics

Problem Description

I was randomly switching the power settings from power saver to balanced etc. then letting system dim screen multiple times. System suddenly froze, nothing written in journalctl. I had to press and hold the power button.
Latest journalctl logs from last boot, freeze happened around 19:39
```
Jul 19 19:38:25 nixos systemd[1]: Starting Hostname Service...
Jul 19 19:38:25 nixos systemd-timesyncd[1308]: Network configuration changed, trying to establish connection.
Jul 19 19:38:25 nixos systemd-timesyncd[1308]: Network configuration changed, trying to establish connection.
Jul 19 19:38:25 nixos systemd-timesyncd[1308]: Network configuration changed, trying to establish connection.
Jul 19 19:38:25 nixos systemd-timesyncd[1308]: Contacted time server 162.159.200.1:123 (2.nixos.pool.ntp.org).
Jul 19 19:38:25 nixos dbus-daemon[1503]: [system] Successfully activated service 'org.freedesktop.hostname1'
Jul 19 19:38:25 nixos systemd[1]: Started Hostname Service.
Jul 19 19:38:33 nixos .org.gnome.Char[51683]: JS LOG: Characters Application exiting
Jul 19 19:38:37 nixos wpa_supplicant[1720]: wlp0s20f3: CTRL-EVENT-SIGNAL-CHANGE above=0 signal=-61 noise=9999 txrate=526600
Jul 19 19:38:41 nixos wpa_supplicant[1720]: wlp0s20f3: CTRL-EVENT-SIGNAL-CHANGE above=1 signal=-61 noise=9999 txrate=526600
```

4JX, 2024-08-04:
Same as #1043, also happens on my 16ACH6H. I feel it happens more when the laptop is doing a high workload situation, but I do not have a reproducible setup. 

Will try to pull logs next time it happens.

4JX, 2024-08-23:
Hey, this seems to be related to https://discourse.nixos.org/t/kernel-panic-system-freezes-no-bsod/45424 and https://forums.developer.nvidia.com/t/series-550-freezes-laptop/284772. I can [recreate the kernel panics](https://forums.developer.nvidia.com/t/series-550-freezes-laptop/284772/8) part pretty solidly and there's a bunch of people with legions over there.

I've changed my driver to `560.31.02` (another solution proposed is to roll back to `53X.X.X` or to try another driver like `nvidia-open`/`noveau`+`nvk`) and it has at least fixed the shutdown panics and sleep freezes from a quick test, will see how it fares long term.
```nix
hardware.nvidia.package = config.boot.kernelPackages.nvidiaPackages.beta; # Currently 560.31.02
```

MAkcanca, 2024-09-12:
Did not work. 560.35 and 6.10 is not compatible
Throws the error below after building

```
inconsistent kernel versions: 6.10.9
6.6.50
```

Also 560 is not available in latest versions of 24.05 channel, only up to 550. I tried only using unstable for Nvidia then keep it same but sadly no luck.

Current relevant configuration.nix part:
```
let 
  unstable = import <nixos-unstable> { config = { allowUnfree = true; }; };
in {
  imports =
    [ 
      <nixos-hardware/lenovo/legion/16irx8h>
      ./hardware-configuration.nix
    ];
    ...
      boot.kernelPackages = unstable.linuxPackages_6_10;
      hardware.nvidia.package = unstable.linuxPackages.nvidiaPackages.latest;
}
```

4JX, 2024-09-12:
>  in latest versions of 24.05 channel

Ah, I follow `nixos-unstable` (which is pretty darn stable despite its name, you should give it a chance).

> inconsistent kernel versions: 6.10.9
> 6.6.50

> boot.kernelPackages = unstable.linuxPackages_6_10;

You probably have some part of your system still [depending on `6.6`](https://discourse.nixos.org/t/error-returned-when-i-try-to-switch-to-another-kernel-version/8668/4?u=akam) somewhere. I'm doing fine on unstable with [this](https://github.com/4JX/nixos-config/blob/cb32df36eaa635e2b7a88f457a63bcfb18a6d1eb/hosts/terra/boot.nix).

closed 2025-02-25 by MAkcanca