NixOS/nixos-hardware #1051: Build failure: uboot on Star64

SpidFightFR, 2024-07-21:
### Steps To Reproduce

Steps to reproduce the behavior:
1. when compiling `uboot-pine64_star64_defconfig-riscv64-unknown-linux-gnu-3.0.4.drv` i end up on:

<details><summary>Error.txt</summary>

```
nix build --accept-flake-config .\#sd-image-cross-8gb
warning: Git tree '/home/<user>/star64-nixos-flakes' is dirty
cannot build on 'ssh://localhost': error: cannot connect to 'localhost': error: unable to execute 'bash': No such file or directory
error: builder for '/nix/store/drjac9lzd7qv5z29mk30my2ry0x2hjq5-uboot-pine64_star64_defconfig-riscv64-unknown-linux-gnu-3.0.4.drv' failed with exit code 1;
       last 10 log lines:
       > --------------------------
       > |diff --git a/board/raspberrypi/rpi/rpi.env b/board/raspberrypi/rpi/rpi.env
       > |index 30228285ed..0327ef74fa 100644
       > |--- a/board/raspberrypi/rpi/rpi.env
       > |+++ b/board/raspberrypi/rpi/rpi.env
       > --------------------------
       > File to patch:
       > Skip this patch? [y]
       > Skipping patch.
       > 1 out of 1 hunk ignored
       For full logs, run 'nix log /nix/store/drjac9lzd7qv5z29mk30my2ry0x2hjq5-uboot-pine64_star64_defconfig-riscv64-unknown-linux-gnu-3.0.4.drv'.
error: 1 dependencies of derivation '/nix/store/j8qyy3cqv136ccajj2v7i7k70j0y8d8w-pine64-star64-spl-riscv64-unknown-linux-gnu.drv' failed to build
error (ignored): error: cannot unlink '/tmp/nix-build-tcpdump-riscv64-unknown-linux-gnu-4.99.4.drv-2/build/tcpdump-4.99.4': Directory not empty
error (ignored): error: cannot unlink '/tmp/nix-build-riscv64-unknown-linux-gnu-rustc-1.77.2.drv-2/build/rustc-1.77.2-src/src/doc': Directory not empty
error (ignored): error: cannot unlink '/tmp/nix-build-ruby-3.1.5.drv-1/build/ruby-3.1.5': Directory not empty
error (ignored): error: cannot unlink '/tmp/nix-build-qemu-8.2.5.drv-1/build/qemu-8.2.5/roms/ipxe/src/drivers/net': Directory not empty
error (ignored): error: cannot unlink '/tmp/nix-build-riscv64-unknown-linux-gnu-binutils-2.41.drv-1/build/binutils-2.41/bfd': Directory not empty
error (ignored): error: cannot unlink '/tmp/nix-build-linux-riscv64-unknown-linux-gnu-6.9.9.drv-1/build/linux-6.9.9/drivers': Directory not empty
error (ignored): error: cannot unlink '/tmp/nix-build-gnugrep-riscv64-unknown-linux-gnu-3.11.drv-1/build/grep-3.11': Directory not empty
error (ignored): error: cannot unlink '/tmp/nix-build-openssl-riscv64-unknown-linux-gnu-3.0.14.drv-1/build/openssl-3.0.14': Directory not empty
error: 1 dependencies of derivation '/nix/store/zzi5a7zhm1siahz92q1lklhn6avynh9l-pine64-star64-uboot-fit-image-riscv64-unknown-linux-gnu.drv' failed to build
error: 1 dependencies of derivation '/nix/store/b0519n10rbr8wx319ln9bf1rdgbwl52i-nixos-sd-image-24.05.20240714.53e81e7-riscv64-linux-pine64-star64.img-riscv64-unknown-linux-gnu.drv' failed to build

```
</details>

### Build log

<details><summary>Build Logs.txt</summary>

```
nix log /nix/store/44anyq48b31347zgqy11p2a9fp5pswa7-uboot-pine64_star64_defconfig-riscv64-unknown-linux-gnu-3.0.4.drv
warning: The interpretation of store paths arguments ending in `.drv` recently changed. If this command is now failing try again with '/nix/store/44anyq48b31347zgqy11p2a9fp5pswa7-uboot-pine64_star64_defconfig-riscv64-unknown-linux-gnu-3.0.4.drv^*'
sourcing setup hook '/nix/store/kp2j7yn0wzwq5piy494r54dafrh83s6s-gcc-wrapper-13.3.0/nix-support/setup-hook'
sourcing setup hook '/nix/store/9rz99vibv1782dxjkwvxr29s7f2f86wl-binutils-wrapper-2.42/nix-support/setup-hook'
sourcing setup hook '/nix/store/w545rbkal83gjx8456wlapcyqg2gl133-install-shell-files/nix-support/setup-hook'
sourcing setup hook '/nix/store/yq6n8b0mnk0qxzbs3ajsjcp8ziwqylrl-patchelf-0.15.0/nix-support/setup-hook'
sourcing setup hook '/nix/store/iks1pihvbilsh5sy8qvpd638k422w9i8-update-autotools-gnu-config-scripts-hook/nix-support/setup-hook'
sourcing setup hook '/nix/store/j1a71cavysmzgv59v5az8ai0agjbrd70-update-autotools-gnu-config-scripts-hook/nix-support/setup-hook'
sourcing setup hook '/nix/store/h9lc1dpi14z7is86ffhl3ld569138595-audit-tmpdir.sh'
sourcing setup hook '/nix/store/m54bmrhj6fqz8nds5zcj97w9s9bckc9v-compress-man-pages.sh'
sourcing setup hook '/nix/store/wgrbkkaldkrlrni33ccvm3b6vbxzb656-make-symlinks-relative.sh'
sourcing setup hook '/nix/store/5yzw0vhkyszf2d179m0qfkgxmp5wjjx4-move-docs.sh'
sourcing setup hook '/nix/store/fyaryjvghbkpfnsyw97hb3lyb37s1pd6-move-lib64.sh'
sourcing setup hook '/nix/store/kd4xwxjpjxi71jkm6ka0np72if9rm3y0-move-sbin.sh'
sourcing setup hook '/nix/store/pag6l61paj1dc9sv15l7bm5c17xn5kyk-move-systemd-user-units.sh'
sourcing setup hook '/nix/store/jivxp510zxakaaic7qkrb7v1dd2rdbw9-multiple-outputs.sh'
sourcing setup hook '/nix/store/ilaf1w22bxi6jsi45alhmvvdgy4ly3zs-patch-shebangs.sh'
sourcing setup hook '/nix/store/cickvswrvann041nqxb0rxilc46svw1n-prune-libtool-files.sh'
sourcing setup hook '/nix/store/xyff06pkhki3qy1ls77w10s0v79c9il0-reproducible-builds.sh'
sourcing setup hook '/nix/store/ngg1cv31c8c7bcm2n8ww4g06nq7s4zhm-set-source-date-epoch-to-latest.sh'
sourcing setup hook '/nix/store/gps9qrh99j7g02840wv5x78ykmz30byp-strip.sh'
sourcing setup hook '/nix/store/4zymxg18qsjc8r9r2ik2pfd3lqpbc302-riscv64-unknown-linux-gnu-gcc-wrapper-13.3.0/nix-support/setup-hook'
sourcing setup hook '/nix/store/kg8j9l84wals2z4rzxk52vc5vi53mjyi-riscv64-unknown-linux-gnu-binutils-wrapper-2.42/nix-support/setup-hook'
calling 'envBuildBuildHook' function hook 'ccWrapper_addCVars' /nix/store/kp2j7yn0wzwq5piy494r54dafrh83s6s-gcc-wrapper-13.3.0
calling 'envBuildBuildHook' function hook 'bintoolsWrapper_addLDVars' /nix/store/kp2j7yn0wzwq5piy494r54dafrh83s6s-gcc-wrapper-13.3.0
calling 'envBuildBuildHook' function hook 'ccWrapper_addCVars' /nix/store/9rz99vibv1782dxjkwvxr29s7f2f86wl-binutils-wrapper-2.42
[... 153 lines cut ...]
--------------------------
|diff --git a/board/raspberrypi/rpi/rpi.env b/board/raspberrypi/rpi/rpi.env
|index 30228285ed..0327ef74fa 100644
|--- a/board/raspberrypi/rpi/rpi.env
|+++ b/board/raspberrypi/rpi/rpi.env
--------------------------
File to patch: 
Skip this patch? [y] 
Skipping patch.
1 out of 1 hunk ignored

```
</details>

### Additional context

Riscv64 CPU, Star64 SBC

### Notify maintainers

### Metadata

Please run `nix-shell -p nix-info --run "nix-info -m"` and paste the result.

N.A: cannot run the command as of now, stuck in rebuild as of now.

SpidFightFR, 2024-08-23:
Hello, after a *quite long* investigation on my side, it appears it's because of the patch being updated. In fact in a previous state of the repo the patch was compatible.

I think this is because the uboot repo for the star64 is stuck in time, probably unlike the rpi variant.

I tried to make an overlay but unlike packages like mesa i'm unable to understand how to make an overlay that ignores the rpi patch...

I tried this but without success:
```
{ config, lib, pkgs, inputs, ... }:
{
    nixpkgs.overlays = [
        (final: prev: {
                ubootPine64 = inputs.nixpkgs.legacyPackages.${pkgs.system}.ubootPine64.overrideAttrs (oldAttrs: {
                patches = [];
                });
            }
        )
    ];
}
```

Not gonna lie, i lack the skills, and the absence of documentation on this doesn't make things easier.
if someone could help me here it would be much appreciated. :pray:

humaidq, 2024-08-24:
Hello! I finally got it built. Check out these commits:
https://github.com/humaidq/u-boot/commit/b801e9996c537f3225bb7acb2e27feeb9a98a198
https://github.com/humaidq/nixos-hardware/commit/1d80ca39335442296a275a9b3e2b0de73bdca93c

This is working on nixos-unstable.

SpidFightFR, 2024-08-24:
> Hello! I finally got it built. @SpidFightFR check out these commits: [humaidq/u-boot@b801e99](https://github.com/humaidq/u-boot/commit/b801e9996c537f3225bb7acb2e27feeb9a98a198) [humaidq@1d80ca3](https://github.com/humaidq/nixos-hardware/commit/1d80ca39335442296a275a9b3e2b0de73bdca93c)
> 
> This is working on nixos-unstable.

Hey thanks so much for your answer, you're finally bringing me hope here !

I've added your repo to the inputs of my flake, looks like that overall:

```
inputs = {
    nixpkgs = {
        url = "github:nixos/nixpkgs/nixos-24.05";
    };

    nixpkgs-unstable = {
            url = "github:nixos/nixpkgs/nixos-unstable";
        };

    nixos-hardware.url = "github:nixos/nixos-hardware";

    nixos-hardware-patched.url = "github:humaidq/nixos-hardware";

    home-manager = {
        url = "github:nix-community/home-manager/release-24.05";
        inputs.nixpkgs.follows = "nixpkgs";
    };
  };
```

I also exposed them with the `output option`.

If i wish to make something like:
Everything stays on the regular 24.05 branch, with the regular nixos-hardware, except for uboot that is compiled using your work...

If it doesn't work with 24.05 that's not a problem i can use the unstable input.
But how can i make the overlay for uboot ?

For something like mesa i've made :
```
  nixpkgs.overlays = [
    (final: prev: {
      mesa = inputs.nixpkgs-unstable.legacyPackages.${pkgs.system}.mesa;
    })
  ];

    environment.systemPackages = with pkgs; [
        mesa
    ];
```
Wouldn't something like that be possible with uboot ?

humaidq, 2024-08-24:
The version of u-boot should be the same regardless of if you are on 24.05 or unstable. You can see it pinned here and similarly in my fork:
https://github.com/NixOS/nixos-hardware/blob/master/pine64/star64/uboot.nix#L4

SpidFightFR, 2024-08-24:
> Hello! I finally got it built. @SpidFightFR check out these commits: [humaidq/u-boot@b801e99](https://github.com/humaidq/u-boot/commit/b801e9996c537f3225bb7acb2e27feeb9a98a198) [...]

I tried just changing every mention of `nixos-hardware` by your fork.
It works, on 24.05, however i'm not quite convinced by this solution, i'm not sure you're willing to sync your repo to upstream.

My best shot is what i got in mind.

In the meantime 24.05 works. :+1:

humaidq, 2024-08-24:
Yep, this is just a hacky fix. The board was collecting dust and I just wanted it to finally boot 😄
I am willing to make a proper fix and getting it upstreamed.

SpidFightFR, 2024-08-24:
> Yep, this is just a hacky fix. The board was collecting dust and I just wanted it to finally boot 😄 I am willing to make a proper fix and getting it upstreamed.

If you could that would be perfect, indeed, but i'm pretty sure i can hack my way around that issue, with or without your fix up-streamed (although if it were upstreamed, it would be great for people passing by after us.) :smile:

SpidFightFR, 2025-04-29:
Hey again:

To this day i get the following error in the unstable channel:
```
error: builder for '/nix/store/191vbvg9w09snn8jmq8hhlwlr6y42mkw-uboot-pine64_star64_defconfig-riscv64-unknown-linux-gnu-3.0.4.drv' failed with exit code 2;
       last 25 log lines:
       >    95 | static inline u8 readb(const volatile void __iomem *addr)
       >       |                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~
       > drivers/video/dw_hdmi.c:98:43: error: passing argument 1 of 'readl' makes pointer from integer without a cast []
       >    98 |                 return readl(hdmi->ioaddr + (offset << 2));
       >       |                              ~~~~~~~~~~~~~^~~~~~~~~~~~~~~
       >       |                                           |
       >       |                                           ulong {aka long unsigned int}
       > ./arch/riscv/include/asm/io.h:113:54: note: expected 'const volatile void *' but argument is of type 'ulong' {aka 'long unsigned int'}
       >   113 | static inline u32 readl(const volatile void __iomem *addr)
       >       |                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~
       >   CC      cmd/unzip.o
       >   AR      drivers/video/starfive/built-in.o
       > make[2]: *** [scripts/Makefile.build:253: drivers/video/dw_hdmi.o] Error 1
       > make[2]: *** Waiting for unfinished jobs....
       >   CC      cmd/lzmadec.o
       >   CC      cmd/fs_uuid.o
       >   CC      cmd/ximg.o
       >   CC      cmd/gpt.o
       >   CC      cmd/nvedit.o
       >   UPD     cmd/config_data_gz.h
       >   CC      cmd/config.o
       >   AR      common/built-in.o
       > make[1]: *** [scripts/Makefile.build:394: drivers/video] Error 2
       > make: *** [Makefile:1817: drivers] Error 2
       >   AR      cmd/built-in.o
       For full logs, run 'nix log /nix/store/191vbvg9w09snn8jmq8hhlwlr6y42mkw-uboot-pine64_star64_defconfig-riscv64-unknown-linux-gnu-3.0.4.drv'.
```

I tried to ask - without success - in the nixos-riscv matrix, without success... I sincerely hope to find help here... 🙏