NixOS/nixos-hardware #1052: Unexpected / new "error: value is a path while a set was expected" when importing "nixos-hardware.nixosModules.asus-zephyrus-ga402x.amdgpu" or "nixos-hardware.nixosModules.asus-zephyrus-ga402x.nvidia"

mexisme, 2024-07-21:


mexisme, 2024-07-21:
// @NiklasGollenstede

mexisme, 2024-07-21:
I suspect it's not useful, but the full error message (for me) looks like:
```
error:
       … while calling the 'seq' builtin

         at /nix/store/qszplw617r895nbcprgyj139c9a3r0xs-source/lib/modules.nix:322:18:

          321|         options = checked options;
          322|         config = checked (removeAttrs config [ "_module" ]);
             |                  ^
          323|         _module = checked (config._module);

       … while evaluating a branch condition

         at /nix/store/qszplw617r895nbcprgyj139c9a3r0xs-source/lib/modules.nix:261:9:

          260|       checkUnmatched =
          261|         if config._module.check && config._module.freeformType == null && merged.unmatchedDefns != [] then
             |         ^
          262|           let

       … in the left operand of the AND (&&) operator

         at /nix/store/qszplw617r895nbcprgyj139c9a3r0xs-source/lib/modules.nix:261:72:

          260|       checkUnmatched =
          261|         if config._module.check && config._module.freeformType == null && merged.unmatchedDefns != [] then
             |                                                                        ^
          262|           let

       … in the left operand of the AND (&&) operator

         at /nix/store/qszplw617r895nbcprgyj139c9a3r0xs-source/lib/modules.nix:261:33:

          260|       checkUnmatched =
          261|         if config._module.check && config._module.freeformType == null && merged.unmatchedDefns != [] then
             |                                 ^
          262|           let

       … while evaluating a branch condition

         at /nix/store/qszplw617r895nbcprgyj139c9a3r0xs-source/lib/modules.nix:254:12:

          253|
          254|         in if declaredConfig._module.freeformType == null then declaredConfig
             |            ^
          255|           # Because all definitions that had an associated option ended in

       … from call site

         at /nix/store/qszplw617r895nbcprgyj139c9a3r0xs-source/lib/modules.nix:242:28:

          241|           # For definitions that have an associated option
          242|           declaredConfig = mapAttrsRecursiveCond (v: ! isOption v) (_: v: v.value) options;
             |                            ^
          243|

       … while calling 'mapAttrsRecursiveCond'

         at /nix/store/qszplw617r895nbcprgyj139c9a3r0xs-source/lib/attrsets.nix:1201:5:

         1200|     f:
         1201|     set:
             |     ^
         1202|     let

       … from call site

         at /nix/store/qszplw617r895nbcprgyj139c9a3r0xs-source/lib/modules.nix:234:33:

[... 90 lines cut ...]
          335|       # Like unifyModuleSyntax, but also imports paths and calls functions if necessary
          336|       loadModule = args: fallbackFile: fallbackKey: m:
             |                                                     ^
          337|         if isFunction m then

       … from call site

         at /nix/store/qszplw617r895nbcprgyj139c9a3r0xs-source/lib/modules.nix:337:12:

          336|       loadModule = args: fallbackFile: fallbackKey: m:
          337|         if isFunction m then
             |            ^
          338|           unifyModuleSyntax fallbackFile fallbackKey (applyModuleArgs fallbackKey m args)

       … while calling 'isFunction'

         at /nix/store/qszplw617r895nbcprgyj139c9a3r0xs-source/lib/trivial.nix:929:16:

          928|   */
          929|   isFunction = f: builtins.isFunction f ||
             |                ^
          930|     (f ? __functor && isFunction (f.__functor f));

       error: value is a path while a set was expected
```

nipsy, 2024-07-24:
Just ran into this last night when defining nixos-hardware.nixosModules.lenovo-yoga-7-14ARH7.amdgpu in my module list for my laptop's nixosSystem.  Since I had other NixOS systems building without any errors and the inclusion of this hardware flake was pretty much the only difference for that system, I chanced upon the correct solution by commenting out that line in my module list and then found this bug report after the rebuild succeeded that time.

The error output certainly didn't help to narrow it down initially.  The initial error was after the first failure where the extra eval-cache option was necessary now to attempt a rebuild:

```
# nixos-rebuild switch --show-trace --flake .#
error: cached failure of attribute 'nixosConfigurations.ginaz.config'

# nixos-rebuild switch --option eval-cache false --show-trace --flake .#
error:
       … while calling the 'seq' builtin

         at /nix/store/afawjm6nv8m5aagkcqxlxgrr5qpyh6cm-source/lib/modules.nix:334:18:

          333|         options = checked options;
          334|         config = checked (removeAttrs config [ "_module" ]);
             |                  ^
          335|         _module = checked (config._module);

       … while evaluating a branch condition

         at /nix/store/afawjm6nv8m5aagkcqxlxgrr5qpyh6cm-source/lib/modules.nix:273:9:

          272|       checkUnmatched =
          273|         if config._module.check && config._module.freeformType == null && merged.unmatchedDefns != [] then
             |         ^
          274|           let

       … in the left operand of the AND (&&) operator

         at /nix/store/afawjm6nv8m5aagkcqxlxgrr5qpyh6cm-source/lib/modules.nix:273:72:

          272|       checkUnmatched =
          273|         if config._module.check && config._module.freeformType == null && merged.unmatchedDefns != [] then
             |                                                                        ^
          274|           let

       … in the left operand of the AND (&&) operator

         at /nix/store/afawjm6nv8m5aagkcqxlxgrr5qpyh6cm-source/lib/modules.nix:273:33:

          272|       checkUnmatched =
          273|         if config._module.check && config._module.freeformType == null && merged.unmatchedDefns != [] then
             |                                 ^
          274|           let

       … while evaluating a branch condition

         at /nix/store/afawjm6nv8m5aagkcqxlxgrr5qpyh6cm-source/lib/modules.nix:266:12:

          265|
          266|         in if declaredConfig._module.freeformType == null then declaredConfig
             |            ^
          267|           # Because all definitions that had an associated option ended in

       … from call site

         at /nix/store/afawjm6nv8m5aagkcqxlxgrr5qpyh6cm-source/lib/modules.nix:254:28:

          253|           # For definitions that have an associated option
          254|           declaredConfig = mapAttrsRecursiveCond (v: ! isOption v) (_: v: v.value) options;
             |                            ^
          255|

       … while calling 'mapAttrsRecursiveCond'

         at /nix/store/afawjm6nv8m5aagkcqxlxgrr5qpyh6cm-source/lib/attrsets.nix:1201:5:

         1200|     f:
         1201|     set:
[... 99 lines cut ...]
             |                                                     ^
          349|         if isFunction m then

       … from call site

         at /nix/store/afawjm6nv8m5aagkcqxlxgrr5qpyh6cm-source/lib/modules.nix:349:12:

          348|       loadModule = args: fallbackFile: fallbackKey: m:
          349|         if isFunction m then
             |            ^
          350|           unifyModuleSyntax fallbackFile fallbackKey (applyModuleArgs fallbackKey m args)

       … while calling 'isFunction'

         at /nix/store/afawjm6nv8m5aagkcqxlxgrr5qpyh6cm-source/lib/trivial.nix:989:16:

          988|   */
          989|   isFunction = f: builtins.isFunction f ||
             |                ^
          990|     (f ? __functor && isFunction (f.__functor f));

       error: value is a path while a set was expected
```

This is one of my few complaints with NixOS is that the error messages are next to useless sometimes in figuring out where the problem actually lies.  Although I'm open to suggestions if a better command line invocation would have somehow sussed this out as a problem in this particular flake.

mexisme, 2024-07-25:
> Just ran into this last night when defining nixos-hardware.nixosModules.lenovo-yoga-7-14ARH7.amdgpu in my module list for my laptop's nixosSystem. Since I had other NixOS systems building without any errors and the inclusion of this hardware flake was pretty much the only difference for that system, I chanced upon the correct solution by commenting out that line in my module list and then found this bug report after the rebuild succeeded that time.
> 
> The error output certainly didn't help to narrow it down initially. The initial error was after the first failure where the extra eval-cache option was necessary now to attempt a rebuild:
> 
> ```
> # nixos-rebuild switch --show-trace --flake .#
> error: cached failure of attribute 'nixosConfigurations.ginaz.config'
> 
> # nixos-rebuild switch --option eval-cache false --show-trace --flake .#
> error:
>        … while calling the 'seq' builtin
> 
>          at /nix/store/afawjm6nv8m5aagkcqxlxgrr5qpyh6cm-source/lib/modules.nix:334:18:
> 
>           333|         options = checked options;
>           334|         config = checked (removeAttrs config [ "_module" ]);
>              |                  ^
>           335|         _module = checked (config._module);
> 
>        … while evaluating a branch condition
> 
>          at /nix/store/afawjm6nv8m5aagkcqxlxgrr5qpyh6cm-source/lib/modules.nix:273:9:
> 
>           272|       checkUnmatched =
>           273|         if config._module.check && config._module.freeformType == null && merged.unmatchedDefns != [] then
>              |         ^
>           274|           let
> 
>        … in the left operand of the AND (&&) operator
> 
>          at /nix/store/afawjm6nv8m5aagkcqxlxgrr5qpyh6cm-source/lib/modules.nix:273:72:
> 
>           272|       checkUnmatched =
>           273|         if config._module.check && config._module.freeformType == null && merged.unmatchedDefns != [] then
>              |                                                                        ^
>           274|           let
> 
>        … in the left operand of the AND (&&) operator
> 
>          at /nix/store/afawjm6nv8m5aagkcqxlxgrr5qpyh6cm-source/lib/modules.nix:273:33:
> 
>           272|       checkUnmatched =
>           273|         if config._module.check && config._module.freeformType == null && merged.unmatchedDefns != [] then
>              |                                 ^
>           274|           let
> 
>        … while evaluating a branch condition
> 
>          at /nix/store/afawjm6nv8m5aagkcqxlxgrr5qpyh6cm-source/lib/modules.nix:266:12:
> 
>           265|
>           266|         in if declaredConfig._module.freeformType == null then declaredConfig
>              |            ^
>           267|           # Because all definitions that had an associated option ended in
> 
>        … from call site
> 
>          at /nix/store/afawjm6nv8m5aagkcqxlxgrr5qpyh6cm-source/lib/modules.nix:254:28:
> 
>           253|           # For definitions that have an associated option
>           254|           declaredConfig = mapAttrsRecursiveCond (v: ! isOption v) (_: v: v.value) options;
>              |                            ^
>           255|
> 
>        … while calling 'mapAttrsRecursiveCond'
> 
>          at /nix/store/afawjm6nv8m5aagkcqxlxgrr5qpyh6cm-source/lib/attrsets.nix:1201:5:
> 
>          1200|     f:
>          1201|     set:
[... 109 lines cut ...]
>           350|           unifyModuleSyntax fallbackFile fallbackKey (applyModuleArgs fallbackKey m args)
> 
>        … while calling 'isFunction'
> 
>          at /nix/store/afawjm6nv8m5aagkcqxlxgrr5qpyh6cm-source/lib/trivial.nix:989:16:
> 
>           988|   */
>           989|   isFunction = f: builtins.isFunction f ||
>              |                ^
>           990|     (f ? __functor && isFunction (f.__functor f));
> 
>        error: value is a path while a set was expected
> ```
> 
> This is one of my few complaints with NixOS is that the error messages are next to useless sometimes in figuring out where the problem actually lies. Although I'm open to suggestions if a better command line incantation would have somehow sussed this out as a problem in this particular flake.

TL;DR:
I think the only places where the (incorrect) attr-set style was used was in `asus-zephyrus-ga402x` and `lenovo-yoga-7-14ARH7` — both of which were my fault, I'm afraid.

---

IMHO, in this particular situation, it's more that the inner complexities of modules can make it harder to diagnose higher-level errors like this.

As I understand it, part of the reason for @NiklasGollenstede's change was to try to make it easier to diagnose module errors;  unfortunately this particular one is much less obvious because of when it fails.
I only realised it because of a bit of an epiphany!

PR #1055 (2024-07-25): Create lenovo-yoga-7-14ARH7-amdgpu and lenovo-yoga-7-14ARH7-nvidia entries
###### Description of changes

This PR:
- Replaces `lenovo-yoga-7-14ARH7.amdgpu` with a `lenovo-yoga-7-14ARH7-amdgpu` entry.
- Replaces `lenovo-yoga-7-14ARH7.nvidia` with a `lenovo-yoga-7-14ARH7-nvidia` entry.
- Makes `lenovo-yoga-7-14ARH7` throw a deprecation warning.

FIXES: #1052 

Background:
The `lenovo-yoga-7-14ARH7` provides an attr-set with `amdgpu` and `nvidia` entries, to allow users to choose whether to enable only the AMD-GPU driver, or also enable the NVidia driver with (by default) Prime enabled.

However, [... cut]
PR #1053 (2024-07-25): Create asus-zephyrus-ga402x-amdgpu and asus-zephyrus-ga402x-nvidia entries
###### Description of changes

This PR:
- Replaces `asus-zephyrus-ga402x.amdgpu` with a `asus-zephyrus-ga402x-amdgpu` entry.
- Replaces `asus-zephyrus-ga402x.nvidia` with a `asus-zephyrus-ga402x-nvidia` entry.
- Makes `asus-zephyrus-ga402x` throw a deprecation warning.

FIXES: #1052 

Background:
The `asus-zephyrus-ga402x` provides an attr-set with `amdgpu` and `nvidia` entries, to allow users to choose whether to enable only the AMD-GPU driver, or also enable the NVidia driver with (by default) Prime enabled.

However, [... cut]
closed 2024-07-25 by mergify[bot]