NixOS/nixos-hardware #1092: error with hardware.nvidia.open (dell xps 15 7590)

eg-ayoub, 2024-08-29:
on my xps laptop using this flake: 
```nix
      # 2 - mouse : (laptop) Tiling WM + dev tools
      mouse = nixpkgs.lib.nixosSystem {
        specialArgs = { inherit inputs system; };

        modules = [
          ./hosts/mouse/configuration.nix
          nixos-hardware.nixosModules.dell-xps-15-7590-nvidia
        ];
      };
```
in configuration.nix I have:
`hardware.nvidia.prime.offload.enable = true;`

nixos rebuild fails with 
```sh
$ sudo nixos-rebuild switch --flake .#mouse --option eval-cache false
warning: Git tree '/home/ayoub/repos/nix-files' is dirty
building the system configuration...
warning: Git tree '/home/ayoub/repos/nix-files' is dirty
error:
       … while calling the 'head' builtin

         at /nix/store/887hpp8a2i99n9jjwcvz6qkhhhqsvzkg-source/lib/attrsets.nix:1575:11:

         1574|         || pred here (elemAt values 1) (head values) then
         1575|           head values
             |           ^
         1576|         else

       … while evaluating the attribute 'value'

         at /nix/store/887hpp8a2i99n9jjwcvz6qkhhhqsvzkg-source/lib/modules.nix:821:9:

          820|     in warnDeprecation opt //
          821|       { value = addErrorContext "while evaluating the option `${showOption loc}':" value;
             |         ^
          822|         inherit (res.defsFinal') highestPrio;

       (stack trace truncated; use '--show-trace' to show the full trace)

       error: The option `hardware.nvidia.open' is used but not defined.
```
I assumed this option is set automatically by https://github.com/NixOS/nixpkgs/blob/nixos-unstable/pkgs/os-specific/linux/nvidia-x11/default.nix 
am I wrong ? 
thanks for your help in advance :)

BCNelson, 2024-08-29:
It was an intentional breaking change https://github.com/NixOS/nixpkgs/pull/337289#issuecomment-2314851863

eg-ayoub, 2024-08-29:
@BCNelson thanks! 
I'll make a PR by tomorrow. 
If 560 works on my machine would it be acceptable to set  `hardware.nvidia.open = true;` for xps15-7590 ?

BCNelson, 2024-08-29:
I'm not sure about the specifics. But I do know that it defaulted to false before

closed 2025-01-31 by eg-ayoub