NixOS/nixos-hardware #1095: Package request: linux-g14 kernel package

sjhaleprogrammer, 2024-09-01:
**Project description**

linux-g14 is a lot of kernel patches made for asus laptops

**Metadata**

* homepage URL: https://gitlab.com/dragonn/linux-g14
* source URL: https://aur.archlinux.org/pkgbase/linux-g14

---

please help my asus laptop work better on nixos

435vic, 2024-09-05:
I also use an asus laptop and this would help so much

ottersome, 2024-09-07:
+1. I am willing to help with a PR if necessary. Though my experience is limited.

flukejones, 2024-09-08:
It is probably better to follow my fedora packaging which I've started using a single file patch for: https://gitlab.com/asus-linux/fedora-kernel/-/blob/rog-6.10/asus-patch-series.patch?ref_type=heads

It will always be kept up to date with 6.10.n, of the version in the `kernel.spec`. As of today there was a new 6.10 point release - I'm not up to date with that and won't be until fedora pulls it. There is an AMD PMF patch that was merged to upstream.

sjhaleprogrammer, 2024-09-08:
This is really nice thanks for your Asus support

435vic, 2024-09-09:
That's excellent! One quick option to integrate them would be to use `boot.kernelPatches` and import the patch file directly. However, that means on rebuild the entire kernel will be compiled on the computer, which is generally undesirable. I think for it to be cached and compiled automatically with Hyrda it would need to be part of `nixpkgs`, no? Which would mean creating a custom kernel package... would appreciate any help on the matter to see some options

500InternalError, 2024-09-12:
Created a PR for linux-g14: https://github.com/NixOS/nixpkgs/pull/341338

sjhaleprogrammer, 2024-09-17:
https://github.com/sjhaleprogrammer/nixos/blob/master/kernel.nix 
took a while to build but if anyone wants the patches

thrombe, 2024-09-18:
I have been daily driving linux-g14 from arch/asus-linux from fedora on my asus ga402xu for about a year now. it works great for my usecase. tho note that i don't know much about kernels.

here is the [arch linux-g14](https://github.com/thrombe/dotfiles-promax/blob/a24c096fb213f81512c89ac7f1947ac0cce5fe7c/configma/nixos/etc/nixos/ga402xu/configuration.nix#L234) config.
and here is the [fedora asus-linux](https://github.com/thrombe/dotfiles-promax/blob/a24c096fb213f81512c89ac7f1947ac0cce5fe7c/configma/nixos/etc/nixos/ga402xu/configuration.nix#L158) config.

caniko, 2025-04-24:
Hey everyone, I have been working on porting the kernel to nix, [here](https://gist.github.com/caniko/e452a3610a7dcc49cc99b5490863b8d1). I tried automating the installation of the patches by every patch on the repo, but it didn't work for some reason. `rog-control-center` is complaining that the patch is absent.

Will create a PR soon when I fix this. Feel free to give feedback in the meantime.