NixOS/nixos-hardware #1112: Asus GU605MY - Laptop does not wake from sleep

relief-melone, 2024-09-07:
Most of the settings for the Asus Zephyrus G16 GU605MY do work pretty fine as of now. However I am still trying to figure out a way to make the wake up after suspend work.
At the moment the display as well as keyboard lighting will turn on the display will remain black. I do not see any entries in the journal that are pointing me on where the issues lie. 
Working on it but also very happy if somebody else has an idea where to look or how to fix

relief-melone, 2024-09-08:
Ok I was able to get some info from the kernel logs. However I cannot make much of it yet

```txt
Sep 08 17:09:56 rm-az-2024 kernel:  </TASK>
Sep 08 17:09:56 rm-az-2024 kernel:  ret_from_fork_asm+0x1a/0x30
Sep 08 17:09:56 rm-az-2024 kernel:  ? __pfx_kthread+0x10/0x10
Sep 08 17:09:56 rm-az-2024 kernel: ---[ end trace 0000000000000000 ]---
Sep 08 17:09:56 rm-az-2024 kernel:  </TASK>
Sep 08 17:09:56 rm-az-2024 kernel:  ret_from_fork_asm+0x1a/0x30
Sep 08 17:09:56 rm-az-2024 kernel:  ? __pfx_kthread+0x10/0x10
Sep 08 17:09:56 rm-az-2024 kernel: ---[ end trace 0000000000000000 ]---
Sep 08 17:09:56 rm-az-2024 kernel:  </TASK>
Sep 08 17:09:56 rm-az-2024 kernel:  ret_from_fork_asm+0x1a/0x30
Sep 08 17:09:56 rm-az-2024 kernel:  ? __pfx_kthread+0x10/0x10
Sep 08 17:09:56 rm-az-2024 kernel: ---[ end trace 0000000000000000 ]---
Sep 08 17:09:56 rm-az-2024 kernel:  </TASK>
Sep 08 17:09:56 rm-az-2024 kernel:  ret_from_fork_asm+0x1a/0x30
Sep 08 17:09:56 rm-az-2024 kernel:  ? __pfx_kthread+0x10/0x10
Sep 08 17:09:56 rm-az-2024 kernel: ---[ end trace 0000000000000000 ]---
Sep 08 17:09:56 rm-az-2024 kernel:  </TASK>
Sep 08 17:09:56 rm-az-2024 kernel:  ret_from_fork_asm+0x1a/0x30
Sep 08 17:09:56 rm-az-2024 kernel:  ? __pfx_kthread+0x10/0x10
Sep 08 17:09:56 rm-az-2024 kernel: ---[ end trace 0000000000000000 ]---
Sep 08 17:09:56 rm-az-2024 kernel:  </TASK>
Sep 08 17:09:56 rm-az-2024 kernel:  ret_from_fork_asm+0x1a/0x30
Sep 08 17:09:56 rm-az-2024 kernel:  ? __pfx_kthread+0x10/0x10
Sep 08 17:09:56 rm-az-2024 kernel:  ret_from_fork+0x31/0x50
Sep 08 17:09:56 rm-az-2024 kernel:  ? __pfx_kthread+0x10/0x10
Sep 08 17:09:56 rm-az-2024 kernel:  kthread+0xcd/0x100
Sep 08 17:09:56 rm-az-2024 kernel:  ? __pfx__main_loop+0x10/0x10 [nvidia]
Sep 08 17:09:56 rm-az-2024 kernel:  _main_loop+0x8f/0x160 [nvidia]
Sep 08 17:09:56 rm-az-2024 kernel:  os_execute_work_item+0x5e/0x90 [nvidia]
Sep 08 17:09:56 rm-az-2024 kernel:  rm_execute_work_item+0x110/0x170 [nvidia]
Sep 08 17:09:56 rm-az-2024 kernel:  _nv000723rm+0x35/0xf6 [nvidia]
Sep 08 17:09:56 rm-az-2024 kernel:  nv_revoke_gpu_mappings+0x67/0xc0 [nvidia]
Sep 08 17:09:56 rm-az-2024 kernel:  unmap_mapping_range+0x11b/0x140
Sep 08 17:09:56 rm-az-2024 kernel:  zap_page_range_single+0x122/0x1e0
Sep 08 17:09:56 rm-az-2024 kernel:  unmap_single_vma+0xb6/0xf0
Sep 08 17:09:56 rm-az-2024 kernel:  untrack_pfn+0x57/0x130
Sep 08 17:09:56 rm-az-2024 kernel:  follow_phys+0x4b/0x110
Sep 08 17:09:56 rm-az-2024 kernel:  ? follow_pte+0x21c/0x230
Sep 08 17:09:56 rm-az-2024 kernel:  ? asm_exc_invalid_op+0x1a/0x20
Sep 08 17:09:56 rm-az-2024 kernel:  ? exc_invalid_op+0x17/0x80
Sep 08 17:09:56 rm-az-2024 kernel:  ? handle_bug+0x43/0x90
Sep 08 17:09:56 rm-az-2024 kernel:  ? report_bug+0x172/0x1a0
Sep 08 17:09:56 rm-az-2024 kernel:  ? follow_pte+0x21c/0x230
Sep 08 17:09:56 rm-az-2024 kernel:  ? __warn+0x80/0x120
Sep 08 17:09:56 rm-az-2024 kernel:  <TASK>
Sep 08 17:09:56 rm-az-2024 kernel: Call Trace:
Sep 08 17:09:56 rm-az-2024 kernel: PKRU: 55555554
Sep 08 17:09:56 rm-az-2024 kernel: DR3: 0000000000000000 DR6: 00000000ffff07f0 DR7: 0000000000000400
Sep 08 17:09:56 rm-az-2024 kernel: DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
Sep 08 17:09:56 rm-az-2024 kernel: CR2: 000000001954dff8 CR3: 0000000607820003 CR4: 0000000000f70ef0
Sep 08 17:09:56 rm-az-2024 kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Sep 08 17:09:56 rm-az-2024 kernel: FS:  0000000000000000(0000) GS:ffff98491ee80000(0000) knlGS:0000000000000000
Sep 08 17:09:56 rm-az-2024 kernel: R13: ffff9df204177bb0 R14: ffff9841d69f8580 R15: 0000000000000000
Sep 08 17:09:56 rm-az-2024 kernel: R10: 0000000000000000 R11: 0000000000000000 R12: ffff9df204177bb8
Sep 08 17:09:56 rm-az-2024 kernel: RBP: ffff9df204177bf8 R08: ffff9df204177d50 R09: 0000000000000000
Sep 08 17:09:56 rm-az-2024 kernel: RDX: ffff9df204177bb0 RSI: 00007f2d5b311000 RDI: ffff9841e8da1ce0
Sep 08 17:09:56 rm-az-2024 kernel: RAX: 0000000000000000 RBX: 00007f2d5b311000 RCX: ffff9df204177bb8
Sep 08 17:09:56 rm-az-2024 kernel: RSP: 0018:ffff9df204177b78 EFLAGS: 00010246
Sep 08 17:09:56 rm-az-2024 kernel: Code: 24 e8 28 df 95 00 e8 43 1b e2 ff bd ea ff ff ff 5b 89 e8 5d 41 5c 41 5d 41 5e 41 5f 31 d2 31 c9 31 f6 31 ff c3 cc cc cc cc 90 <0f> 0b 90 e9 0c fe ff ff 66 66 2e 0f 1f 84 00 00 00 00 00 90 90 90
Sep 08 17:09:56 rm-az-2024 kernel: RIP: 0010:follow_pte+0x21c/0x230
Sep 08 17:09:56 rm-az-2024 kernel: Hardware name: ASUSTeK COMPUTER INC. ROG Zephyrus G16 GU605MY_GU605MY/GU605MY, BIOS GU605MY.319 04/22/2024
Sep 08 17:09:56 rm-az-2024 kernel: CPU: 1 PID: 1657 Comm: nv_queue Tainted: P        W  O       6.10.8 #1-NixOS
Sep 08 17:09:56 rm-az-2024 kernel:  efivarfs tpm libaescfb ecdh_generic ecc rng_core dmi_sysfs ip_tables x_tables autofs4 ext4 crc32c_generic crc16 mbcache jbd2 hid_generic usbhid hid rtsx_pci_sdmmc mmc_core nvme thunderbolt nvme_core atkbd xhci_pci libps2 nvme_auth xhci_pci_renesas t10_pi vivaldi_fmap xhci_hcd c>
Sep 08 17:09:56 rm-az-2024 kernel:  processor_thermal_rapl ucsi_acpi snd_soc_cs35l56_shared cfg80211 intel_th_gth snd_pcm snd_hda_cs_dsp_ctls drm_gpuvm intel_rapl_common typec_ucsi drm_exec nvidia_wmi_ec_backlight loop snd_soc_cs_amp_lib gpu_sched intel_th_pci processor_thermal_wt_req i2c_i801 mei_me tiny_power_b>
Sep 08 17:09:56 rm-az-2024 kernel:  ac97_bus snd_pcm_dmaengine nls_iso8859_1 nls_cp437 vfat fat iwlmvm joydev mousedev mac80211 hid_multitouch snd_hda_codec_realtek intel_uncore_frequency intel_uncore_frequency_common snd_hda_codec_generic x86_p [... cut ...]
Sep 08 17:09:56 rm-az-2024 kernel: Modules linked in: qrtr snd_seq_dummy rfcomm snd_hrtimer snd_seq snd_seq_device xt_MASQUERADE nf_conntrack_netlink xfrm_user xfrm_algo xt_addrtype overlay nls_utf8 cifs cifs_arc4 nls_ucs2_utils cifs_md4 dns_resolver netfs ccm xt_CHECKSUM ipt_REJECT nf_reject_ipv4 nft_chain_nat a>
Sep 08 17:09:56 rm-az-2024 kernel: WARNING: CPU: 1 PID: 1657 at include/linux/rwsem.h:80 follow_pte+0x21c/0x230
Sep 08 17:09:56 rm-az-2024 kernel: ------------[ cut here ]------------

```

Mic92, 2024-09-09:
The stack trace contains some issue with the nvidia driver?