NixOS/nixos-hardware #1114: Why acpi_call?

ghtau, 2024-09-09:
[TLP's documentation](https://linrunner.de/tlp/faq/battery.html#which-external-kernel-module-do-i-need-for-my-thinkpad) states:

> As of version 5.17, the Linux kernel in combination with TLP 1.5 or later offers **full battery care support** (i.e. charge thresholds and recalibration) for ThinkPads from model year 2011 onwards.

...and later even calls `acpi_call` "obsolete".

So I'm curious why it's used and why the module states that it "makes tlp work for newer thinkpads"?  The `natacpi` + `thinkpad_acpi` combo works for me on older Thinkpads (e.g. X230) and my more recent ones (e.g. T470s).

Mic92, 2024-09-09:
It's probably fair to remove it at this point.

jrasanen, 2024-12-02:
I noticed I'm unable to build NixOS with the latest ACPI call, [here's a thread](https://discourse.nixos.org/t/hash-mismatch-issue/56870) I opened on Discourse, with some findings from the community members (thanks Simon!)

closed 2024-12-02 by Mic92

nixos-discourse, 2024-12-02:
This issue has been mentioned on **NixOS Discourse**. There might be relevant details there:

https://discourse.nixos.org/t/hash-mismatch-issue/56870/6