NixOS/nixos-hardware #1121: Surface Pro 9 won't boot after a seemingly successful installation

kubaspek, 2024-09-10:
I'm building a flake which integrates devices to it's config, but I've encountered issues with this Microsoft Surface Pro 9. I've partitioned the ssd for dualboot between Win11 and NixOS, compiled a NixOS compatible kernel, ran `nixos-install` from an installation thumbdrive and everything seemed to work, but when rebooting into NixOS, a series of scary error messages popped up.

Here is a photo of the faulty boot:
[image]

My config:
https://github.com/kubaspek/nixos/

> In hosts/desktops/surface/ you can find the host specific config for the Surface and `nixos-hardware` imports are in `flake.nix` on lines 18 and 123

I suspect the problem is either somewhere here or perhaps in nixos-hardware itself, but I really don't know:

```
# The system config for Surface

## Sources

{ config, lib, pkgs, inputs, ... }: {

### BOOT
    
    boot.supportedFilesystems = ["ntfs"];
    boot.loader.efi.canTouchEfiVariables = true;
    boot.loader.efi.efiSysMountPoint = "/boot";
    #boot.blacklistedKernelModules = ["ov13858"];

boot.loader.grub = {
      enable = true;
      efiSupport = true;
      #zfsSupport = true;
      efiInstallAsRemovable = false;
      memtest86.enable = true;
      device = "nodev";
      useOSProber = true;
      /*mirroredBoots = [
        {
          devices = ["nodev"];
          path = "/boot";
        }
      ];*/
  };
```

```
# Disk and FS config
{config, pkgs, ...}: {
### ZFS CONFIG
    #boot.zfs.requestEncryptionCredentials = ["nvme"];

  #boot.kernelPackages = config.boot.zfs.package.latestCompatibleLinuxPackages;
    #boot.zfs.package = pkgs.zfs_unstable;
/*
services.zfs = {
      trim.enable = true;
      autoScrub.enable = true;
    };
*/

### MOUNTS
  fileSystems."/" = {
    device = "/dev/disk/by-uuid/b69e469e-3d00-4338-a415-c5abd8a939e7";
    fsType = "ext4";
  };

  fileSystems."/boot" = {
    device = "/dev/disk/by-uuid/DA67-00D0";
    fsType = "vfat";
    #options = ["noexec" "nodev"];
  };
```

I've tried blacklisting the "vo13858" kernel module, because the boot sequence had a problem with it, but I got another error after doing that so the problem is probably somewhere else anyways

Feel free to ask for any additional info
Thanks

xieve, 2024-09-22:
Having the same issue on a Surface Pro 9, this seems to be an issue with linux-surface: linux-surface/linux-surface#1516
Here's the workaround mentioned in that issue for nixOS:
```nix
  boot.blacklistedKernelModules = [
    "intel-ipu6"
    "intel-ipu6-isys"
  ];
```
Worked for me on 6.10.5/`b493dfd4a8cf9552932179e56ff3b5819a9b8381`.
I assume it wouldn't hurt if you posted your failed boot logs to that issue as well :)

kubaspek, 2024-09-23:
This did work, but only for the old version of nixos-hardware, which I downloaded a few weeks ago
I got into the system, `sudo nix flake update` did update my flake, but `sudo nixos-rebuild boot` tried to recompile the kernel and failed at it. This is a problem, as the everything, including kde, system packages and other imports are tied to this rebuild. If `nixos-hardware` doesn't work, nothing does..

Should I try some sort of a workaround or would it be better to somehow use the old inputs for nixos-hardware so that I don't need to recompile the kernel each time someone makes a comment in `nixos-hardware`?

Or should I just wait until `nixos-hardware` starts working again? 
btw don't need cameras to work, I don't mind

Thank you for the support

xieve, 2024-09-24:
> This did work, but only for the old version of nixos-hardware, which I downloaded a few weeks ago I got into the system, `sudo nix flake update` did update my flake, but `sudo nixos-rebuild boot` [...]

Very odd, I compiled the latest commit a few days ago, and nixos-hardware hasn't received any surface-related updates in a while as far as I can see. 

This sounds like a new problem? If the kernel won't compile, that's a different issue than a segfault while booting, no?

> Should I try some sort of a workaround or would it be better to somehow use the old inputs for nixos-hardware so that I don't need to recompile the kernel each time someone makes a comment in nixos-hardware?

If you're using flakes, you can specify a fixed commit for an input like this:
```nix
inputs = [
  nixos-hardware.url = "github:nixos/nixos-hardware?rev=b493dfd4a8cf9552932179e56ff3b5819a9b8381 # or any other commit hash, obviously
]
```
and you can get the old hash from the `flake.lock` version before you ran `flake update`(I hope your config is version controlled).

Afaik, you don't have to recompile linux each time `nixos-hardware` receives a commit, just like you don't have to re-download every package every time `nixpkgs` is updated. Nix is smart like that :D You do have to recompile when the kernel version is bumped, obviously. But in this case it's definitely a good approach to "lock" the version input like above.

kubaspek, 2024-09-24:
Thank you!
I'll try this fixed build and maybe update `nixos-hardware` every now and then
It's quite a pain recompiling the kernel only to find out it won't boot
Other than that it's perfect for me

closed 2024-09-24 by kubaspek