NixOS/nixos-hardware #1124: Minisforum UH125 Pro

alejandrosame, 2024-09-13:
I've tried installing NixOS in this mini PC to experiment with the NPU of its Intel Core Ultra chip.

The machine doesn't boot, keeping a black screen after "EFI stub: Loaded initrd from LINUX_EFI_INITRD_MEDIA_GUID device path". I've tried with Linux 6.10, 6.6 and 6.8, all give a black screen. 

Installing with Linux 5.15 gives some output with a boot error, but as this machine doesn't have a serial output port, so I don't know how to retrieve the full stack trace.  

Ubuntu 24.04 with kernel 6.8 boots on this machines, so the machine does have Linux support somewhere.

Leaving this report here, as I'd like to use NixOS with it, in case someone finds how to make it boot.

mastoca, 2024-09-22:
I too ran into this issue (i have x7i) and reported it here https://discourse.nixos.org/t/nix-iso-unable-to-boot-in-uefi-mode-but-other-distros-can/16473/25

I'm currently using SUSE on it as it boots without issue....as do a bunch of other distros.

NOTE: that there are some others I tried that also fail so it isn't just nixos...so they've messed up in a similar way?

JohnMolotov, 2024-09-23:
I don't think it's that they've "messed up", but rather simply that it's newish hardware so you'll need the latest kernel and driver to get it working, which aren't typically the default for stability reasons.

mastoca, 2024-09-23:
@JohnMolotov The text install hangs as well, which is weird. I've tried all the suggestions in the discourse posting I shared above. I'm happy to try different things...

JohnMolotov, 2024-09-23:
@mastoca No clue what the issue is then sorry.