NixOS/nixos-hardware #1136: Request Asus ROG Zephyrus GA403

AchmadFathoni, 2024-09-21:
This laptop is not configured yet. Should I start with GA402 or GA402x? How to make the keyboard backlight adjustable by keyboard shortcut? What do you think? @Lyndeno

hatosu, 2024-09-22:
I would also like this a HUGE amount. I myself also have a GA403 model and am disappointment to see there being no option for it here :(

AchmadFathoni, 2024-09-23:
Hi @hatosu , have you tried [asus-linux](https://asus-linux.org/guides/nixos/) with liurlnux 6.11? I'm using these, and the only think I complain is keyboard LED and slash LED that can't controlled.

Lyndeno, 2024-09-24:
> Should I start with GA402 or GA402x? 

Either or I think, probably 402x if the 403 has a dedicated gpu.

> How to make the keyboard backlight adjustable by keyboard shortcut?

Not sure on that one, the 402 and 402x have [this](https://github.com/NixOS/nixos-hardware/blob/d0cb432a9d28218df11cbd77d984a2a46caeb5ac/asus/zephyrus/ga402x/shared.nix#L57C10-L61C14) for the mic button.

sekasorto93, 2024-10-13:
Would also like to see some GA403 support!