NixOS/nixos-hardware #1142: Allow increasing `gpu_mem` for Raspberry PI

M-Gregoire, 2024-09-26:
Hello :wave: ,

After installing NixOS and using `nixos-hardware` to setup my Raspberry Pi 4, the allocated GPU memory is 76M.
```
# vcgencmd get_mem gpu
gpu=76M
```

I'd like to increase it to at least 128M. I believe the standard way would have been:
```
  boot.loader.raspberryPi = {
    enable = true;
    firmwareConfig = ''
      gpu_mem=128
    '';
  };
```

But I've read in [an other issue](https://github.com/NixOS/nixos-hardware/pull/780#issue-1993455970) that `firmwareConfig` is gonna be deprecated.

I'm not quite sure what it the proper way to increase the memory allocation for the GPU.

Any help would be appreciated :)

Mic92, 2024-09-27:
Not sure if this can be actually deprecated because I assume that the bootloader won't respect dts. At least this is my understanding. So `firmwareConfig` is probably still the way to go?

M-Gregoire, 2024-10-07:
Thank you for your answer.

I've tried the snippet above which produced the following error :
```
Failed assertions:
       - Only Raspberry Pi >= 3 supports aarch64.
```
(Coming from [here](https://github.com/NixOS/nixpkgs/blob/ecbc1ca8ffd6aea8372ad16be9ebbb39889e55b6/nixos/modules/system/boot/loader/raspberrypi/raspberrypi.nix#L142)) when the target is a Raspberry pi 4 with
` nixos-hardware.nixosModules.raspberry-pi-4` in it's flake.

So, upon further inspection, it seems that It might be because `firmwareConfig` requires `loader.raspberryPi` ([Here](https://github.com/NixOS/nixpkgs/blob/ecbc1ca8ffd6aea8372ad16be9ebbb39889e55b6/nixos/modules/system/boot/loader/raspberrypi/raspberrypi.nix#L6)) where `nixos-hardware` defaults to the generic one ([here](https://github.com/NixOS/nixos-hardware/blob/ecfcd787f373f43307d764762e139a7cdeb9c22b/raspberry-pi/4/default.nix#L35)).

Documentation for `loader.raspberryPi.enable` states:
```
Whether to create files with the system generations in /boot. /boot/old will hold files from old generations.

These options are deprecated, unsupported, and may not work like expected.
```

So it does feel like this might be not the right way to go with this, but again, I'm not quite sure how to change the `gpu_mem` otherwise.

Would you have any advice by any chance?

Mic92, 2024-10-08:
You may need to set the version to 4 as well to avoid the evaluation error that you see.