NixOS/nixos-hardware #1163: Does intel-media-driver supersede intel-vaapi-driver ?

matdibu, 2024-10-01:
Is there any usecase where the older intel-vaapi-driver is a better choice when compared with intel-media-driver?

SuperSandro2000, 2024-10-02:
If you have old hardware https://github.com/NixOS/nixpkgs/blob/21576aba43e604a95cc45c25248f39f3989ebe36/nixos/modules/hardware/graphics.nix#L77

matdibu, 2024-10-02:
got it, thanks!

matdibu, 2024-10-02:
and does the list order in `hardware.graphics.extraPackages` matter? I see that in https://github.com/NixOS/nixos-hardware/blob/11c43c830e533dad1be527ecce379fcf994fbbb5/common/gpu/intel/default.nix#L29 `intel-media-driver` is after `vaapiIntel`/`intel-vaapi-driver`

closed 2024-10-02 by matdibu

SuperSandro2000, 2024-10-02:
I don't think that matters. The files they contain are sorted differently anyway and the driver or maybe program then chooses a preference for one over the other.