NixOS/nixos-hardware #1166: Legion Pro 7 16IRX8H fails building

i-am-logger, 2024-10-06:
[image]

```bash
error: builder for '/nix/store/17cxk5fyi5wcfdg98wmkkwi246fg2jjw-lenovo-legion-module-0.0.12.drv' failed with exit 
code 2;
       last 25 log lines:
       > no configure script, doing nothing
       > Running phase: buildPhase
       > build flags: SHELL=/nix/store/izpf49b74i15pcr9708s3xdwyqs4jxwl-bash-5.2p32/bin/bash O=\$\(buildRoot\) CC=/nix/store/zznja5f8v3jafffyah1rk46vpfcn38dv-gcc-wrapper-13.3.0/bin/cc HOSTCC=/nix/store/zznja5f8v3jafffyah1rk46vpfcn38dv-gcc-wrapper-13.3.0/bin/cc HOSTLD=/nix/store/b74nxf0yn2dzha02mgdxyklaqjaijzqp-binutils-wrapper-2.42/bin/ld ARCH=x86_64 SHELL=bash KERNELVERSION=6.11.1 KSRC=/nix/store/k3j2lh3hfc7r7s63zjyryydqxyis9wdk-linux-6.11.1-dev/lib/modules/6.11.1/build INSTALLDIR=/nix/store/a90q523321wm9vgrpz0qy6a0arv8zsx5-lenovo-legion-module-0.0.12/lib/modules/6.11.1/kernel/drivers/platform/x86 MODDESTDIR=/nix/store/a90q523321wm9vgrpz0qy6a0arv8zsx5-lenovo-legion-module-0.0.12/lib/modules/6.11.1/kernel/drivers/platform/x86 DKMSDIR=/nix/store/a90q523321wm9vgrpz0qy6a0arv8zsx5-lenovo-legion-module-0.0.12/lib/modules/6.11.1/misc
       > make -C /nix/store/k3j2lh3hfc7r7s63zjyryydqxyis9wdk-linux-6.11.1-dev/lib/modules/6.11.1/build M=/build/source/kernel_module modules
       > make[1]: Entering directory '/nix/store/k3j2lh3hfc7r7s63zjyryydqxyis9wdk-linux-6.11.1-dev/lib/modules/6.11.1/build'
       >   CC [M]  /build/source/kernel_module/legion-laptop.o
       > /build/source/kernel_module/legion-laptop.c:1727:9: warning: no previous prototype for 'ecram_memoryio_write' [8;;https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wmissing-prototypes-Wmissing-prototypes8;;]
       >  1727 | ssize_t ecram_memoryio_write(const struct ecram_memoryio *ec_memoryio,
       >       |         ^~~~~~~~~~~~~~~~~~~~
       > /build/source/kernel_module/legion-laptop.c:3278:27: warning: no previous prototype for 'ec_to_wmi_powermode' [8;;https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wmissing-prototypes-Wmissing-prototypes8;;]
       >  3278 | enum legion_wmi_powermode ec_to_wmi_powermode(int ec_mode)
       >       |                           ^~~~~~~~~~~~~~~~~~~
       > /build/source/kernel_module/legion-laptop.c:3294:26: warning: no previous prototype for 'wmi_to_ec_powermode' [8;;https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wmissing-prototypes-Wmissing-prototypes8;;]
       >  3294 | enum legion_ec_powermode wmi_to_ec_powermode(enum legion_wmi_powermode wmi_mode)
       >       |                          ^~~~~~~~~~~~~~~~~~~
       > /build/source/kernel_module/legion-laptop.c:6058:19: error: initialization of 'void (*)(struct platform_device *)' from incompatible pointer type 'int (*)(struct platform_device *)' [8;;https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wincompatible-pointer-types-Werror=incompatible-pointer-types8;;]
       >  6058 |         .remove = legion_remove,
       >       |                   ^~~~~~~~~~~~~
       > /build/source/kernel_module/legion-laptop.c:6058:19: note: (near initialization for 'legion_driver.<anonymous>.remove')
       > cc1: some warnings being treated as errors
       > make[3]: *** [/nix/store/k3j2lh3hfc7r7s63zjyryydqxyis9wdk-linux-6.11.1-dev/lib/modules/6.11.1/source/scripts/Makefile.build:244: /build/source/kernel_module/legion-laptop.o] Error 1
       > make[2]: *** [/nix/store/k3j2lh3hfc7r7s63zjyryydqxyis9wdk-linux-6.11.1-dev/lib/modules/6.11.1/source/Makefile:1926: /build/source/kernel_module] Error 2
       > make[1]: *** [/nix/store/k3j2lh3hfc7r7s63zjyryydqxyis9wdk-linux-6.11.1-dev/lib/modules/6.11.1/source/Makefile:224: __sub-make] Error 2
       > make[1]: Leaving directory '/nix/store/k3j2lh3hfc7r7s63zjyryydqxyis9wdk-linux-6.11.1-dev/lib/modules/6.11.1/build'
       > make: *** [Makefile:13: all] Error 2
       For full logs, run 'nix log /nix/store/17cxk5fyi5wcfdg98wmkkwi246fg2jjw-lenovo-legion-module-0.0.12.drv'.
error: 1 dependencies of derivation '/nix/store/fvfzni3icx849fcxyc2c33cxpicnaww0-linux-6.11.1-modules.drv' failed to build
error (ignored): error: cannot unlink '"/tmp/nix-build-nvidia-x11-560.35.03-6.11.1.drv-4/build/NVIDIA-Linux-x86_64-560.35.03"': Directory not empty
error: 1 dependencies of derivation '/nix/store/dhzb2wrr22mncd2w077m9lbw8i4ib41d-nixos-system-handlink-24.11.20241004.bc947f5.drv' failed to build
```

DrVoidest, 2025-02-03:
I am also getting this on lenovo-legion-module-0.0.18. Error log: ```Running phase: unpackPhase
@nix { "action": "setPhase", "phase": "unpackPhase" }
unpacking source archive /nix/store/gxprayzqiqikai10iw9zhlggja4pb5da-source
source root is source/kernel_module
Running phase: patchPhase
@nix { "action": "setPhase", "phase": "patchPhase" }
Running phase: updateAutotoolsGnuConfigScriptsPhase
@nix { "action": "setPhase", "phase": "updateAutotoolsGnuConfigScriptsPhase" }
Running phase: configurePhase
@nix { "action": "setPhase", "phase": "configurePhase" }
no configure script, doing nothing
Running phase: buildPhase
@nix { "action": "setPhase", "phase": "buildPhase" }
build flags: SHELL=/nix/store/4fvc5fm8bszmkydng1ivrvr5cbvr1g60-bash-5.2p37/bin/bash ARCH=x86_64 CROSS_COMPILE= KBUILD_OUTPUT=/nix/store/6b9j17qfxdvyh9kmi9fjgh6zg0ix8if5-linux-6.13-dev/lib/modules/6.13.0/build SHELL=bash KERNELVERSI>
make -C /nix/store/6b9j17qfxdvyh9kmi9fjgh6zg0ix8if5-linux-6.13-dev/lib/modules/6.13.0/build M=/build/source/kernel_module modules
make[1]: Entering directory '/nix/store/6b9j17qfxdvyh9kmi9fjgh6zg0ix8if5-linux-6.13-dev/lib/modules/6.13.0/build'
make[2]: Entering directory '/build/source/kernel_module'
  CC [M]  legion-laptop.o
legion-laptop.c:6146:10: error: 'struct platform_driver' has no member named 'remove_new'; did you mean 'remove'?
 6146 |         .remove_new = legion_remove,
      |          ^~~~~~~~~~
      |          remove
legion-laptop.c:1816:16: warning: 'ecram_memoryio_write' defined but not used [-Wunused-function]
 1816 | static ssize_t ecram_memoryio_write(const struct ecram_memoryio *ec_memoryio,
      |                ^~~~~~~~~~~~~~~~~~~~
make[4]: *** [/nix/store/6b9j17qfxdvyh9kmi9fjgh6zg0ix8if5-linux-6.13-dev/lib/modules/6.13.0/source/scripts/Makefile.build:194: legion-laptop.o] Error 1
make[3]: *** [/nix/store/6b9j17qfxdvyh9kmi9fjgh6zg0ix8if5-linux-6.13-dev/lib/modules/6.13.0/source/Makefile:1989: .] Error 2
make[2]: *** [/nix/store/6b9j17qfxdvyh9kmi9fjgh6zg0ix8if5-linux-6.13-dev/lib/modules/6.13.0/source/Makefile:251: __sub-make] Error 2
make[2]: Leaving directory '/build/source/kernel_module'
make[1]: *** [/nix/store/6b9j17qfxdvyh9kmi9fjgh6zg0ix8if5-linux-6.13-dev/lib/modules/6.13.0/source/Makefile:251: __sub-make] Error 2
make[1]: Leaving directory '/nix/store/6b9j17qfxdvyh9kmi9fjgh6zg0ix8if5-linux-6.13-dev/lib/modules/6.13.0/build'
make: *** [Makefile:13: all] Error 2```

DrVoidest, 2025-05-02:
I am no longer having this issue. I have no idea when it got fixed.

closed 2025-05-04 by Mic92