NixOS/nixos-hardware #1174: framework 16 - CPU stuck below 1ghz under 100% load after sleep resume

Eddio0141, 2024-10-10:
After sleeping and resuming, the performance of the laptop drops by a significant amount
The CPU clock cannot go over 1ghz when at 100% or near 100% CPU load
Power profile hasn't changed, switching to other profiles and back to performance doesn't change anything

Mic92, 2024-10-23:
On other Intel hardware I have seen 600mhz as the minimal frequency. But this was often linked with some period of overheating. In any case this was only resolved by a reboot, which resets the power management.

Eddio0141, 2024-10-23:
The CPU temperature doesn't go above 50c while this is happening
If this is an option in power-profiles-daemon or something, I'd love to disable it

Mic92, 2024-10-24:
Might be something else than.

Sigmanificient, 2024-12-17:
I am currently running into a similar issue with a different laptop model. Running on a [i7-1370P](https://www.intel.com/content/www/us/en/products/sku/232146/intel-core-i71370p-processor-24m-cache-up-to-5-20-ghz/specifications.html) which has 14 cores, 20 thread and a a frequency range from 400Mhz to 5.2Ghz.

I am facing issues with the CPU frequency when trying to do heavy work (for instance, compiling a lib with cmake). In such scenario the frequency throttle down to 800Mhz:
```
$ cat /sys/devices/system/cpu/cpu*/cpufreq/scaling_cur_freq
700000
699992
699978
700006
699998
700003
700003
700005
699997
700009
699993
699992
700014
700000
700000
699992
700013
700000
700007
700000
```
(other method report similar frequencies)

---

I have set the latest available kernel from nixpkgs, using `boot.kernelPackages = pkgs.linuxPackages_latest;`, as I heard it had better support with newer CPU.

```
$ uname -a
Linux Sigmachine 6.12.3 #1-NixOS SMP PREEMPT_DYNAMIC Fri Dec  6 06:20:46 UTC 2024 x86_64 GNU/Linux
```

Then I started tweaking governor settings:

I made sure to use the following setting for powerManagement:
```nix
  powerManagement = {
    enable = true;
    cpuFreqGovernor = "performance";
  };
```

and also tried the following variant for this kernel parameters which i found on other similar discourse posts:
```
boot.kernelParams = [ "intel_pstate=active" ];
boot.kernelParams = [ "intel_pstate=disable" ];
boot.kernelParams = [ "intel_pstate=passive" ];
```

Disable this options, as a post suggested it could be part of the problem:
```nix
services.power-profiles-daemon.enable = false;
```

I also tried settings `cpufreq.min` setting to `5.0Ghz`, and tried `auto-cpufreq`:
```nix
  services.auto-cpufreq.enable = true;
  services.auto-cpufreq.settings = {
    battery = {
      governor = "powersave";
      turbo = "never";
    };
    charger = {
      governor = "performance";
      turbo = "auto";
      scaling_min_freq = 520000000;
    };
  };
```

bobzurad, 2025-02-18:
I was noticing this behavior as well. The CPU in my Framework 16 would not go above 1ghz after resuming from sleep, or when powering on the laptop without it plugged in.

I fixed it by adding amd_pstate=active to my kernel parameters. Note that adding this requires a Linux kernel of 6.3 or higher. I then installed this tool to manage the power profiles: https://github.com/jothi-prasath/auto-epp

Since then my CPU runs as expected at all times :)

Eddio0141, 2025-02-19:
Unless I'm wrong, isn't it doing exactly that in https://github.com/NixOS/nixos-hardware/blob/master/common/cpu/amd/pstate.nix
I think it's being set to passive due to my kernel version being `Linux 6.12.13 x86_64`
I'm on nixpkgs unstable, so I guess I just need to update it explicitly, or wait? Since using `inputs.nixos-hardware.nixosModules.framework-16-7040-amd` already includes `pstate.nix`, so

abmantis, 2025-05-22:
> Unless I'm wrong, isn't it doing exactly that in [`master`/common/cpu/amd/pstate.nix](https://github.com/NixOS/nixos-hardware/blob/master/common/cpu/amd/pstate.nix?rgh-link-date=2025-02-19T18%3A28%3A44.000Z) I think it's being set to passive due to my kernel version being `Linux 6.12.13 x86_64` I'm on nixpkgs unstable, so I guess I just need to update it explicitly, or wait? Since using `inputs.nixos-hardware.nixosModules.framework-16-7040-amd` already includes `pstate.nix`, so

It should use `active` for anything newer than 6.3, so 6.12 should be using `active`. You can check it with `cat /proc/cmdline` for example.

I am already using `active` and still have this issue, so it is not fixed by that.

boozedog, 2026-11-10:
Still present on a Framework 16 (Ryzen 7 7840HS, BIOS 04.05, kernel 7.2.15-zen1), and it did not require sleep. Uptime was 3 days. amd_pstate=active was already on the cmdline. Governor, EPP, and platform profile were all performance, scaling_max was 5.14 GHz.

Under an all-core compile the firmware delivered CPPC performance 38 against a software floor of 42 (~1.0 GHz). Tctl stayed ~50 °C. Switching profiles was not the missing step; they were already on performance. Unplugging the charger did not clear it (AC offline, pack discharging ~31 W, clocks still ~1 GHz). A reboot did.