NixOS/nixos-hardware #1176: Enabling USB on RPI CM4

bddvlpr, 2024-10-11:
When attempting to enable USB's on my PI CM4 I thought enabling xhci (through `hardware.raspberry-pi."4".xhci.enable = true;`) would fix it, but it in turn causes the following build error. Am I missing something or is this just not supported yet?

```
Processing source device tree broadcom/bcm2711-rpi-4-b.dtb...
  Applying overlay enable-xhci
  Applying overlay rpi4-cpu-revision
Saving final device tree broadcom/bcm2711-rpi-4-b.dtb...
Processing source device tree broadcom/bcm2711-rpi-400.dtb...
  Applying overlay enable-xhci
  Skipping overlay rpi4-cpu-revision: {'raspberrypi,4-model-b'} is incompatible with {'raspberrypi,400'>
Saving final device tree broadcom/bcm2711-rpi-400.dtb...
Processing source device tree broadcom/bcm2711-rpi-cm4-io.dtb...
  Applying overlay enable-xhci
Traceback (most recent call last):
  File "/nix/store/206l37y6cvcwl3ck1ppwgnxn4cnm818i-apply_overlays.py", line 102, in <module>
    main()
  File "/nix/store/206l37y6cvcwl3ck1ppwgnxn4cnm818i-apply_overlays.py", line 92, in main
    dt = apply_overlay(dt, overlay.fdt)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/206l37y6cvcwl3ck1ppwgnxn4cnm818i-apply_overlays.py", line 50, in apply_overlay
    raise FdtException(err)
libfdt.FdtException: pylibfdt error -1: FDT_ERR_NOTFOUND
...skipping...
Processing source device tree broadcom/bcm2711-rpi-4-b.dtb...
  Applying overlay enable-xhci
  Applying overlay rpi4-cpu-revision
Saving final device tree broadcom/bcm2711-rpi-4-b.dtb...
Processing source device tree broadcom/bcm2711-rpi-400.dtb...
  Applying overlay enable-xhci
  Skipping overlay rpi4-cpu-revision: {'raspberrypi,4-model-b'} is incompatible with {'raspberrypi,400'>
Saving final device tree broadcom/bcm2711-rpi-400.dtb...
Processing source device tree broadcom/bcm2711-rpi-cm4-io.dtb...
  Applying overlay enable-xhci
Traceback (most recent call last):
  File "/nix/store/206l37y6cvcwl3ck1ppwgnxn4cnm818i-apply_overlays.py", line 102, in <module>
    main()
  File "/nix/store/206l37y6cvcwl3ck1ppwgnxn4cnm818i-apply_overlays.py", line 92, in main
    dt = apply_overlay(dt, overlay.fdt)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/206l37y6cvcwl3ck1ppwgnxn4cnm818i-apply_overlays.py", line 50, in apply_overlay
    raise FdtException(err)
libfdt.FdtException: pylibfdt error -1: FDT_ERR_NOTFOUND
```

melektron, 2025-07-09:
I am have come across the same issue, and it seems to be related to the GPU (added in https://github.com/NixOS/nixos-hardware/pull/261) not working on CM4 either, as discussed in https://github.com/NixOS/nixos-hardware/issues/631.

As far as I understand, this is because enabling the fkms-3d modesetting patch removes all device tree variants except those of the Pi 4B and Pi 400 since https://github.com/NixOS/nixos-hardware/pull/693 because it failed to apply the overlays to the [bcm2711-rpi-cm4-io device tree variant](https://github.com/raspberrypi/linux/blob/rpi-6.6.y/arch/arm/boot/dts/broadcom/bcm2711-rpi-cm4-io.dts), which is also the case for the xhci it seems. When enabling the fkms-3d patch, even though the GPU doesn't actually work, the CM4 bootloader loads the Pi 4B device tree and therefore enables USB even without the xhci.enable setting. 

I am not exactly sure what the CM4-IO DT variant is used for, I cannot find any documentation about it from Raspberry Pi or anywhere else. It seems to be for the CM4 IO Board, but it removes a bunch of CPU peripherals (like GPU), so I cannot imagine it being a replacement for the regular CM4 version when running on the IO board. Does anyone else know about this?

I have gotten both the GPU and USB to work on the CM4 by enabling both the ```fkms-3d``` and ```xhci``` options and modifying [this](https://github.com/NixOS/nixos-hardware/blob/7ced9122cff2163c6a0212b8d1ec8c33a1660806/raspberry-pi/4/modesetting.nix#L35C4-L35C55) line to the following:
```nix
hardware.deviceTree.filter = "bcm2711-rpi-[4c][-0m][b04].dtb";
```
This allows both the 4B, 400 and CM4 variants, but still excludes the problematic CM4-IO. I seems to not cause any issues, so I don't know why it was decided against allowing the CM4 in https://github.com/NixOS/nixos-hardware/issues/631 and https://github.com/NixOS/nixos-hardware/pull/693 (except for the horrible glob syntax), but maybe I am missing something or something has changed since then. I am testing this using 25.05.

Also I don't quite understand which device trees we are modifying here, as the the ones included on the firmware partition (presumably loaded by the bootloader?) are [copied directly from the upstream raspberry pi firmware repository](https://github.com/NixOS/nixpkgs/blob/eb9dbcd4e116ecb69dde3040102f6a7bfa8b1aab/nixos/modules/installer/sd-card/sd-image-aarch64.nix#L81), where the cm4-io variant isn't even included... so when and by what component are these modified device trees loaded?

Majiir, 2025-07-20:
> I've also tried using dwc2 in every mode possible but none of them seem to make my ports work again.

As a workaround, I found this to work on a couple devices as long as `otg_mode` is **not** set in the firmware `config.txt` file.

```nix
{
  hardware.raspberry-pi."4".dwc2 = {
    enable = true;
    dr_mode = "host";
  };
}
```