NixOS/nixos-hardware #1199: `ssd.nix`:`fstrim` should not be enabled by default and only activated on `ext4` filesystems

ahydronous, 2024-10-21:
This is because `btrfs`, `zfs` and `xfs` get automatically mounted with `discard=async` since kernel 6.2.

Right now, this causes those systems to trim with much more frequency, effectively increasing wear.

Mic92, 2024-10-22:
Much more means here, once per week more. How often does the discard=async effectively run trim?

Mic92, 2024-10-22:
https://btrfs.readthedocs.io/en/latest/Trim.html

> the tool fstrim starts a trim operation on the whole filesystem, no mount options need to be specified, so it’s up to the filesystem to traverse the free space and start the trim, this is suitable for running it as periodic service

If async discard is enabled, does it still run trim with fstrim?

ahydronous, 2024-10-22:
`discard` is ran immediately with `sync` and as soon as possible with `async`. `fstrim` is ran whenever it is called upon by a cronjob or systemd.

But digging a bit deeper, it seems SSDs even as recent as the Samsung 800 series still have issues with queued trim commands. So out of caution / as a general rule, it should be the other way around, always `fstrim` and avoid using `discard`. Damn

closed 2025-05-21 by ahydronous

Mic92, 2025-05-30:
Ubuntu does the same.