NixOS/nixos-hardware #1201: Bluetooth is loaded but It does not work.

sincorchetes, 2024-10-22:
Hi folks,

I have system 24.05 version already installed but, I have kernel, linux-firmware and nvidia drivers from unstable.

uname -a: `Linux tank0 6.11.4 #1-NixOS SMP PREEMPT_DYNAMIC Thu Oct 17 13:27:02 UTC 2024 x86_64 GNU/Linux`

This is my motherboard: https://www.asus.com/motherboards-components/motherboards/tuf-gaming/tuf-gaming-x870-plus-wifi/

```
[    7.365501] Bluetooth: Core ver 2.22
[    7.365509] NET: Registered PF_BLUETOOTH protocol family
[    7.365510] Bluetooth: HCI device and connection manager initialized
[    7.365512] Bluetooth: HCI socket layer initialized
[    7.365514] Bluetooth: L2CAP socket layer initialized
[    7.365516] Bluetooth: SCO socket layer initialized
[    7.865402] Bluetooth: hci0: HW/SW Version: 0x00000000, Build Time: 20240816133202
[    9.195980] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    9.195984] Bluetooth: BNEP socket layer initialized
[   26.250299] Bluetooth: hci0: Device setup in 18109067 usecs
[   26.250303] Bluetooth: hci0: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[   26.410455] Bluetooth: MGMT ver 1.23
[   26.429836] Bluetooth: RFCOMM TTY layer initialized
[   26.429840] Bluetooth: RFCOMM socket layer initialized
[   26.429843] Bluetooth: RFCOMM ver 1.11

```

lsmod
```
lsmod |grep bt
btusb                  77824  0
btrtl                  32768  1 btusb
btintel                69632  1 btusb
btbcm                  24576  1 btusb
btmtk                  32768  1 btusb
bluetooth            1073152  44 btrtl,btmtk,btintel,btbcm,bnep,btusb,rfcomm
```

systemctl status bluetooth.service
```
● bluetooth.service - Bluetooth service
     Loaded: loaded (/etc/systemd/system/bluetooth.service; enabled; preset: enabled)
    Drop-In: /nix/store/5lavl80b872zxc3xv7qaiigm2q7y44cd-system-units/bluetooth.service.d
             └─overrides.conf
     Active: active (running) since Tue 2024-10-22 21:15:50 CEST; 3h 0min ago
       Docs: man:bluetoothd(8)
   Main PID: 1107 (bluetoothd)
     Status: "Running"
         IP: 0B in, 0B out
         IO: 1.9M read, 0B written
      Tasks: 1 (limit: 75921)
     Memory: 2.9M (peak: 3.3M)
        CPU: 27ms
     CGroup: /system.slice/bluetooth.service
             └─1107 /nix/store/4jk7imhlawwja56kdvqwiiznjv19zv91-bluez-5.75/libexec/bluetooth/bluetoothd -f /etc/bluetooth/main.conf

oct 22 21:16:08 tank0 bluetoothd[1107]: Endpoint registered: sender=:1.90 path=/MediaEndpoint/A2DPSource/opus_05
oct 22 21:16:08 tank0 bluetoothd[1107]: Endpoint registered: sender=:1.90 path=/MediaEndpoint/A2DPSink/opus_05_duplex
oct 22 21:16:08 tank0 bluetoothd[1107]: Endpoint registered: sender=:1.90 path=/MediaEndpoint/A2DPSource/opus_05_duplex
oct 22 21:16:08 tank0 bluetoothd[1107]: src/gatt-database.c:database_add_chrc() Failed to create characteristic entry in database
oct 22 21:16:08 tank0 bluetoothd[1107]: src/gatt-database.c:database_add_service() Failed to add characteristic
oct 22 21:16:08 tank0 bluetoothd[1107]: src/gatt-database.c:database_add_app() Failed to add service
oct 22 21:16:08 tank0 bluetoothd[1107]: src/gatt-database.c:client_ready_cb() Failed to create GATT service entry in local database
oct 22 21:18:32 tank0 bluetoothd[1107]: Controller resume with wake event 0x0
oct 22 23:55:30 tank0 bluetoothd[1107]: Controller resume with wake event 0x0
oct 22 23:56:24 tank0 bluetoothd[1107]: Controller resume with wake event 0x0
```

bluetoothctl
```
Waiting to connect to bluetoothd...
[bluetooth]# hci0 new_settings: powered bondable ssp br/edr le secure-conn wide-band-speech cis-central cis-peripheral iso-broadcaster sync-receiver
[bluetooth]# Agent registered
[bluetooth]# [CHG] Controller XX:XX:XX:5A:42:79 Pairable: yes
[bluetooth]# scan on
[bluetooth]# SetDiscoveryFilter success
[bluetooth]# Failed to start discovery: org.bluez.Error.InProgress
[bluetooth]# scan off
[bluetooth]# Failed to stop discovery: org.bluez.Error.Failed
```

sincorchetes, 2024-10-27:
Solved https://github.com/bluez/bluez/issues/995

closed 2024-10-27 by sincorchetes

rafeiropt, 2025-09-07:
I have the same problem, how you solved?
Thanks

sincorchetes, 2025-09-08:
> I have the same problem, how you solved? Thanks

I couldn't get it working; it was a hardware problem. I'd recommend buying a TP-Link Bluetooth adapter instead. This is the one I got, and it works perfectly. https://www.amazon.com/TP-Link-Bluetooth-Receiver-Controller-UB500/dp/B09DMP6T22/ref=sr_1_1?sr=8-1