NixOS/nixos-hardware #1202: Network Ethernet card module is loaded but It does not listed

sincorchetes, 2024-10-22:
Hi folks,

I have this motherboard https://www.asus.com/motherboards-components/motherboards/tuf-gaming/tuf-gaming-x870-plus-wifi/

The Ethernet card is recognised:
```
06:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller [10ec:8125] (rev 0c)
	DeviceName: Realtek RTL8125D LAN
	Subsystem: ASUSTeK Computer Inc. Device [1043:88e1]
	Kernel modules: r8169
```

The module is loaded:
```
r8169                 126976  0
mdio_devres            12288  1 r8169
libphy                225280  3 r8169,mdio_devres,realtek
led_class              20480  6 r8169,snd_hda_codec_generic,mt76,videodev,input_leds,asus_wmi
```

But It does not list
```
ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute
       valid_lft forever preferred_lft forever
2: wlp7s0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether ___ brd ff:ff:ff:ff:ff:ff
    inet _____ brd _______ scope global dynamic noprefixroute wlp7s0
       valid_lft 32069sec preferred_lft 32069sec
    inet6 _____ scope link noprefixroute
       valid_lft forever preferred_lft forever
```

nmcli
```
GENERAL.DEVICE:                         wlp7s0
GENERAL.TYPE:                           wifi
GENERAL.HWADDR:                        _____________-
GENERAL.MTU:                            1500
GENERAL.STATE:                          100 (connected)
GENERAL.CONNECTION:                     MOVISTAR_PLUS_15D8
GENERAL.CON-PATH:                       /org/freedesktop/NetworkManager/ActiveConnection/2
IP4.ADDRESS[1]:                         _________________
IP4.GATEWAY:                            ______________-
IP4.ROUTE[1]:                           dst = -_______________-, nh = 0.0.0.0, mt = 600
IP4.ROUTE[2]:                           dst = 0.0.0.0/0, nh = -__________-, mt = 600
IP4.DNS[1]:                             ---________________-
IP4.DNS[2]:                             -___________-
IP6.ADDRESS[1]:                         -________________--
IP6.GATEWAY:                            --
IP6.ROUTE[1]:                           dst = fe80::/64, nh = ::, mt = 1024

GENERAL.DEVICE:                         lo
GENERAL.TYPE:                           loopback
GENERAL.HWADDR:                         00:00:00:00:00:00
GENERAL.MTU:                            65536
GENERAL.STATE:                          100 (connected (externally))
GENERAL.CONNECTION:                     lo
GENERAL.CON-PATH:                       /org/freedesktop/NetworkManager/ActiveConnection/1
IP4.ADDRESS[1]:                         127.0.0.1/8
IP4.GATEWAY:                            --
IP6.ADDRESS[1]:                         ::1/128
IP6.GATEWAY:                            --

GENERAL.DEVICE:                         p2p-dev-wlp7s0
GENERAL.TYPE:                           wifi-p2p
GENERAL.HWADDR:                         (unknown)
GENERAL.MTU:                            0
GENERAL.STATE:                          30 (disconnected)
GENERAL.CONNECTION:                     --
GENERAL.CON-PATH:                       --
```

dmesg
```
[    0.042998] percpu: Embedded 86 pages/cpu s229376 r8192 d114688 u524288
[    0.043001] pcpu-alloc: s229376 r8192 d114688 u524288 alloc=1*2097152
[    7.258298] r8169 0000:06:00.0: error -ENODEV: unknown chip XID 688, contact r8169 maintainers (see MAINTAINERS file)
```

Mic92, 2024-10-23:
So

```
[    7.258298] r8169 0000:06:00.0: error -ENODEV: unknown chip XID 688, contact r8169 maintainers (see MAINTAINERS file)
```

Seems to be the issue.
Have you tried the latest linux kernel as well?

Mic92, 2024-10-23:
Ok. looks like you tried 6.11: https://github.com/NixOS/nixos-hardware/issues/1201

Mic92, 2024-10-23:
https://github.com/torvalds/linux/blob/c2ee9f594da826bea183ed14f2cc029c719bf4da/drivers/net/ethernet/realtek/r8169_main.c#L5426 I think you than actually need to contact the linux kernel maintainer of r8169

Mic92, 2024-10-23:
If you are lucky it's a one-line fix: https://github.com/torvalds/linux/blob/c2ee9f594da826bea183ed14f2cc029c719bf4da/drivers/net/ethernet/realtek/r8169_main.c#L2233

sincorchetes, 2024-10-25:
The r8169 module was reply via email:

> Thanks for the report. This is a new chip version (RTL8125D) which isn't supported yet.
> This has been reported few days ago already, and adding support for this chip version is
> in progress. It should land in linux-next in the next days, and then in kernel 6.13.
> You'll need the firmware for this chip version (rtl8125d-1.fw), which has been submitted
> to linux-firmware few days ago. So please ensure that you have an up-to-date linux-firmware
> package.

Mic92, 2024-11-02:
Giving it will be eventually fixed, we can close this and wait?

closed 2024-11-02 by Mic92

sincorchetes, 2024-11-04:
Mmm

I've installed the latest linux-firmware and It does not solve the issue. :/

default.nix
```
with import <nixpkgs> {};

stdenv.mkDerivation {
  name = "linux-firmware-custom";
  version = "1.0";

  src = fetchFromGitLab {
    owner = "kernel-firmware"; # El propietario del repositorio
    repo = "linux-firmware";    # El nombre del repositorio
    rev = "main";               # La rama o revisión que deseas usar
    sha256 = "8ozeLxvDKAMQTl89YvDje3e93wRc1enrha2Nvdy9Y3A=";
  };

  installPhase = ''
    mkdir -p $out/lib/firmware
    cp -r * $out/lib/firmware/
  '';
}
```

I've added into the config

```
environment = {
    systemPackages = with pkgs; [
       ...
       bridge-utils
       (import /home/sincorchetes/build/default.nix)
       #linux-firmware
    ];
  };

```

List the firmware files:
```
[sincorchetes@tank0:~]$ ls /nix/store | grep linux-firmware
092rdbfz7r2n0xnd028k4g3w4ag610ar-linux-firmware-main.tar.gz.lock
0crsxhjlghhzni2vp69ab451718y0li9-linux-firmware-custom.drv
11mk3bjxamqwyzjbzzjl62dbahim06w8-linux-firmware-custom-1.0
6jg7rxypn5gscsz729n2kig5m9msj8nj-linux-firmware-custom
77cjkh6j4z13m4h008sccg8xgmj2dsz3-linux-firmware-custom-1.0.drv
bkl8dfkwjjpy933bd06zfhypfci77819-linux-firmware-main.tar.gz
dr0m3xvsfk6l40g17hqp1yrx8lq87m2g-linux-firmware-20241017.drv
drmvca85kva8f3qj29gnd7x59b1bqbi0-linux-firmware-20241017.drv
dy975n4qn6gm4vn8dvim9c7glv1g4lb0-linux-firmware-custom-1.0.drv
gm5vg2x9zrm18a7wv16gngskzj0x11md-linux-firmware-main.tar.gz.drv
gx1bj4kdyfhxbs1kznrzk9hwg5dfdcmh-linux-firmware-20241017-zstd.drv
i4az1q57hywag64bjl9caaizlbfbqvrg-linux-firmware-custom-1.0.drv
jxr2rzdjn3s4s7dfvqmni0snnd5s9pqp-linux-firmware-20241017.drv
kwbn3axxzjpvin4pbzxnj7x54hglflsp-linux-firmware-custom-1.0.drv
lwyxgz7brfks89jm8l612mmwyv2k7nh5-linux-firmware-20241017-zstd
m9h4llqk64psbqwzc6h2qh39fz58y67c-linux-firmware-20241017
ma26777nfavb9p26fpr16gaaj9h5xbf6-linux-firmware-custom-1.0.drv
pxsnb09mr9k1mxzz94axb5vfija2l13c-linux-firmware-20241017-zstd.drv
r9xyf8mz81hfw8hpqcqd16c41jaxsi1w-linux-firmware-20241017-zstd
x4vgz3arb2zdrhdj5cvg9m90dbpcq9xr-linux-firmware-20241017
xphyvyp9d5vj3ahbpsf5sccp2n3lz13k-linux-firmware-main.tar.gz
zxk06npclwcb1l17cs1v4vd7cyzvlwag-linux-firmware-20241017-zstd.drv

```

[    4.252381] r8169 0000:06:00.0: error -ENODEV: unknown chip XID 688, contact r8169 maintainers (see MAINTAINERS file)

Do I have to wait for the new kernel?

Mic92, 2024-11-04:
Yes. The check happens in the linux kernel, not in the firmware file.