NixOS/nixos-hardware #1205: Intel & AMD cpu: add config (off by default) option that disables relevant security mitigations for huge performance uplift

ahydronous, 2024-10-24:
I have a blurb in my own nixos config for certain CPUs to disable either `retbleed` and/or `downfall` mitigations.

Both of these are pretty much lab-only exploits that are virtually impossible to exploit without extreme setup and conditions. And at least for `retbleed`, the primary danger is to cloud providers, not personal computers. This is not worth paying a 20-40% (average 22.5%) performance cost for.

Nonetheless, I understand it would make people uncomfortable, so this "smart mitigations" option should be off by default.

`retbleed` affects Intel 6th-8th gen and AMD Zen1-ZEn2+ afaik.
`downfall` affects Intel 6th-11th gen.

Mic92, 2024-12-02:
When you say cloud providers are mainly affected, what types of attacks does this entail?
VM escapes? Is my browser also a VM in this case?

abowen, 2025-03-13:
How different is this from

```
  boot.kernelParams = [
    "mitigations=off"
  ];
```

Mic92, 2025-03-13:
@abowen `mitigations=off` has a bigger impact as it includes more the mitigations named here.