NixOS/nixos-hardware #1207: Lenovo Legion 5 15ARH05

SoSeDiK, 2024-10-26:
There's already a profile for [15ARH05H](https://github.com/NixOS/nixos-hardware/blob/master/lenovo/legion/15arh05h/default.nix) model, and it seems to work fine so far with my `15ARH05`.
https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/legion-series/legion-5-15arh05/

Possibly could be almost 1:1 copy, the only issue I noticed is a different AMD GPU bus id, which is `PCI:5:0:0` instead of `PCI:6:0:0`:
```sh
lspci | grep -i vga                        
01:00.0 VGA compatible controller: NVIDIA Corporation TU117M [GeForce GTX 1650 Ti Mobile] (rev a1)
05:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Renoir [Radeon Vega Series / Radeon Vega Mobile Series] (rev c7)
```

LucasFA, 2024-12-23:
Could this be a case similar to #1038? Definitely sounds like there is more than one version, and it is unlikely than the original creator simply missed the correct bus ID.

SoSeDiK, 2024-12-23:
It is similar indeed, Lenovo seems to have three versions for this model:

<details>
<summary>Lenovo product search</summary>

[image]

</details>

My test setup for my model works so far and seems to be correct on paper, but I'm holding back due to a couple of unresolved issues with NVIDIA GPU (not properly working sleep/suspend on both open and closed driver).

LucasFA, 2024-12-23:
I missed the H being missing in your3 title and first line. Woops.
Regarding your sleep issues, have you seen [#336723](https://github.com/NixOS/nixpkgs/issues/336723) or directly this [Discourse thread](https://discourse.nixos.org/t/suspend-resume-cycling-on-system-resume/32322/9)?

SoSeDiK, 2024-12-24:
I'm not on GNOME (mostly running Hyprland), so sadly the fix doesn't apply.

Suspend works most of the time (at very rare occasions screen turns off, but laptop still works; in that case screen turns on from any key/mouse action and then suspend will work from the second try), the issues mostly arise upon waking up from the sleep as I've had the laptop freeze a bunch of times.

I also can't get neither S01x nor D3 to work despite reports on the Internet that it works for other people with the same/similar laptop/GPU models (the usual workaround of closed driver with `NVreg_EnableGpuFirmware=0` didn't work).

<details>
<summary>Details</summary>

cat /proc/driver/nvidia/gpus/0000\:01\:00.0/power

Runtime D3 status:          Not supported
Video Memory:               Active

GPU Hardware Support:
 Video Memory Self Refresh: Not Supported
 Video Memory Off:          Supported

S0ix Power Management:
 Platform Support:          Not Supported
 Status:                    Disabled

</details>

For some more context, I'm on nixos-unstable with 6.12.5-xanmod1 kernel and stable 565.77 closed nvidia driver.

SoSeDiK, 2024-12-26:
I suspect the finegrained power management (`hardware.nvidia.powerManagement.finegrained`) to be the cause of my issues. Disabling it and using instead `NVreg_DynamicPowerManagement=0x01` (because otherwise it's disabled by default for Turing GPUs) enables RTD3:

<details>
<summary>cat /proc/driver/nvidia/gpus/0000\:01\:00.0/power</summary>

```sh
Runtime D3 status:          Enabled (coarse-grained)
Video Memory:               Active

GPU Hardware Support:
 Video Memory Self Refresh: Not Supported
 Video Memory Off:          Supported

S0ix Power Management:
 Platform Support:          Not Supported
 Status:                    Disabled
 ```

</details>

I've done a few dozen suspend & hibernate cycles without issues: laptop properly shutdowns and wakes up.
I've also switched to the open nvidia driver and had no issues either after a few cycles (same kernel and driver version as before).

Setting `NVreg_DynamicPowerManagement=0x02` makes it `Not supported` again, not sure why. Can't get Video Memory Self Refresh and S0ix to work either, even though they supposedly are supported by the hardware?

LucasFA, 2024-12-26:
>I suspect the finegrained power management (hardware.nvidia.powerManagement.finegrained) to be the cause of my issues. 

I just looked into this and realised the card does not (officially) support the option on the open modules. The [finegrained](https://search.nixos.org/options?channel=24.11&show=hardware.nvidia.powerManagement.finegrained&from=0&size=50&sort=relevance&type=packages&query=finegrained) option mentions the [chapter 22 of the nvidia docs](https://download.nvidia.com/XFree86/Linux-x86_64/565.77/README/dynamicpowermanagement.html#KnownIssuesAndW6426e), which do say it supports Turing and newer, but the [open kernel modules docs](https://download.nvidia.com/XFree86/Linux-x86_64/565.77/README/kernel_open.html) say that they only support RTD3 from Ampere (30 series) onwards. Hence why the closed driver should make it work, but, well, your empirical testing seems to show that no.

Have you tested manually adding the `NVreg_DynamicPowerManagement=0x01` to the kernel boot options, with the closed driver? Instead of using the `hardware.nvidia.powerManagement.enabled` option. I'm thinking there's something going on with the default, which is `0x03` and [it disables RTD3 features](https://download.nvidia.com/XFree86/Linux-x86_64/565.77/README/dynamicpowermanagement.html#DriverSettingsfe624)

SoSeDiK, 2024-12-26:
> Have you tested manually adding the `NVreg_DynamicPowerManagement=0x01` to the kernel boot options, with the closed driver? Instead of using the `hardware.nvidia.powerManagement.enabled` option. I'm [...]

Default `0x03` disables dynamic power management by default for Turing, so the manual override is needed indeed.

Basic coarse-grained `0x01` power management works on both open and closed drivers, fine-grained `0x02` doesn't.

There are a bunch of reports like https://github.com/pop-os/nvidia-graphics-drivers/issues/216 on the Internet, which make me think that the GPU itself should support fine-grained too, but the workarounds didn't work for me.

SoSeDiK, 2024-12-27:
I've dug up some threads from years ago, notably like this one: https://forums.developer.nvidia.com/t/nvidia-d3-runtime-power-management-only-works-on-window/196553/3.

According to the users with the same issue, NVIDIA blocked fine-grained RTD3 on Linux on Ryzen 4xxx and 3xxx models (I have Ryzen 5 4600H) despite the hardware support being there and it working fine on Windows. I couldn't find any official statements from NVIDIA or sources besides users just reporting the issue, but seems plausible, and sucks if that still the case after all these years.

My system advertises support for `S0ix` mode too, defaulting to `S3`:
```sh
cat /sys/power/mem_sleep
s2idle [deep]
```

But NVIDIA's unhappy, and booting with `s2idle` breaks suspend, causing system freezing.

4JX, 2025-02-07:
Skimmed through the conversation here, I'm a `16ach6h` (5 Pro 2021, 5800H, 3070m) user.

For the PCI ID dilemma, it most commonly depends on [how many drives](https://github.com/NixOS/nixos-hardware/tree/master/lenovo/legion/16ach6h#using-multiple-drives-with-this-configuration) you have in your system. For my device we opted to leave it at `5:0:0` as a default, and then just have people [manually override](https://github.com/4JX/nixos-config/blob/ad9185d732a75644f29e728b22eb7e68e0620152/hosts/terra/hardware-configuration.nix#L45) it as needed.

> Suspend works most of the time (at very rare occasions screen turns off, but laptop still works; in that case screen turns on from any key/mouse action and then suspend will work from the second [...]

The sleep issues mentioned here also apply to me and a few other users (there's other issues in this repo about it as well, haven't checked non-legions) with exactly the same general symptoms. Using GNOME with Wayland, so I'll fiddle around and comment on the relevant issue.