NixOS/nixos-hardware #1219: Display configuration for Asus tuf A14 FA401WV

WillerWasTaken, 2024-10-31:
Hello there,

Just received a new laptop that I am setting up and trying to get it to work with nixos. I had some issue with the amdgpu igpu but I believe it might too too recent to I disabled it in the BIOS to solely use Nvidia dgpu.

I am importing the configuration like so
```
  imports =
    [ # Include the results of the hardware scan.
      /etc/nixos/hardware-configuration.nix
      "${builtins.fetchGit { url = "https://github.com/NixOS/nixos-hardware.git"; }}/common/gpu/nvidia"
      "${builtins.fetchGit { url = "https://github.com/NixOS/nixos-hardware.git"; }}/common/hidpi.nix"
    ];
```
This seems to be almost working, the main issue I have is that all GTK windows are zoomed in like so
[image]
Terminal and status bar have normal size but the browser is super zoomed in which is odd.

To make it work I found [that issue](https://discourse.nixos.org/t/how-to-correct-nvidia-driver-resolution/14183/3) that is adding
```
  services.xserver.dpi = 97;
  environment.variables = {
    GDK_SCALE = "0.5";
  };
```
to the configuration and it is actually working but seems like a weird workaround.

One more weird behavior is that I have only one resolution available on xrandr for the main display (laptop screen)

```
$ xrandr
Screen 0: minimum 8 x 8, current 6000 x 1600, maximum 32767 x 32767
DP-0 disconnected (normal left inverted right x axis y axis)
DP-1 disconnected (normal left inverted right x axis y axis)
HDMI-0 connected primary 3440x1440+2560+0 (normal left inverted right x axis y axis) 800mm x 334mm
   3440x1440     84.96*+  49.95  
   3840x2160     59.94    50.00    29.97    25.00    23.98  
   2560x1440    120.00  
   2560x1080    119.88   100.00    59.94    50.00  
   1920x1080     60.00    59.94    50.00  
   1280x720      59.94  
   1024x768      60.00  
   800x600       60.32  
   720x576       50.00  
   720x480       59.94  
   640x480       59.94    59.93  
DP-2 connected 2560x1600+0+0 (normal left inverted right x axis y axis) 302mm x 188mm
   2560x1600     60.00*+ 165.00
```
DP-2 corresponds to the main laptop screen and is only available in 2560x1600.

Any idea of what could be optimized in the configuration?

Mic92, 2024-11-02:
Firefox has its own dpi option called `layout.css.devPixelsPerPx` in about:config.
I think this is a software issue, not a hardware one. It works these days if you use wayland-based compositors for example i.e. I don't see this problem in KDE, but I remember having it before.

closed 2024-11-02 by Mic92

WillerWasTaken, 2024-11-02:
I does not seem to be specific to firefox

[image]

Every gtk application are zoomed in like 200%. Here there's also `qgit`, `keepassxc` and even the i3 tabs that are bigger that they should be. Only the terminal and the status bar are displayed correctly.

Even when logging in, the logging screen is way bigger (I use lightdm). There's probably a misconfiguration, the only thing that works for me is to add
```
  services.xserver.dpi = 97;
  environment.variables = {
    GDK_SCALE = "0.5";
  };
```

Only then will I have the correct size for all windows, browser, i3 tabs, gtk app etc and not only the terminal + status bar

Mic92, 2024-11-02:
Okay. We cannot have environment variables though by default. They are not working correctly when people use wayland and there is no way to detect this.