NixOS/nixos-hardware #1256: A Framework AMD fingerprint question

3timeslazy, 2024-11-18:
Hello everyone, 

The [/framework/13-inch/7040-amd](https://github.com/NixOS/nixos-hardware/tree/master/framework/13-inch/7040-amd#getting-the-fingerprint-sensor-to-work) README says that in order to activate the fingerprint support one must downgrade the fwupd package.

However, recently I imported the nixos-hardware module, rebuild the system, executed `sudo fprintd-enroll $USER` and the fingerprint just worked without any additional work.

Do you think there was a fix somewhere that made it possible to active fingerprint support without pinning the fwupd version or is it me missing something? 

I'm using the latest 3.05 13-inch amd firmware version

LucasFA, 2024-12-22:
I don't have this hardware, but this [guide in the framework website](https://community.frame.work/t/guide-lvfs-fingerprint-reader-firmware-for-framework-laptop-13-13th-gen-intel-core-amd-7040-series-fingerprint-reader-on-fedora-39-and-ubuntu-22-04/41898) seems to be about this issue.

It seems it was an upstream issue in fwupd according to an article linked in [this comment](https://community.frame.work/t/guide-lvfs-fingerprint-reader-firmware-for-framework-laptop-13-13th-gen-intel-core-amd-7040-series-fingerprint-reader-on-fedora-39-and-ubuntu-22-04/41898/47). Article [here](https://9to5linux.com/fwupd-1-9-14-fixes-updating-the-fingerprint-reader-on-framework-laptops).

The README should be updated. Are you up for that? As far as I know it's just deleting that section

griffi-gh, 2025-01-04:
Yeah, had the same experience
worked for me OOTB with up-to-date drivers

closed 2025-01-07 by mergify[bot]