NixOS/nixos-hardware #1278: Intel Xe driver on older platorms (Tiger Lake)

matdibu, 2024-12-18:
There doesn't seem to be a consensus on whether Xe should be used on older platforms, like Tiger Lake

https://docs.kernel.org/next/gpu/rfc/xe.html#xe-platforms

The `common/gpu/intel/tiger-lake/default.nix` module enables Xe:
https://github.com/NixOS/nixos-hardware/blob/b12e314726a4226298fe82776b4baeaa7bcf3dcd/common/gpu/intel/tiger-lake/default.nix#L5

But Xe won't be loaded unless something like this is configured:
```nix
boot.extraModprobeConfig = ''
  options xe force_probe=9a60
  options i915 force_probe=!9a60
'';
```

The jury is still out on Xe stability, some call it trash, others call it perfect

So, should Xe be `force_probe`'d in nixos-hardware?

Mic92, 2025-01-07:
We can enable it on selected devices where people tested it. Otherwise I would rely on kernel defaults.

matdibu, 2025-01-07:
sure, but then this issue still stands
we're "enabling" it in the nix module, but it won't actually work on Tiger Lake unless `force_probe`'d

so that means we should take `Xe` out of `nixos-hardware/common/gpu/intel/tiger-lake/default.nix` and let it be selected in device-specific modules, right?

Mic92, 2025-01-07:
We should probably remove it from that file than.

closed 2025-01-09 by Mic92

dimitarg, 2025-04-19:
FWIW, I ran a test on Alder lake, `boot.kernelParams = [ "i915.force_probe=!46a6" "xe.force_probe=46a6" ];`
For me, there's a 0.5 second screen tear during boot stage 1, reproducible consistently. Syslog is also full of warnings that this is not officially supported (mesa, etc).

Probably not worth it unless you know what you're doing, which I don't.