NixOS/nixos-hardware #1295: Framework 13 hardware.framework.laptop13.audioEnhancement not working

neutralinsomniac, 2025-01-03:
I can't seem to get the "Framework Speakers" pipewire sink to show up. I have my configuration set up here: https://github.com/neutralinsomniac/nix/blob/master/hw/theseus/default.nix and it seems to be getting used, but I don't see any new sinks being created:

```
[jeremy@theseus:~/nix]$ nix repl
Nix 2.24.11
Type :? for help.
nix-repl> :lf .#
Added 9 variables.

nix-repl> nixosConfigurations.theseus.config.services.pipewire.wireplumber.configPackages
[
«derivation /nix/store/mz75271vf68drrasp5x9abhwsi5js4pp-99-laptop.conf.drv»
]

[jeremy@theseus:~/nix]$ ls -l /nix/store/mz75271vf68drrasp5x9abhwsi5js4pp-99-laptop.conf.drv
-r--r--r-- 2 root root 2472 Dec 31  1969 /nix/store/mz75271vf68drrasp5x9abhwsi5js4pp-99-laptop.conf.drv
```

I also seem to have way too many soundcards in kde's audio device configurator:
[image]

neutralinsomniac, 2025-01-03:
Figured it out. Had to remove `hardware.pulseaudio.enable=true;` from my config!

closed 2025-01-03 by neutralinsomniac

Titaniumtown, 2025-01-29:
ran across something similar, had to do `hideRawDevice = false;`