NixOS/nixos-hardware #1298: hardware.framework.laptop13.audioEnhancement.enable causes audio profile to be used for headphones as well

neutralinsomniac, 2025-01-06:
The pipewire/wireplumber config that gets applied to the speakers sounds great, but the profile also seems to affect the headphone jack, which sounds really bad. I'm not sure if there's a way to prevent the profile from being applied to the headphone jack too, but that's probably what we want here.

Mic92, 2025-01-07:
cc @miketth

griffi-gh, 2025-01-15:
just wanted to report this as well. 
also, it hides the headphone/speaker port selection, but that's prob not possible to fix.

miketth, 2025-01-16:
Unfortunately, this is a known issue; PipeWire/WirePlumber does not have a way to apply a sound profile to only the speaker, since it's the same device as the headphones from its perspective. The workaround is to set `framework.laptop13.audioEnhancement.hideRawDevice = false;`. This also solves the headphone/speaker port selection not being visible (they are only visible under the normally hidden "raw" device).

There's a more complicated fix outlined in the original PR: https://github.com/NixOS/nixos-hardware/pull/1253#issuecomment-2494687041

Feel free to highlight these caveats in the option's description (or I can submit a PR for that too). Unfortunately these two issues are known, but I don't use the headphone jack, so I don't plan on fixing them. If you find a way that does not cause these issues, feel free to submit a PR or point me in a direction that I can submit one (without the complicated alsa tinkering).

nyabinary, 2025-01-22:
> Unfortunately, this is a known issue; PipeWire/WirePlumber does not have a way to apply a sound profile to only the speaker, since it's the same device as the headphones from its perspective. The workaround is to set `framework.laptop13.audioEnhancement.hideRawDevice = false;`. This also solves the headphone/speaker port selection not being visible (they are only visible under the normally hidden "raw" device).
> 
> There's a more complicated fix outlined in the original PR: [#1253 (comment)](https://github.com/NixOS/nixos-hardware/pull/1253#issuecomment-2494687041)
> 
> Feel free to highlight these caveats in the option's description (or I can submit a PR for that too in the coming days). Unfortunately these two issues are known, but I don't use the headphone jack, so I don't plan on fixing them. If you find a way that does not cause these issues, feel free to submit a PR or point me in a direction that I can submit one (without the complicated alsa tinkering).

Prob should just submit a PR  :3